jbruce12000/kiln-controller #97: Ignore emergencies

captaincaden, 2022-06-18:
Within one of the past updates, the ignore emergency feature was added. 

During my firing this morning, a single line of an unkown thermocouple error (???:True) was logged, and (as it's written) the controller stopped controlling, and stopped output to the kiln. Even with my notification additions in place that texted me the thermocouple had an error, it still took about 30 minutes before I saw the message and was able to check the status, which lost about 150 ish degrees. I have the kiln set up so that in the event there's an issue with the controller, just by diverting power the kiln can run on it's own controls again (which for a glaze firing, is not a super big deal to me) but I wish there were a better option to restart than having to use the API  and a time basis to restart in the middle of a schedule. 

I think ignore emergencies need to be something we can do on an individual basis, not a single ignore all emergencies function. While I want it to ignore unknown errors, I don't want it to ignore temp exceed, or ignore NC thermocouple errors. While I can write this in for myself, I figured it'd be better to bring this issue up here, because I'd imagine anyone else who uses the ignore emergency feature probably also doesn't want to ignore ALL emergency conditions. The ???: True emergency only happens rarely, and every time it's happened in the past, it's been during cooling or idle, never while it was actually running, and everytime it happens, I always load the web socket and see that there's no interruption to temp values, and the temp is still being logged in the console and accurate. 

My question regarding ignore emergencies is, say ignore emergencies is active, and a total disconnection to the thermocouple occurs, and temp input is null, what happens? I'd imagine the schedule just see's a major 'kiln must catch up' and outputs max power? 

Touching back on the interruption part, it's something that has been on my mind with this controller project from the start, I only add that here because I've also been trying to think of a way that could solve it, but it seems like it would be challenging to implement. I wish there was a way that if there's an error, or a power outage, etc. etc. and the need arises to restart a schedule, I could simply restart it by pressing start on that schedule again. That if the kiln was at say 1400 it would be able to just pick up where the 1400 degree point was in the schedule (like how a commercial controller would do). This last part is just my thoughts, from my understanding of the script, I understand why this wouldn't exactly be easy to do.

captaincaden, 2022-06-18:
Adding here because I just noticed the honour_thermocouple_short_errors feature, but am unsure if it does the one part that I was questioning or not.

jbruce12000, 2022-06-18:
If there is no connection to the thermocouple and config.ignore_emergencies=True... then no reset would happen. Normally 40 temps are read every duty cycle [2s by default]... in this case, no temps would be read at all. The temps array would be empty. I'm guessing this would expose a bug and the temperature would be static, unchanging from the last successful average over the duty cycle.

I have not tried this... but looking at the code, I think that's what would happen.

jbruce12000, 2022-06-18:
Not that my watcher is any improvement over yours, but I did write a watcher script that alerts after 60s [by default].

https://github.com/jbruce12000/kiln-controller/blob/master/docs/watcher.md

jbruce12000, 2022-06-18:
I'll have to look at the code and think about how to handle some things being an emergency and other things not.

As far as automated or manual restarts, you'd need a few things recorded every duty cycle. Maybe just write get_state to a json file every duty_cycle.

1. profile that was running
2. time elapsed in that profile
3. timestamp of last successful duty cycle

You'd have to pick a reasonable window for restarts [maybe 10 or 15 minutes], after which it's too late and you just want the software to give up and NOT restart. If it's inside that window, restart the last running profile where it was.

captaincaden, 2022-06-18:
> If there is no connection to the thermocouple and config.ignore_emergencies=True... then no reset would happen. Normally 40 temps are read every duty cycle [2s by default]... in this case, no temps [...]

That's about along the lines I thought would happen. If there is nothing in the temps array, does it continue to output? or would it stop? My thermocouple is easy to disconnect, so I could theoretically test it and see how it reacted while running a profile, and while ignore_emergencies=True. 

> Not that my watcher is any improvement over yours, but I did write a watcher script that alerts after 60s [by default].
> 
> https://github.com/jbruce12000/kiln-controller/blob/master/docs/watcher.md

I use both yours and mine. Mine is just an SMTP function in the oven.py file that piggybacks off of already defined variables like temp and the thermocouple errors and excessive temp functions, so the text/email got sent immediately when oven.py determined that the ???:True condition had arised. The other part there is it runs inside of the script, so anytime kiln-controller is running, and my configuration for notifications is set to true in config then notifications are sent at those predefined variables or events. The downside of the watcher.py script is that I have to have something else up and running to run the script, and in this case, I didn't have your watcher running yet.  However, when I'm not at home, having both your watcher running and the notifications I have written alongside the camera's that are on my kilns, it makes a great team of making sure everything is working well and as expected if I have to leave the kiln alone for a bit. That being said, I still am usually around when it's firing, but it's nice to have all of these measures watching out. 

> I'll have to look at the code and think about how to handle some things being an emergency and other things not.

I was thinking the easiest way to do that would just be to make separate config.ignore(said event) here, and in the oven.py file where the reset is called for each of those, make them correspond to the individual config.ignore(event), instead of just a grand ignore all emergencies configuration. But that was just my way of understanding it, and I didn't dig too far into seeing how you currently have ignore_emergencies set up.

captaincaden, 2022-06-18:
Something like: 

in config.py:
`# There are all kinds of emergencies that can happen including:
# - temperature is too high (emergency_shutoff_temp exceeded)
# - lost connection to thermocouple
# - unknown error with thermocouple
# - too many errors in a short period from thermocouple
# and some people just want to ignore all of that and just log the emergencies but do not quit
ignore_temp_too_high = False
ignore_lost_connection_tc = False
ignore_unknown_tc_error = False
ignore_too_many_tc_errors = False`

and then in oven.py:
`    def reset_if_emergency(self):
        '''reset if the temperature is way TOO HOT, or other critical errors detected'''
        if (self.board.temp_sensor.temperature + config.thermocouple_offset >=
            config.emergency_shutoff_temp):
            log.info("emergency!!! temperature too high")
            if not config.ignore_temp_too_high== True:
                self.reset()

        if self.board.temp_sensor.noConnection:
            log.info("emergency!!! lost connection to thermocouple")
            if not config.ignore_lost_connection_tc== True:
                self.reset()

        if self.board.temp_sensor.unknownError:
            log.info("emergency!!! unknown thermocouple error")
            if not config.ignore_unknown_tc_error == True:
                self.reset()

        if self.board.temp_sensor.bad_percent > 30:
            log.info("emergency!!! too many errors in a short period")
            if not config.ignore_too_many_tc_errors == True:
                self.reset()`
                
                
This is just how I was going to test doing it, but again, I'm not sure if you used ignore_emergencies anywhere else or not, or if this would even work.

captaincaden, 2022-06-27:
I looked through the code a bit to see if you had used config.ignore_emergencies anywhere other than in the part of oven.py that's responsible for acting on those different emergencies, and didn't find anything, so I went ahead and implemented the idea mentioned in my response above, simply because I'd prefer not to turn off emergencies for real emergencies, but don't want to take the risk of it shutting off for an erroneous unknown tc error. I just simply separated each emergency into it's own config.ignore_????_??_?? variable so I could configure each individually, basic tests of using it this way seem to work fine. 

If you come up with a different way of doing this or making it more functional or complex somehow else, I'm prepared to switch over to it though, for now though, that solution seems to work, and accomplishes all that I personally need it to.

jbruce12000, 2022-08-10:
working on this. should get the branch pushed up in the next few days. no ground-breaking implementation... basically just what's been discussed here.

I'll need you to test the branch before it gets merged to master if you can do so.

captaincaden, 2022-08-10:
Sounds good.

I can do that. I've had it implemented the way I outlined on June 18th for a month or two now and it has worked well, for what that information is worth.

jbruce12000, 2022-08-11:
ok, ready to test.

branch name = configurable-emergencies

captaincaden, 2022-08-11:
Other than the line in the run function for shorts to ground and VCC, it's written exactly as I had it (other than one of the names of one of the ignore variables), which is what I've been running now for the past month and it has worked well. 

I've updated it to match the branch, and still seems to function well as well. I've tested it the way I had it prior numerous times the past month, so I imagine that it will work well as it is now as well. 

With it updated to the branch, testing the ignore temp too high = true and forcing the too high temp, also functions as expected, logging, but not stopping the profile.

jbruce12000, 2022-08-11:
good news. I think these new variables being set to False is sane. If folks run into problems with their hardware, they can then can set to True as needed.

captaincaden, 2022-08-11:
I think that makes the most sense as well. 

What types of TC errors add to the variable responsible for too_many_errors?

jbruce12000, 2022-08-11:
noConnection
shortToGround
shortToVCC
unknownError

jbruce12000, 2022-08-11:
merged to master. open new issues if you find any.

closed 2022-08-11 by jbruce12000

captaincaden, 2022-08-11:
> noConnection shortToGround shortToVCC unknownError

That's what I thought. The way I have had it set up to combat my random one line unknown error issue (???:True) is to have unknown errors ignored true, but still keeping the too_many_errors false, so that way if there is a genuine unknown error that causes too many errors, it still stops the firing. I don't know if that makes sense or functions appropriately, but it has prevented numerous unnecessary endings of firings. I haven't had it set the too many errors variable yet, so I like that safety net still there in case there is a genuine unknown error that would cause temp or something like that to be unreadable.