jbruce12000/kiln-controller #98: automated restarts

jbruce12000, 2022-06-18:
https://github.com/captaincaden requested a way to restart a profile that failed at the last known good point.

I think this should be controlled by a setting:

`automatic_restart = True`

You'd need a few things recorded every duty cycle: 

- profile that was running
- time elapsed in that profile
- timestamp of last successful duty cycle

Maybe just write get_state to a json file every duty_cycle.

You'd have to pick a reasonable window for restarts [maybe 10 or 15 minutes], after which it's too late and you just want the software to give up and NOT restart. If it's inside that window, restart the last running profile where it was.

`automatic_restart_window = 15`

captaincaden, 2022-06-18:
Even if it was just automatic to the point of knowing where it currently is in the profile based on where the temp is would be a nice improvement to the restarting issue. I wouldn't mind having to manually tell the kiln to start a profile again after an issue arose, this is how my commercial digital kiln works. I would just like to not have to use the API and figure out where I'd need to be roughly in the profile to restart. Mainly just because if I guess incorrectly, then I'm pretty much 10 degrees (the kiln_must_catch up value) too high or too low the rest of the firing from that point. 

The automatic restart in this scenario would really only be applicable for failures or human intervention in relation to needing to stop and restart the kiln for some reason. A power outage probably isn't something that this would work on (although it'd be really cool if it could) this setup, I'd imagine it'd get complicated in that sense. However I do know my digital kiln does that (if power outage, and power resumes within __ amount of time, the schedule will continue when power is restored) but I'm not sure how it achieves it.

rondoc, 2022-06-18:
If you get a power outage and the pi goes down at same time. When power
comes back on what do expect to happen? Putting the pi on a UPS or simple
battery makes sense to me.
Ronald.

captaincaden, 2022-06-18:
> If you get a power outage and the pi goes down at same time. When power comes back on what do expect to happen? Putting the pi on a UPS or simple battery makes sense to me. Ronald.

That was why I had added the part about it not really being applicable to this. I've thought about putting it on a UPS, but power outages are rare but definitely not unheard of here. I was just adding that part to say it would be a cool feature if it was able to resume after a power outage. 

I don't think commercial digital kilns have on board UPS's or batteries though, and they're able to restart the firing if the power is interrupted without user input. Just not sure how they achieve it exactly.

jbruce12000, 2022-06-19:
My vision of what happens in a power outage:

1. pi stops and kiln stops
2. power comes back
3. pi automatically reboots (which I think needs to be configured)
4. kiln-controller.py automatically starts (which must be configured)
5. kiln-controller.py checks if it was running a profile recently... within automatic_restart_window... if so
6. kiln-controller.py restarts last known profile at point where power went off
7. kiln_controller.py runs as normal based on your settings... which means if kiln_must_catch up=True, it pauses the profile while heating back up. If kiln_must_catch_up=False, the kiln would probably be below the setpoint for a time while it struggles to heat.

jbruce12000, 2022-06-19:
So, if this could be more easily solved by either changing the api so that it uses elapsed time in seconds instead of minutes... or every time the process stops or is reset, log the api statement that COULD be used to restart at that point (you know copy/paste a curl cmd).

I'm all for simplicity.

captaincaden, 2022-06-19:
Addressing #3, for me at least, I don't believe I changed any configurations about auto rebooting, but when power is taken away from the pi and reapplied it automatically reboots, and with start_on_boot active, it also starts kiln-controller upon reboot for me. 

Again, while power outages aren't a common occurrence (for me), this would be a great feature, and a big relief of stress about power outages or any other accidental loss of power to the pi.

jbruce12000, 2022-06-19:
this feature is implemented in a branch named automatic-restart. I am testing it in simulations and it seems ok... but I haven't done any testing with my kiln yet.

jbruce12000, 2022-06-19:
interesting. I can't store state in /tmp. During reboots, the contents of that directory evidently get cleaned up sometimes and the state file disappears. I'll have to warn folks about that in the docs.

captaincaden, 2022-06-19:
Cool! I was going to add the changes over to a virtual machine pi that I have, but I realized I really wouldn't be able to test much since I can't reboot that vm without it wiping it to a fresh operating system, so I'll add it to my actual kiln and see how it works if it's ready to do that? I was looking over it earlier to figure out how I can test it without performing a bunch of hard shutdowns on the pi, but I think just a normal controlled reboot should test relatively well?

jbruce12000, 2022-06-20:
This is not quite ready to test. I'll post here when I'm done with development and ready to have you help.

When the time comes, I've tested by killing the process on the command line to start [which covers a lot of cases], and then

`sudo shutdown -r now
`
for a more realistic power loss test. All of that seems to work OK right now, it's the normal functions [like clicking stop in the web interface] that are broken. working.

jbruce12000, 2022-06-20:
I'm not sure I understand your question @rondoc . The purpose of this feature is to restart where a profile left off if there is a failure for any reason... it could be a power outage like a brown out or a black out. It could be something that causes the kiln-controller.py process to die like a bug or an edge case or someone accidentally killing the process on the command line.

This feature is configurable, so if folks don't want to use it, they don't have to. As far as needing different behavior with automated restarts for different kinds of kilns [pottery/glass/enamel/metal], I'm all ears.

captaincaden, 2022-06-20:
In this case, at least from my understanding of this, use case scenario wouldn’t make a difference. If you’re a glass artist running a glass profile, then it would just resume the profile if it was configured true, and it was interrupted while a profile was in process. This wouldn’t make a difference if you were using it for pottery, glass, enamel, etc. the profile would still be the same. 

And from my understanding, the major kiln manufacturers use the same controllers for glass kilns vs. ceramic kilns, just with different profiles and different kiln hardware.

rondoc, 2022-06-20:
My ? was that most ceramicists just let kiln cool naturally but glass
workers control the cooling.  There is a slight concern of mine that say
kiln reaches 1200F and a power outage occurs and kiln temp drops to 800F
 Then power is restored you would not want full power on to get back to
1200 as this would mean very rapid transition thru the quartz area where
change occurs.
Wouldn't it be best to just pick up the profile at 800F?
Just a thought...
Ronald.

captaincaden, 2022-06-20:
@rondoc I believe that's one of the main purposes for the restart only during ___ time frame. Being that if it ____ amount of time had already passed, do not restart. The other part of it is that the way it currently functions, it would be a challenge as is if you lost power at 1200f to figure out how to resume and then call it to the api if the kiln was at 800 anyways without the automated restart feature.

However I do agree on the point that it would be nice if it was able to instead of continuing where it left off, it instead continued the schedule at the temperature/correlating time that it was currently at. And if it was outside ___ window, don't restart. I think that complicates it more though.

rondoc, 2022-06-21:
I guess the more features this app gets the more likely it will have bugs.
True? Sometimes simple is better.
Thanks
Ronald.

jbruce12000, 2022-06-21:
I cannot argue the point that adding more features makes the code more complex. In this case, I have wrapped all the new code in the setting that controls the feature. if `automatic_restarts =False`, none of the code will run.

I agree with @captaincaden about the `automatic_restart_window` setting. If you know that your kiln could pass through some critical phase during that window, you can make the window smaller... or not use the feature at all.

@captaincaden @rondoc it's certainly possible to do what your asking. I have the last time the state file was written to [last modified time], and now [the current time]. I know exactly how much time has passed since the start of the failure. It would be possible to restart the profile at that point. It makes things marginally more complex [like checking to make sure going that far foward in time doesn't go past the end of the profile].

I think I'm gonna finish the current feature before I add another one though. ^^^ could be a future setting.

jbruce12000, 2022-06-21:
@captaincaden the automatic-restart branch is ready to be tested. I tested

1. starting a profile in the web interface and api
2. stopping a profile using the web interface and api
3. stopping the kiln-controller.py process while a profile is running and restarting [causes restart at same pt]
4. killing the kiln-controller.py process while a profile is running and restarting [causes restart at same pt]
5. starting a profile at X minutes using the api
6. turning the feature off and testing that everything works

let me know if you find any bugs.

captaincaden, 2022-06-21:
I will check back over the code I copied vs. the code that's current. My original issue was caused because when I just had it showing changes made, I was seeing an older version. I remember seeing a line about something being too noisy. 

I really need to take some time to learn how to store all of my changes and be able to just let it update the rest of the code from the command line instead of copy pasting. It just seems like it's easier and allows me to make sure everything goes smoothly doing it manually (minus the parts where I'm copying the wrong things or older versions of things

captaincaden, 2022-06-21:
In regards to clicking stop, but restarts, yes, like I figured, I had missed a part of the code when I copied it over. It works as I'd expect now! Gonna play around with some tests and simulations in a few. Like I said, especially at this current point, if I report any issues relative to this, don't exclude the possibility that it was an error that I caused lol. 

State file too old repeating is still true. When the run is started it stops, and same when it's interrupted (assuming it's because new data is stored there) that repeating message goes away. But, after the file becomes >15(or the time set in config) minutes old, it would return logging repeatedly when idle.

jbruce12000, 2022-06-21:
yeah, I don't want any message printing once/second unless it is informative or actionable. I'll get that fixed.

glad to hear that you had a copy/paste issue where you skipped something. I tested on my end again, and could not replicate, so I was worried.

I'm gonna keep testing and figure out a better way to name the state file so it does not require manual work before use.

I'll keep testing for a while [maybe a week] before merging the changes to master. That merge will require everyone to update config.py, so not a minor change.

jbruce12000, 2022-06-21:
cool. I just committed the latest set of changes which includes...

1. more generalized state file naming in config
2. a sleep so the ovenwatcher can be set properly (so the web client displays the right profile)
3. less verbose logging

captaincaden, 2022-06-21:
Updated to the last set of changes, and seems to be working well still. No more logging of state file too old at idle. I do notice that nothing gets logged when it won't restart the profile. I set the automatic restart time to 1 minute (for testing purposes), and forced a stop, waited the minute, and restarted the process, and as expected, it did not restart the profile, but also didn't log anything relative to it. I don't know that it needs to, or if it's supposed to, but I thought I remembered seeing another part where "state file too old" was written, and was expecting to see a log about it. 

I imagine that the watcher script would probably detect an error during the restarts and notify me, but I added a line in the code to use the smtp notifier that I added to also send me a text/email when a restart has happened. I'm going to test and see what the watcher does during this too just to see.

jbruce12000, 2022-06-22:
what happens if there is no lan connectivity... either wireless or wired?

the kiln-controller.py process would start at power-up. If a profile needs to restarted automatically it would do that. The kiln would be locally controlled using the GPIO on the pi. No web or api clients would be able to connect.

Sorry you disabled wireless and got locked out. bummer.

captaincaden, 2022-06-22:
Interesting, so it wouldn't cause any issues? I'm picturing that it would reboot once power came back on, and restart the process, then maybe 2 minutes into all of that internet would be restored. Once internet restored, nothing else would need to be done? You'd be able to connect to the web/api clients just once the pi was connected again? I may read back through some of my added smtp codes, I remember having an issue where it broke oven.py when it tried to send a notification without it being connected to network yet, but I believe I put in exceptions on everything to just pass the notification and log it if it couldn't log into the smtp server, but I need to double check that. 

No worries. I should have bought the micro usb to usb connector when I bought the pi anyways. When I formatted the SD card I configured the network before it was connected to the pi, so I was able to do everything headless with SSH and remote desktop for this long. 
I was trying to find old usb cables to make one real quick, but didn't have any that would be able to plug a mouse into after, and no stores near me have any micro usb to usb adapters or anything that is cheap enough for me to want to cut apart and solder, so internet to the rescue. Should have an adapted by Friday.

jbruce12000, 2022-06-22:
the pi will run the kiln-controller.py process without issue. It is thread based, and the threads that control the kiln have no networking dependencies. So, once networking is restored [ie interface comes up] then access via api or browser based control will work as normal.

I cannot speak to the smtp changes you made.

captaincaden, 2022-06-22:
Good to know! Once I have the adapter to use a mouse and can get back out of the pi not having network, then I'll do some more tests. But everything I tested prior to that incident worked well in my opinion. 

Did you comment on the not logging anything when the kiln won't restart? When I was looking back through the code I saw another line of 'state file too old' that wasn't commented out, but I didn't look further to the conditions that that one required. But when it was interrupted when running a profile, and I made sure the state file was too old, it didn't restart, but also didn't log anything.

captaincaden, 2022-06-23:
Got my pi back in my controls, lol. While I had it locally connected to a keyboard, I tested the automated restart by killing it at the command line, and then killing internet, and then restarting, then reconnecting internet, and as you said, it ran flawlessly. My added smtp codes also ran as expected. 

Note to self: don't turn off wireless LAN again when your only method of controlling the device is via wireless LAN lol.

A unrelated question about time.sleep() in this complex type program, when you use for example time.sleep(1) in a function, does it make the whole process stop for 1 second? or just that line of the function?

jbruce12000, 2022-06-23:
time.sleep(1) in this specific case causes a single thread to sleep... The reason is because other threads are initializing and creating objects. I must wait on the ovenwatcher thread to be created so it can be set in the oven class. This is a terrible and delicate solution, but it works.

A better way to handle this might be to continually poll to see if the ovenwatcher has been set... and sleep until it gets set.

captaincaden, 2022-06-23:
Good to know, I was just curious. It seems to work well as it is, in all of the scenarios I've tested at least.

I just tested another scenario that I had mentioned earlier I believe, is that when an error occurs that stops the process (with ignore emergencies = false, temp too high, tc errors, etc.) it basically just becomes a fight between the automatic restart and the process shutting down. I tested this by simulating the emergency temp, and once it shuts itself down, the automatic restart restarts it, then emergency shuts it down again, then restarts it, over and over and over until it's manually interrupted. I'd imagine this would be the same in the event of a thermocouple failure. I think this would be an issue. Of course, it would be fixed with ignoring emergencies, but maybe there should be some kind of configuration to tell what emergencies should be automatically restarted after somehow? for example, I'd like it to try to restart if an erroneous thermocouple error happened, but I would not want it to restart if temp was too high. This configuration part also maybe solved by making the emergencies themselves configurable like we'd talked about in the other feature. Or maybe simpler way to resolve this issue would be both having emergencies configurable like I mentioned in the other feature, and  like I mentioned earlier in this feature, just put a limit for the amount of times a restart can be attempted, so that if it actually is a real emergency like a tc disconnect or a temp too high, it only attempts the restart ____ amount of times before it gives up.

jbruce12000, 2022-06-24:
regarding emergencies. this is a bug. if there is an emergency, restarts should not happen. this is easily fixed by writing to the state file just like I do for stop/abort.

I have another feature open for configurable emergencies. still thinking about how to implement that. there are lots of ways, but I want the most flexible with least trouble.

daemon.log encoding... hmmm. I have never seen this happen. Try this to view the file:

`cat /var/log/daemon.log|strings`

`strings` will strip any non-printable characters from the file. You could also output this to a file and open it if needed:

`cat /var/log/daemon.log|strings > /home/pi/anyfile.txt`

Then open anyfile.txt with your favorite editor.

jbruce12000, 2022-06-24:
quick fix for restarts vs emergencies. see if it works ok for you.

captaincaden, 2022-06-24:
That's what I figured. Is there a way to tie that into the configurable emergencies? Like I said, I'd like to try a restart after an unknown thermocouple error, but wouldn't want to try to restart after temp to high or a disconnected tc error. 

When I first ran `cat /var/log/daemon.log|strings` it ran a bunch of lines, and then crashed the pi. 

I rebooted, and opened the daemon.log file in Thonny, and it loaded, but only loaded the file up to a certain point. I re ran `cat /var/log/daemon.log|strings` again, and it completed this time, with logs up to the present time. 

Will check out the update.

captaincaden, 2022-06-24:
Updated, and seems to work as expected. Does not get in a restart loop on a temp emergency.

jbruce12000, 2022-06-24:
with configurable emergencies, my vision is... anything that is configured to be an emergency should stop the profile and it should not restart. everything else [non-emergencies] is just logged.

crash... umm. sounds like the SD card is corrupt to me. I cannot explain the intermittent behavior.

captaincaden, 2022-06-24:
That sounds like a good goal for configurable emergencies. 

> crash... umm. sounds like the SD card is corrupt to me. I cannot explain the intermittent behavior.

I was afraid of that, lol. I pictured an sd card corruption to be more catastrophic, lol. Nothing else seems to be affected besides daemon. At least everything else that I use with any frequency/that would be noticeable.

captaincaden, 2022-06-24:
Well, did a fresh install on a new SD card just to be safe. All appears to be working well, and can see the daemon log normally now. 

I'm curious, I've heard of Raspberry Pi's past issues with corrupting SD cards, and while from my understanding it doesn't happen as much anymore as it used to, for a process like this, would it be overkill to install a backup fully ready to go on a spare SD card? That way in the event it would happen in the middle of a firing or during a power outage in the middle of a firing or something, a simple SD card swap and reboot would get things back running? 

Follow up question, would just saving the .json file of a profile put in the profiles folder bring that profile over to my new sd card? or would I need to re write them again?

jbruce12000, 2022-06-24:
having a back-up sd card seems kinda like overkill to me.

how would swapping out an sd-card affect restarts? well, restarts depend on the state file... without the state file being there, restarts would not happen.

can I just copy json profiles over and have them work? yep. that works fine. they don't have to be created by the web interface. I create them all the time from the command line... mostly because it's faster and easier. also, a new feature request is to create a library where folks can share profiles of all kinds... ceramics, glass, metal, scientific and even subcategories within those. Folks would be able to using anything they want to create those.

captaincaden, 2022-06-24:
I was kind of thinking the same. I always have the manual kiln controls as a major back up. And I bought a second SD card today when I bought the other and installed Raspberri Pi OS on it, so it'd just need kiln-controller installed if I ever needed it. 

Good to know. Everything's copied over now including the profiles I've been using and working as expected. Daemon log is working as well on the new SD. I had seen that feature request but hadn't read through it. That sounds pretty cool and useful.

jbruce12000, 2022-06-26:
I added logging capable of ignoring duplicate messages. Restart log messages now use it so they only get logged the first time.

captaincaden, 2022-06-26:
Will update to it now! It only applies to the restart logs, and not the logs of the whole process? What happens if a restart happens twice before kiln-controller was reset? Or say someone was running the program without shutting it off and ran more than one firing, with more than one instance of the possibility of a restart happening? Would it ignore the logs still if say a restart happened yesterday, and then another one happened today?

jbruce12000, 2022-06-27:
Yeah, dupfree only impacts a couple of messages. To be specific, it will log non-unique messages only once while the process is running. In all the cases I can think of [like you mentioned] a restart is required for the automatic restart to kick in.

jbruce12000, 2022-06-28:
merged to master. this feature is complete. if you find bugs, open a new issue.

closed 2022-06-28 by jbruce12000