jbruce12000/kiln-controller #99: repo for profiles

jbruce12000, 2022-06-18:
- create a repo for profiles
- add a configuration variable for local filesystem directory from which to pull profiles

jbruce12000, 2022-06-21:
the current code slurps all the json files in a directory. this is probably not going to scale, but might be worth testing. reality is that there will be a limit to how far folks want to scroll in a drop down list box.

jbruce12000, 2022-07-01:
getting started on this feature. here's the repo: https://github.com/jbruce12000/kiln-profiles

let me know if you want access.

I'm adding some stuff to the json format for these files that will make it easier to search in the future. I've added fields **description** and **tags**.

jbruce12000, 2022-07-01:
created the kiln-profiles branch to support loading of profiles from the new repo.

jbruce12000, 2022-07-01:
basic loading works with configurable path

solnus, 2024-08-23:
@jbruce12000 I noticed this repo/local files the other day, and was wondering if you'd want to switch to just using a git submodule?