jbruce12000/kiln-controller #100: Firing Program stopping early

Gardy1, 2022-07-18:
Hi, I am very much a newbie when it comes to building and programming but thanks to your excellent instructions I have managed to build a working controller for my wife's pottery kiln.  However I am having an issue that I hope you can provide some guidance on.

I test ran the kiln on a 04 Bisque to 1871 Fahrenheit and everything was working correctly all the way up to around 1600 when the pi displayed that the program was complete and the kiln began cooling.

I then ran a rapid straight line ramp to 1900 Fahrenheit to check the coils (this was an older kiln we bought as a fixer upper).  This time the program ran to full temp then stopped correctly.

I went back and created a new 04 Bisque straight line ramp program and ran it.  Again it reached around 1600 before displaying that the program was complete and starting to cool.

Any suggestions as to what might be causing the early ending of the programs?  Would it normally display an error if the program ended at the wrong time?  Any help gratefully received.

jbruce12000, 2022-07-18:
A profile should not stop for no reason. I need to see the logs so I can understand what's happening.  There is a program you can run to zip up the logs which you can post here.

`cd kiln-controller
./ziplogs`

That'll create a file called **kiln.logs.gz**. Transfer that from your raspberry pi to a desktop and post it here so I can check it out.

jbruce12000, 2022-07-18:
My guess is that your getting errors at higher temps with your thermocouple. You might want to try setting:

`ignore_emergencies = True`

in config.py. This would prevent the kiln-controller.py process from exiting if there are lots of errors from your thermocouple.

jbruce12000, 2022-07-18:
Congrats on getting this far without help. We'll figure this problem out.

Gardy1, 2022-07-18:
[kiln.logs.gz](https://github.com/jbruce12000/kiln-controller/files/9136350/kiln.logs.gz)

Gardy1, 2022-07-19:
I think that should be it.  Please note when it failed the last time I tried to recover the firing by setting a new program starting at 1350 Fahrenheit (as the kiln had cooled to around there) and going back to 1871.  Unfortunately that too failed.  Just in case you see the odd firing program

jbruce12000, 2022-07-19:
2022-07-16 09:11:20
"cmd":"RUN","profile":{"type":"profile","data":[[0,70],[18000,1971]],"name":"Max Temp Test"}}

so 5 hours from 70F to 1971F

finished right on time...

2022-07-16 14:11:57,585 INFO oven: schedule ended, shutting down

2022-07-17 11:45:49,415 INFO kiln-controller: Received (control): {"cmd":"RUN","profile":{"type":"profile","data":[[0,70],[600,90],[36000,1828]],"name":"Med Bisque 06"}}

kiln must catch up configured...
so ten hours to 1828

which ran fine until...

2022-07-17 20:05:41,563 INFO oven: temp=1532.88 and then you thermocouple errors
...

2022-07-17 20:05:43,566 INFO oven: emergency!!! unknown thermocouple error

then, 19 minutes later you tried to create a new profile and attempted to deal w
ith the failure...

2022-07-17 20:27:29,347 INFO kiln-controller: Received (control): {"cmd":"RUN","profile":{"type":"profile","data":[[0,1350],[7200,1828]],"name":"Catch up bisque 1500"}}

so, you could enable automatic restarts.
you also should be using the watchdog.

you weren't happy with progress, so you resaved and restarted again...

2022-07-17 20:30:26,935 INFO kiln-controller: Received (control): {"cmd":"RUN","profile":{"type":"profile","data":[[0,1350],[10800,1828]],"name":"Catch up bisque 1500"}}

you lost connection to the thermocouple an hour and 11 mins later...

2022-07-17 21:41:32,903 INFO oven: emergency!!! lost connection to thermocouple

and I think you gave up here...

jbruce12000, 2022-07-19:
ok, so a couple of things. You definitely want 

ignore_emergencies = True 

in config.py for now. I'm making changes to this code in the near future so you can configure exactly what you want to ignore.

There is an api so you can restart any profile at any specific minute. This way you don't need to create a profile like you did.

see https://github.com/jbruce12000/kiln-controller/blob/master/docs/api.md

There is a watcher that can watch your kiln process so if it has a problem it pages you. This way you wont be 20 minutes late reacting to a problem. 

see https://github.com/jbruce12000/kiln-controller/blob/master/docs/watcher.md

I just added automatic restarts... which might work in this case, not sure... but you can test that if you like. It was designed to handle power outages, but might work for this.

The good news is, it looks like your PID tuning is good.

jbruce12000, 2022-07-19:
btw, the **lost connection to thermocouple** error seems bad. I've never had that happen during a firing. make sure your connections to the max-31855 or max-31856 are good and the connections to the thermocouple are good.

jbruce12000, 2022-07-19:
[final.pdf](https://github.com/jbruce12000/kiln-controller/files/9137006/final.pdf)

Graphs look good for the test fire. There is some oscillation at low temps. That's pretty normal. Error is increasing at high temps, but it's still within acceptable limits.

jbruce12000, 2022-07-19:
[2022-07-08.cone5-long-bisque.pdf](https://github.com/jbruce12000/kiln-controller/files/9141029/2022-07-08.cone5-long-bisque.pdf)
This is the graph from July 8th. It stopped at 1583F with a thermocouple error, but it is revealing. There are stable oscillations for the first two hours and then the amplitude begins to reduce. To fix this, you need less P and more D. When I make PID changes, I change one value by doubling or halving. test. repeat. You can certainly get more stability at low temps. This will likely improve the error at high temps as well. It is oscillating at high temps too, but the period is really short, so you cant see it on the graphs.

Gardy1, 2022-07-20:
Thanks, it could be my first time soldering job wasn't up to scratch on the thermocouple.  I'll take a proper look at the weekend and also make the changes you have suggested.  
Thanks again for the insights and fast responses, it's much appreciated

captaincaden, 2022-07-23:
@jbruce12000 If I remember from my testing correctly, automated restarts wouldn't have helped here. It would be the same outcome as when I tested it with emergency temp too high, since they use the same abort_run function. I've had multiple firings saved by configuring it to ignore just unknown TC errors over the past few weeks, I look forward to seeing what you come up with for individual configuration of those ignore functions compared to what I did, and I think it will be a great addition for everyone else too once you finish it.

jbruce12000, 2022-08-15:
@Gardy1 The latest master branch now includes configurable emergencies so that you can ignore the errors that are causing the kiln-controller to stop. See the very bottom of config.py.

Please try this out and open a new issue if you need.

closed 2022-08-15 by jbruce12000

rondoc, 2022-10-11:
My guess is that the thermocouple is shorting to ground ( earth) . The max
chip will not work if that oc curs. Then the pi and app think the tc is
bad. Using a VOM check the resistance from one of the tc leads to ground it
must be many methods.
Ronald.