jbruce12000/kiln-controller #101: A question about startup-on-boot/graceful shutdown

maabaum, 2022-07-20:
I don't know if this is really an Issue so much as something I don't understand. Prior to doing some more firings I was updating my installation to the latest kiln-controller code, and I tried to enable start-on-boot. (Mostly I've just fired the Raspberry Pi up and then ssh'd in to start things.)

Anyway, rebooting the Pi via basically switching it off and turning it on again, the supposedly enabled kiln-controller service fails thusly:

```
* kiln-controller.service - kiln-controller
     Loaded: loaded (/etc/systemd/system/kiln-controller.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-07-20 16:24:59 EDT; 25min ago
    Process: 401 ExecStart=/home/pi/kiln-controller/venv/bin/python /home/pi/kiln-controller/kiln-controller.py (code=exited, status=1/FAILURE)
   Main PID: 401 (code=exited, status=1/FAILURE)
        CPU: 1.659s

Jul 20 16:24:59 raspberrypi python[401]:   File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/server.py", line 192, in get_list>
Jul 20 16:24:59 raspberrypi python[401]:     return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)
Jul 20 16:24:59 raspberrypi python[401]:   File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/server.py", line 288, in _tcp_lis>
Jul 20 16:24:59 raspberrypi python[401]:     sock.bind(address)
Jul 20 16:24:59 raspberrypi python[401]:   File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/_socketcommon.py", line 563, in b>
Jul 20 16:24:59 raspberrypi python[401]:     return self._sock.bind(address)
Jul 20 16:24:59 raspberrypi python[401]: OSError: [Errno 99] Cannot assign requested address: ('192.168.1.11', 8081)
Jul 20 16:24:59 raspberrypi systemd[1]: kiln-controller.service: Main process exited, code=exited, status=1/FAILURE
Jul 20 16:24:59 raspberrypi systemd[1]: kiln-controller.service: Failed with result 'exit-code'.
Jul 20 16:24:59 raspberrypi systemd[1]: kiln-controller.service: Consumed 1.659s CPU time.
```
Before you ask, yes, that's the correct network address. Anyway this kind of socket bind error goes waaaaay past my understanding of network issues, so I thought I'd ask what it all means?

Note that if I then ssh into the pi and start the service up manually:
> sudo systemctl start kiln-controller.service

it starts up without complaint and seems to run fine.

This leads to another question: there seems to be no graceful way to shut down the kiln-controller process once it's running. I can turn the power off, or I can shell in and kill the service, or find the running process and kill it or whatever, but considering that it's got this little web server running, and logging processes and so on, is that a bad thing? Should we have some sort of a close-things-down-gracefully command maybe through the API?

jbruce12000, 2022-07-21:
graceful shutdown

The kiln-controller.py process is managed by systemd if you configured it to run at start-up. You should use [systemd commands](https://www.digitalocean.com/community/tutorials/systemd-essentials-working-with-services-units-and-the-journal) to control the process.

api shutdown

I think, if you have a need to shut down the process, you're probably troubleshooting a problem. You'll need to login to the pi anyway to check out the logs, diagnose the issue, and decide what to do. Since you're already logged in, you should use systemd commands to manage the process.

could not bind (no active network)

Have you tried **Wait for Network on Boot** on the pi?

`sudo raspi-config`

then select **Wait for Network on Boot** and **Slow Wait for network connection before completing boot**. click finish. reboot.

maabaum, 2022-07-21:
> You should use systemd commands to control the process.

I think that's what I'm doing. systemctl is a utility for managing systemd.

> Have you tried Wait for Network on Boot on the pi?

Ah. I didn't know that even existed. I bet that will do it, thanks.

jbruce12000, 2022-07-21:
yeah, I saw the systemctl cmds after I made my post. Did not edit, I guess for posterity.

close this if **Wait for Network on Boot** fixes your problem.

maabaum, 2022-07-22:
Well, this is not working as well as I expected. I used raspi-config to set "Select wait for network connection on boot". I presume that had effect though I haven't yet figured out where that setting is set, if you follow me, so that I can check it. Anyway, the same thing happens. I get that same failure message as above from systemd when I reboot the pi. Again, if I use systemctl to start the kiln service manually after the pi is up and running, no problem.

Acting on a hint from stack exchange, I added lines to kiln service script:

>[Unit]
Description=kiln-controller
Requires=network.target
After=network.target

to see if that added emphasis to the wait-until-the-network-exists strategy. No change in outcome.

Here's a thing. The kiln-controller.py script **does** get called and starts running during the boot sequence. It starts up the log process and starts writing log messages. It gets as far as :

>2022-07-21 17:20:59,814 INFO kiln-controller: listening on 192.168.1.11:8081

and then nothing. It looks as if things are choking on the line (in def main):

>server = WSGIServer((ip, port), app, handler_class=WebSocketHandler)

jbruce12000, 2022-07-23:
in config.py, what is listening_ip set to? I have a feeling you changed the default. change it back to:

`listening_ip = "0.0.0.0"`

and test again.

jbruce12000, 2022-07-23:
In the context of servers, 0.0.0.0 means all IPv4 addresses on the local machine. If a host has two IP addresses, 192.168.1.1 and 10.1.2.1, and a server running on the host listens on 0.0.0.0, it will be reachable at both of those IPs.

This means that any network this is currently connected to the pi or if one connects in the future, you'll be able to reach the server.

If this works, I think I'll modify config.py and remove the ip, but leave the port configurable.

maabaum, 2022-07-24:
Yes, you're right, that does seem to fix it.

I didn't understand that at, the meaning of the listening address. I still don't entirely follow it, but will eventually, I guess. Anyway, thanks, yet again!!!

closed 2022-07-24 by maabaum