jbruce12000/kiln-controller #105: MAX31855 Reads 32

MurrayPottery, 2022-09-12:
Continued from #49 post closing notes>
Temp only reads 32 

jbruce12000 suggested starting a profile and asked to open a new issue. 

@jbruce12000 are you talking about starting a firing profile?  If so, I have done that and nothing changes. Still reads 32 and time clock doesn't start

MurrayPottery, 2022-09-13:
This is the log with 200-250 test log profile started.  I let it run for several seconds and stopped it.  Temp still read 32F.  

thomasware@raspberrypi:~/kiln-controller $ ./kiln-controller.py
2022-09-12 22:42:04,271 INFO kiln-controller: Starting kiln controller
2022-09-12 22:42:04,280 INFO kiln-controller: this is a real kiln
2022-09-12 22:42:04,281 INFO oven: import MAX31855 
2022-09-12 22:42:04,281 INFO oven: init MAX31855
/home/thomasware/kiln-controller/lib/max31855.py:33: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.cs_pin, GPIO.OUT)
/home/thomasware/kiln-controller/lib/max31855.py:34: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.clock_pin, GPIO.OUT)
2022-09-12 22:42:04,313 INFO oven.dupfree: automatic restart not possible. state file does not exist or is too old.
2022-09-12 22:42:04,315 INFO oven: ovenwatcher set in oven class
2022-09-12 22:42:04,318 INFO kiln-controller: listening on 0.0.0.0:8081
2022-09-12 22:42:20,981 INFO geventwebsocket.handler: 127.0.0.1 - - [2022-09-12 22:42:20] "GET /picoreflow/index.html HTTP/1.1" 304 66 0.068728
{'type': 'backlog', 'profile': None, 'log': []}
{"type": "backlog", "profile": null, "log": []}
2022-09-12 22:42:21,234 INFO kiln-controller: websocket (status) opened
2022-09-12 22:42:21,278 INFO kiln-controller: websocket (control) opened
2022-09-12 22:42:21,313 INFO kiln-controller: websocket (config) opened
2022-09-12 22:42:21,338 INFO kiln-controller: websocket (storage) opened
2022-09-12 22:42:21,767 INFO kiln-controller: GET command received
2022-09-12 22:42:38,477 INFO kiln-controller: Received (control): {"cmd":"RUN","profile":{"data":[[0,100],[480,200],[2000,200],[2300,250],[3600,250]],"type":"profile","name":"test-200-250"}}
2022-09-12 22:42:38,478 INFO kiln-controller: RUN command received
2022-09-12 22:42:38,479 INFO oven: Running schedule test-200-250 starting at 0 minutes
2022-09-12 22:42:38,479 INFO oven: Starting
2022-09-12 22:42:39,371 INFO oven: kiln must catch up, too hot, shifting schedule
2022-09-12 22:42:39,372 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:41,375 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:41,378 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:41,379 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:43,382 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:43,384 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:43,385 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:45,387 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:45,391 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:45,392 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:47,396 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:47,398 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:47,399 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:49,402 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:49,404 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:49,405 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:51,409 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:51,412 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:51,413 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:53,416 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:53,418 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:53,419 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:55,422 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:55,425 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:55,426 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:42:57,429 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.00, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:42:57,432 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:42:57,433 INFO oven: kiln outside pid control window, max heating
[... cut ...]
2022-09-12 22:43:11,473 INFO oven: temp=32.00, target=100.00, error=68.00, pid=0.00, p=1700.01, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=3600, time_left=3599
2022-09-12 22:43:11,477 INFO oven: kiln must catch up, too cold, shifting schedule
2022-09-12 22:43:11,477 INFO oven: kiln outside pid control window, max heating
2022-09-12 22:43:11,790 INFO kiln-controller: Received (control): {"cmd":"STOP"}
2022-09-12 22:43:11,791 INFO kiln-controller: Stop command received
2022-09-12 22:43:13,481 INFO oven.dupfree: automatic restart not possible. state = IDLE

MurrayPottery, 2022-09-14:
Temp reads normal on a standalone test app from MAX 31855
Rewired according to the MAX31855 instructions and temps read normally.
What issues am I going to run into with bypassing the resistors according to the schematics?

jbruce12000, 2022-09-14:
What issues am I going to run into with bypassing the resistors according to the schematics?

D0 is an input. You need to check the max current supplied by the MAX-31855 to make sure it does not overload the input on the PI.
CS and CLK are outpus from the PI to the MAX-31855. The same is true here. Check the max current supplied by the PI and make sure it does not overload the inputs on the MAX-31855.

I think, before removing the resistors, you should verify config.py settings related to pinouts:
```
### Thermocouple Connection (using bitbang interfaces)
gpio_sensor_cs = 27
gpio_sensor_clock = 22
gpio_sensor_data = 17
```

These are BCM pin numbers. If you don't know that that means, read this:

https://github.com/jbruce12000/kiln-controller/blob/master/docs/troubleshooting.md

rondoc, 2022-09-14:
I use the 31855 board and it works. Just make sure that your thermocouple
is not grounded. The tc input to the 31855 must be floating . That means no
connection to ground!!
Also make config is set up for the tc correctly K type tc is most common.
Ron.

MurrayPottery, 2022-09-14:
Will double check all and get back as soon as I can

jbruce12000, 2022-10-01:
please update if you're still having issues. close if not.

MurrayPottery, 2022-10-03:
Thanks for the reminder. I still need to run the kiln for setting the numbers, but everything else is working ok

jbruce12000, 2022-10-03:
ok, sounds like the problem is solved. if you have problems tuning the PID numbers, post and let us know. there are lots of people here that can help.

closed 2022-10-03 by jbruce12000