jbruce12000/kiln-controller #106: Problems Error Oven:  Problem reading temp

tjerram, 2022-09-30:
I just installed everything and had no errors during installation,  when I run the program (not a simulation) I get the following error:
oven problem: problem reading temp N/C:True Gnd:false VCC:false ???:false

Any Ideas?

This looks like GREAT software to run a kiln for glass. I hope we can it it working It sure is a lot easier than a PID controller.

Thanks for helping

rondoc, 2022-09-30:
Check connections. Thermocouple must be isolated from ground.

jbruce12000, 2022-09-30:
I need a few things to help troubleshoot...

1. config.py
2. logs so I can see what you did
3. type of thermocouple and card (max-31855 or max-31856)

btw, this is a PID controller. It has a web interface which makes things easier in some ways, but it's still very complex. We can help you overcome hurdles though.

tjerram, 2022-09-30:
[Kiln.zip](https://github.com/jbruce12000/kiln-controller/files/9686895/Kiln.zip)
I got it to run by doing gpio readall I noticed that all the pins except 22 and 27 were in (they we labeled out) so I ran gpio mode 22 in and gpio mode 27 in now it runs but it says that the temp is always 32f.  I am using the Max-31855 and an old ceramic type k thermistor.  could I have a bad thermistor?

I knew this was a pid controller but it has a MUCH better interface than the packaged ones out there.

Thanks for your help.

tjerram, 2022-09-30:
new issue:
ModuleNotFoundError: No module named 'bottle'

tonymoon61, 2022-10-01:
I'm currently battling similar issues after working great for well over a year. I'm getting it working again slowly, but you'll want to run the following on a command line

pip3 install -r requirements.txt

Then, try to run it using python3 specifically, as follows:

python3 kiln-controller.py

jbruce12000, 2022-10-01:
sounds like you're making progress. You checked the gpio pins are configured them correctly.

now you're running into a library install problem. are you using a virtualenv like the instructions?

My hope is that the installation instructions are explicit and that they work. If this is not the case, I need to know where things broke so I can fix them [or you can submit a pr and I'll try it].

tjerram, 2022-10-01:
yes I issued the command

virtualenv -p python3 venv

before running   python3 kiln-controller.py

I get

Traceback (most recent call last):
  File "/home/pi/kiln-controller/kiln-controller.py", line 8, in <module>
    import bottle
ModuleNotFoundError: No module named 'bottle'

*Tom Jerram*

***@***.***
(607) 235-0316

jbruce12000, 2022-10-02:
can you post the output of this:

```
source venv/bin/activate
pip freeze

```

tjerram, 2022-10-02:
I got no output just ran for a minute and returns command prompt (venv)
***@***.***:~ $

*Tom Jerram*

***@***.***
(607) 235-0316

jbruce12000, 2022-10-02:
pip freeze lists all the libraries that are installed in a virtual environment. The output should be similar to this:

```
bottle==0.12.19
colorzero==2.0
gevent==21.1.2
gevent-websocket==0.10.1
gpiozero==1.6.2
greenlet==1.1.0
pkg_resources==0.0.0
RPi.GPIO==0.7.0
zope.event==4.5.0
zope.interface==5.4.0

```

Your test shows the libraries are NOT installed in the virtual environment. Start with a clean slate

In the directory where you created your virtual environment:

```rm -rf ./venv```

then try to recreate a new virtual environment:

```
virtualenv -p python3 venv
source venv/bin/activate
export CFLAGS=-fcommon
pip3 install --upgrade setuptools
pip3 install greenlet bottle gevent gevent-websocket
pip3 install -r requirements.txt
```

and finally, check what got installed:

```pip freeze```

and post it. if you get nothing again, I'm at a loss and I'm open to suggestions from anyone following this thread.

rondoc, 2022-10-02:
I had issues similar and decided best course was to re install complete on
new mem chip. It's just not worthwhile chasing odd failures that could be
hardware or software issues when in less than hour you can start afresh.
Also when it's all working Ok make a duplicate and pi makes that very
simple.
Just my opinion...
Ron

tjerram, 2022-10-03:
I reimaged and followed the original instructions and that didn't do it so
I added the instruction you sent above and that seemed to work

Then I ran into this:

2022-10-02 21:32:55,367 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:32:55,375 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:32:57,383 INFO oven: temp=74.17, target=79.63, error=5.46,
pid=0.00, p=136.50, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00,
run_time=22, total_time=48780, time_left=48757
2022-10-02 21:32:57,394 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:32:57,401 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:32:59,411 INFO oven: temp=74.31, target=79.63, error=5.32,
pid=0.00, p=133.06, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00,
run_time=22, total_time=48780, time_left=48757
2022-10-02 21:32:59,422 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:32:59,429 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:33:01,435 INFO oven: temp=74.31, target=79.63, error=5.32,
pid=0.00, p=133.06, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00,
run_time=22, total_time=48780, time_left=48757
2022-10-02 21:33:01,442 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:33:01,448 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:33:03,454 INFO oven: temp=74.31, target=79.63, error=5.32,
pid=0.00, p=133.06, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00,
run_time=22, total_time=48780, time_left=48757
2022-10-02 21:33:03,468 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:33:03,475 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:33:05,481 INFO oven: temp=74.31, target=79.63, error=5.32,
pid=0.00, p=133.06, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00,
run_time=22, total_time=48780, time_left=48757
2022-10-02 21:33:05,491 INFO oven: kiln must catch up, too cold, shifting
schedule
2022-10-02 21:33:05,503 INFO oven: kiln outside pid control window, max
heating
2022-10-02 21:33:05,662 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False
2022-10-02 21:33:05,717 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False
2022-10-02 21:33:05,773 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False
2022-10-02 21:33:05,829 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False
2022-10-02 21:33:05,884 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False
2022-10-02 21:33:05,939 ERROR oven: Problem reading temp N/C:True GND:False
VCC:False ???:False

Looks like it could be a bad thermistor or a bad Max 31855  I am going to
order replacements and see if that's the problem.

I'll let you know how that works out.  Any other thoughts?

*Tom Jerram*

***@***.***
(607) 235-0316

rondoc, 2022-10-03:
Why is your PID setup like this? 0,136,0?

rondoc, 2022-10-03:
Sorry I meant 136,0,0 for your PID. I am guessing that this computes to a
full heating . Did you try 20,20,20 ?
I finally used the method suggested by a test heating  and get great
tracking.
Ronald.

jbruce12000, 2022-10-03:
The only time I get this error is when my thermocouple is disconnected from my MAX-31855:

`ERROR oven: Problem reading temp N/C:True GND:False VCC:False ???:False`

Check that connection.

jbruce12000, 2022-10-03:
I just saw that 136,0,0 config. @rondoc what that would do is aggressively attempt to reach the set point. If temp is slightly above set point, no heating. If it is slightly below set point, full heating. It would basically be a very sensitive on-off control.

What this usually causes is a big overshoot, and with no active cooling, a long time to get back below the set pt. rinse. repeat.

Although this is not a good pid config, he needs to deal with the connectivity errors first.

tjerram, 2022-10-04:
I am not sure where to adjust those numbers.  I don't see them in
config.py.  I have ordered a new Max 32855 and a new Type K thermistor.  I
will try again when they arrive and let you know if that was the issue.

One last question, what pins do I attach the ssr to.

Thanks for all your help so far

*Tom Jerram*

***@***.***
(607) 235-0316

rondoc, 2022-10-04:
It doesn't look like you understand the connection to the SSR. I am an EE
and would be happy to help with this. If it's not really ok on this site
you can email
Me direct at ***@***.***
Ronald.

nemisis2005, 2022-10-04:
total newbie please how do i get to run out of simulation and change to centigrade i having problems. all set up correctly i think.
please some one. thanks nem

tjerram, 2022-10-04:
open config.py and go to line 84 change

simulate = True

to

simulate = False

*Tom Jerram*

***@***.***
(607) 235-0316

nemisis2005, 2022-10-05:
thanks for that but my biggest issue is finding the config py file. i checked the sd card for the file but is not there please help.

nemisis2005, 2022-10-05:
i am using rasp pi 4. running through putty .the file you mention /home/pi/kiln-controller how to access this desktop environment on the pi? 
i entered command in putty environment doesn't come up not in directory.
sorry to be a pain but once i find the file to make the changes I think I'll be good. tried in cd kiln-controller aswell.
thanks
nem

nemisis2005, 2022-10-05:
i think i just answered my own question found it tom i am checking now.
thanks
bud

nemisis2005, 2022-10-05:
all working now thanks again for the help easy when you know how.
all in centigrade just got to box it now and tune it.
thanks for the effort you guys have put in .
nem

tjerram, 2022-10-11:
I assume you are accessing your pi from. Windows computer.

Open a command prompt.
ssh ***@***.*** pi's ip
 nano /home/pi/kiln-controller/config.pi

Change the appropriate line (I sent you before)

Ctrl x
Y
Reboot

That should work.

Tom Jerram
(607) 235-0316

jbruce12000, 2022-10-11:
@tjerram what pins do I attach the SSR to? If you follow the schematic, then one side is connected to 5V and the other side is connected to the collector of a transistor [BC-547]. This schematic uses the transistor as a switch. The base of the transistor is connected to a GPIO pin set as an output and configured in config.py as gpio_heat [pin 23 by default]. The emitter of the transistor is connected to ground.

using a transistor like this means that you can supply plenty of current to the SSR to get it to switch.

rondoc, 2022-10-11:
If what you write is correct then there is a piece missing you must have a
resistor, from 1 to 5 kohms, between the base of the transistor and the
control  output pin [23]. This limits the base current to a reasonable
figure . I don't have schematic so it might already show this.
Ronald

nemisis2005, 2022-10-12:
you are correct 1 k resistor between pin 23

nemisis2005, 2022-10-12:
and the schematic is on the on the page just for info.

closed 2022-10-14 by tjerram