jbruce12000/kiln-controller #107: max cooling outside PID range

tjerram, 2022-10-14:
I think I finally have this all set up properly but I am getting strange behavior that is probably related to a setting somewhere.  I don't remember changing or even seeing the settings that are probably wrong.  Here is what I am seeing:

2022-10-13 22:55:39,265 INFO oven: temp=74.98, target=80.00, error=5.02, pid=0.00, p=125.48, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=420, time_left=419
2022-10-13 22:55:39,278 INFO oven: kiln must catch up, too cold, shifting schedule
2022-10-13 22:55:39,285 INFO oven: kiln outside pid control window, max heating
2022-10-13 22:55:41,292 INFO oven: temp=74.98, target=80.00, error=5.02, pid=0.00, p=125.48, i=0.00, d=0.00, heat_on=2.00, heat_off=0.00, run_time=0, total_time=420, time_left=419
2022-10-13 22:55:41,312 INFO oven: kiln must catch up, too cold, shifting schedule
2022-10-13 22:55:41,316 INFO oven: kiln outside pid control window, max heating

This is just running on a table (No Kiln only heat is from me holding the thermocouple)

tjerram, 2022-10-14:
Sorry forgot to add the config.py
[config.zip](https://github.com/jbruce12000/kiln-controller/files/9782128/config.zip)

jbruce12000, 2022-10-14:
In config.py you have

pid_control_window = -5 #degrees

set to a negative number. That is untested and is probably going to cause all kinds of unexpected behavior. I would set this to 10F or 5C to start. This means that PID control will only happen when the kiln is pretty close to the set point. Inside the window is the only time that Integral accumulation happens. So, since you are using F, set it to this:

pid_control_window = 10

I looked through the other settings are everything else looks ok. Use

thermocouple_offset= -3

only if you need to. this should be set to zero unless you have tested your tc and it is off.

tjerram, 2022-10-14:
Thanks so much it now works.

I noticed several other programs in the directory.  But I don't see
instructions. I assume kiln-turner would tune the PID how do I run it (is
it as simple as python kiln-tunner.py)  I am also wondering about
kiln-logger and kiln-watcher.

This is great I look forward to using it.

*Tom Jerram*

***@***.***
(607) 235-0316

jbruce12000, 2022-10-14:
see https://github.com/jbruce12000/kiln-controller/blob/master/docs/pid_tuning.md
and https://github.com/jbruce12000/kiln-controller/blob/master/docs/watcher.md

kiln-logger.py is a web client that grabs data and logs it to a csv file for later analysis. I also have a project that just takes the raw logs and graphs them...https://github.com/jbruce12000/kiln-stats

please close this and the other issue if they are solved.

closed 2022-10-14 by tjerram