jbruce12000/kiln-controller #108: start on boot

nemisis2005, 2022-10-15:
hi question on how to get the start on boot function working where do i enable it or do i  need to move some code files.
thanks in advance
nem

captaincaden, 2022-10-15:
Open a terminal, and run:
`$ /home/pi/kiln-controller/start-on-boot`

nemisis2005, 2022-10-16:
hi if i run in terminal this is what it says bash: /home/pi/kiln-controller/start-on-boot: No such file or directory.
where to go from here.

captaincaden, 2022-10-16:
Open your file manager in /home/pi/kiln-controller (the same folder that config.py is in and kiln-controller.py, etc.) and see if there's a file named start-on-boot. If you followed the installation instructions/used git to this project, then the start-on-boot file should be in that location unless it was moved, or deleted. 

The start on boot file is just a shell script file that runs a few commands to set it up. I'd imagine you could just run those commands on their own, but using the shell script file is easier, and I'm not for certain if running those lines individually would work appropriately, so I'd wait for @jbruce12000 to comment on doing it that way. 

If the file is missing for some reason from /home/pi/kiln-controller then you could theoretically just make a new shell script file and copy the contents of the start-on-boot file in this repository and paste them into the shell script file.

jbruce12000, 2022-10-16:
sounds like you did not install the kiln controller software in /home/pi. If this is the case, you'll need to change start-on-boot to match the path you installed to.

/home/pi is the default user and home directory created by Raspberry PI OS. you should use this default unless you have some linux expertise.

jbruce12000, 2022-10-16:
to find the path you installed to...

login to the pi and then check to see if the software is installed in that home directory...

`ls -la`

and look for **kiln-controller** directory. If it is there, type

`pwd`

to give you the present working directory. replace /home/pi in start-on-boot with the directory output from the pwd command.

jbruce12000, 2022-10-27:
@nemisis2005 please update this issue, or close if the problem is solved.

jbruce12000, 2022-11-15:
I'm assuming the problem is resolved @nemisis2005. open a new issue if you're still having trouble.

closed 2022-11-15 by jbruce12000