jbruce12000/kiln-controller #111: Could not determine platform

BoojumHH, 2022-10-28:
Hello!
A great project! I'm trying to implement it on my end, but always fail with the error message below.
I use a MAX31856 and a RasPi 4 model B.
Apparently the Adafruit_GPIO library seems to be out of date. Unfortunately, I can't get anywhere on my own at this point.
Can someone please help me with this problem?
Thank you very much,
Chris

```

 pip freeze
Adafruit-GPIO==1.0.3
Adafruit-MAX31855==1.6.1
Adafruit-PureIO==1.1.9
bottle==0.12.23
gevent==22.10.1
gevent-websocket==0.10.1
greenlet==1.1.3.post0
RPi.GPIO==0.7.1
spidev==3.5
websocket-client==1.4.1
wiringpi==2.60.1
zope.event==4.5.0
zope.interface==5.5.0

2022-10-28 18:01:13,085 INFO kiln-controller: Starting kiln controller
2022-10-28 18:01:13,088 INFO kiln-controller: this is a real kiln
2022-10-28 18:01:13,089 INFO oven: import MAX31856 
2022-10-28 18:01:13,089 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 43, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 95, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 140, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
  File "/home/pi/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln-controller/venv/lib/python3.9/site-packages/Adafruit_GPIO/GPIO.py", line 426, in get_platform_gpio
    raise RuntimeError('Could not determine platform.')
RuntimeError: Could not determine platform.

```

jbruce12000, 2022-10-28:
This library has been deprecated. See if this works:

`vi /home/pi/kiln-controller/venv/lib/python3.9/site-packages/Adafruit_GPIO/GPIO.py
`
replace line 415 which says

`Adafruit_GPIO.get_platform_gpio()
`
with

`plat = Platform.RASPBERRY_PI`

and save the file with :wq! or shift zz

jbruce12000, 2022-10-28:
The real issue here is deprecation. The solution above might be a temporary fix, but long term, I have to make a decision about blinka. In the past, I refused to switch over to the new blinka library because it did not support generic linux. This meant that folks could not run simulations on linux to test drive the application. I need to check if generic linux support has been added.

jbruce12000, 2022-10-28:
this issue says generic linux support was added... I'll have to test that https://github.com/adafruit/Adafruit_Blinka/issues/60

BoojumHH, 2022-10-28:
Hello Jason,
Thanks for your quick and helpful reply.
The error with the platform is now fixed. Unfortunately now I get a different error message .
```
2022-10-28 19:56:44,111 INFO kiln-controller: Starting kiln controller
2022-10-28 19:56:44,115 INFO kiln-controller: this is a real kiln
2022-10-28 19:56:44,115 INFO oven: import MAX31856 
2022-10-28 19:56:44,115 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 43, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 95, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 140, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
  file "/home/pi/kiln-controller/lib/max31856.py", line 140, in __init__
    self.cr1 = (((self.avgsel & 7) << 4) + (self.tc_type & 0x0f))
TypeError: unsupported operand type(s) for &: 'str' and 'int'.
```
Probably because I changed the entry in config.py to thermocouple_type = "MAX31856.MAX31856_S_TYPE". 
With thermocouple_type = MAX31856.MAX31856_S_TYPE I got the error 'Could not import config file.'
After researching the source code I now have changed to thermocouple_type = 0x6.
After the following messages my pi is now not responding and a reboot did not help either :(
I guess I will have to reinstall it.

```
2022-10-28 20:09:27,871 INFO kiln-controller: Starting kiln controller
2022-10-28 20:09:27,874 INFO kiln-controller: this is a real kiln
2022-10-28 20:09:27,874 INFO oven: import MAX31856 
2022-10-28 20:09:27,875 INFO oven: init MAX31856
client_loop: send disconnect: Broken pipe
```

jbruce12000, 2022-10-28:
at the top of config.py uncomment this line...

`#from lib.max31856 import MAX31856`

by removing the # in front. next uncomment:

`#thermocouple_type = MAX31856.MAX31856_S_TYPE`

and let me know if that fixes it.

BoojumHH, 2022-10-28:
Oh, this is embarrassing...
Everything seems to be working now. 
Now only the thermocoupler is missing. I will attach the circuit to the oven tomorrow and test it.
Thank you very much for your help!
Chris

jbruce12000, 2022-10-28:
no worries. close this issue. I'll explore replacing current HW libs with blinka.

open a new issue if you have trouble getting things running, or if you have trouble tuning the pid values.

jbruce12000, 2022-10-29:
ok, closing for you.

closed 2022-10-29 by jbruce12000