jbruce12000/kiln-controller #112: blinka support

jbruce12000, 2022-10-28:
Switch from current GPIO libraries to blinka.

That's the underlying issue for https://github.com/jbruce12000/kiln-controller/issues/111 and others.

References:

- https://learn.adafruit.com/adafruits-raspberry-pi-lesson-13-power-control/software

rondoc, 2022-10-28:
I am a software novice so please answer these. Once I have a working kiln
controller running do I need to keep making updates?
Does internet connectivity make any difference to this question?
Thank you.
Ron.

jbruce12000, 2022-10-30:
@rondoc **Do I need to keep doing updates?** 

nope. if the kiln-controller.py is working for you at a specific version, you can pin to that version and never update.  Every commit has it's own unique signature and you can checkout code using that signature. This means that you can effectively never update. if you have bugs or problems with a specific version, you'd have to tell me when you file the issue. check out the output of `git log` in the kiln-controller directory if you want to see the commit history. See those long hex strings that look like this?

`commit 3d8333a16bf553bde9acf537b8e564524f857db6`

with any one of those, you can do a git checkout like:

`git checkout 3d8333a16bf553bde9acf537b8e564524f857db6
`
then you are pinned to that version.

**Does internet connectivity make any difference to this question?**

hmm. if you want to update directly from the git repo to your raspberry pi, you must be connected to the internet. If you want to pin to a specific version, you don't need to be connected to the internet.

jbruce12000, 2022-10-30:
Adding blinka could mean a flood of questions about specific hardware I cannot answer because I have no experience with them. Looks like fifty boards are supported https://circuitpython.org/blinka

jbruce12000, 2022-10-31:
wow, it's actually 72 boards that have SPI support. I'm gonna need some help testing the adafruit-31856. Any volunteers? I'll help as much as I can, but I don't have one of these.

afreeth, 2022-11-01:
I could try and help but perhaps a more software orientated person might be
more use and I don't often get time to mess about with kilns.
I wonder perhaps if the community would be better off sending you a MAX
31856.

I have a spare one here but I am located in the UK.

Where are you based?

Cheers
Andrew

jbruce12000, 2022-11-01:
Thanks @afreeth. I appreciate the offer. I think it would be cheaper for me to get one straight from adafruit.

jbruce12000, 2022-11-01:
ok, I have everything working but thermocouple exceptions. These will take a little time because the underlying adafruit libraries are inconsistent.

jbruce12000, 2022-11-02:
I have documentation changes to make, but I'm gonna need help testing this. Does not matter if you're using an adafruit-31855 or adafruit-31856. Does not matter what rapsberry pi version you're using.

phdbrewer, 2022-11-02:
I could assist on testing the adafruit. I am setting up a kiln and intend to use a max chipset. The max has been delayed, so I could order an adafruit, assemble,and test. I understand the code and would be happy to contribute.William

jbruce12000, 2022-11-03:
@phdbrewer cool. post here when your hardware is ready. I should have all the docs completed by then.

jbruce12000, 2022-11-04:
documentation is complete except for the replacement schematic. I'll get that done soon. Tons of changes. Lots of new docs. Modified the kiln-tuner. Added some docs about scheduling kiln runs which I think will benefit lots of folks:

https://github.com/jbruce12000/kiln-controller/blob/blinka/docs/schedule.md

Something everyone needs to understand that is when I commit this change to master, it will be a **breaking change**. Everyone will likely need to move around a couple of wires to connect to raspberry-pi SPI pins. So, you can pin at an old version and do nothing, or use the new version, change some wiring, and test.

None of the PID code is changing, No functionality is going away... but the underlying 31855 and 31856 libraries are going away and being replaced by blinka - which adafruit owns and maintains.

jbruce12000, 2022-11-16:
Testing this branch is blocked on me. I have to rebuild the high voltage/high current wiring in my personal controller.

jbruce12000, 2022-11-28:
Finished rebuilding my controller and started testing yesterday. Tested autotuner. Completed a test profile on my kiln.

jbruce12000, 2022-12-04:
I've done lots of testing on this branch. The autotuner works well and it required only slight adjustment for my preferences on my kiln [about half as much integral]. Temp control and error handling seem bug-free. New settings work well. test-output, test-thermocouple, and gpioreadall all worked well during initial setup. docs changes look ok.

I'll test with a long bisque fire this week.

jbruce12000, 2022-12-22:
more testing today. ran into a bug last night with config variable naming. ran my profile 90% and then died. bummer.... but fixed the bug.

chipgarner, 2022-12-24:
@jbruce12000 Can you explain the naming bug? I would like to see if I can create a test that will catch this and other similar bugs.

jbruce12000, 2022-12-24:
setting in the code was changed from 

config.ignore_too_many_tc_errors

to

config.ignore_tc_too_many_errors

see for yourself with...

git diff 90d10eb125a4ab0b3862616e10b278f0765d5f57 abaf155e5a34fb1259c0aa4adb5b197a8c5d8a13

Around 1600F my thermocouple started getting lots of errors [which is normal] and since the variable was named incorrectly, my setting in config.py was ignored and the code ended my schedule because it though there was an emergency. These errors are caused by plasma in the kiln making it my tc appear to have a short circuit.

chipgarner, 2022-12-24:
I use Pycharm and it flags errors like this, making them much easier to find! There is a free version. There are other python code checkers. I will go through the blinka branch and check for "cannot find reference" issues.

I need to learn about plasma. So far I have left it to the fusion physicists.

jbruce12000, 2022-12-24:
I think doing lint checking is great. I think we should strive to:

1. have automated tests that 
2. can be executed via github actions

I'm no testing expert, but there are lots of testing libraries out there for python. Let me ask one of my developers at work and we can get started on that after the holiday season.

chipgarner, 2022-12-24:
I checked all the .py files in blinka and did not find any issues.

I use pytest, it is pretty great and is widely used. I wrote a couple of unit tests for the additions I made for the seek function.

chipgarner, 2023-01-01:
@jbruce12000 Have you tested the MAX31856 yet? I may have found an issue. 

I don't have a kiln controls connected but I do have a Pi and both a MAX31855 and MAX31856 with thermocouples, so I can do some testing with this.

jbruce12000, 2023-01-01:
@chipgarner I don't own a MAX31856. testing blinka with that would be great. I might switch to the 31856 soon.

chipgarner, 2023-01-02:
@jbruce12000 Found some issues with the '56 code. I made some minor changes and have it running. I will clean it up and do a pull request in a day or two.

jbruce12000, 2023-01-02:
@chipgarner sounds good

rondoc, 2023-01-03:
I wonder if you are really getting errors from plasma. For this to occur
need a pretty close pair of "electrodes" at 240 bolts. Most TCS are
enclosed in a ceramic sleeve so plasma cannot occur. Can you send me a pix
of where this occurs? Much more likely to be the tc wires shorting together
and if so put beads on the the wires.
Ronald

jbruce12000, 2023-01-03:
It's temperature related, 1600F and above only. The thermocouple leads are isolated and cannot physically contact each other.

jbruce12000, 2023-01-03:
Well, plasma doesn't exist at temps this low... so it's something else. Maybe thermal expansion is causing contact near the tip of the TC.

rondoc, 2023-01-03:
Have you checked the tc with an electronic thermometer at high temps? It
could be that the welded end of the tc is poor. I use a Fluke thermometer
in parallel with the pi system and so far very close readings.

chipgarner, 2023-01-05:
I am experimenting with both a MAX31856 and MAX31855 and 2 thermocouples. The '56 seems to be a lot better. I am getting lots of lots of short circuit to ground and short circuit to power errors with the '55:
[image]
 It is also always much noisier, here is a graph of both that shows the relative jumpiness at 5 second readings.
[image]

My thermocouple wires are ridiculously thin, much less than 24 guage. I have some fatter ones on order.

chipgarner, 2023-01-05:
My MAX31855 works similarly badly with both the old and new Adafruit libraries. The above is using the new CircuitPython library.

rondoc, 2023-01-05:
I used both chips...from Adafruit and they worked ok. No problems. The 55
makes sense as we use the K type TCS. The tc wire in the kiln should be as
heavy as possible. Once it's out of the heat you can change to thin double
tc lead wire.
R.

chipgarner, 2023-01-05:
Now the noise went away on my MAX31855. I am still seeing grounding errors but not as many. The jumps in the figure are about 0.25C, which is the resolution of the A to D converter.
[image]

chipgarner, 2023-01-06:
I replaced the cheap and very thin thermocouple wires with better 24 gauge thermocouple wires. They are yellow and red, instead of yellow and blue, which is reassuring for type K thermocouple wires. I am getting much better results, the noises is low and I am not seeing "short circuit to power" and "short circuit to ground" errors I was getting with the cheap wire. Zero errors after over an hour, this was during cool down from about 1600f to 1400f. I was getting several errors per minute before changing the wires.

@jbruce12000 I don't think it is normal to be getting lot of thermocouple errors, regardless of the temperature. We should try to find the source if this is happening. I will continue to test.

rondoc, 2023-01-06:
Re tc stuff.
1. Check 55 or 56 output with scope. Is it noisy?
2. If noisy check tc . If in kiln is the kiln power a phase controller ? If
so filter power leads.
3. Use a lithium battery to power the 55 module.
4. Where is 55module located. The tc output and hence amplification in the
module is large so easy to pick up noise.
5. If above not successful put a tc right next to the module and heat with
heat gun.
Is noise still problem?
Finding noise and eliminating it is a logic ladder and if you cannot
eliminate the problem then check on power supplies as they are often noisy.
Change to batteries. Also you can filter the output of the 55 module with a
couple of 1k resistors and a 0.1 uF capacitor.

chipgarner, 2023-01-06:
@jbruce12000 PR 128 is up

chipgarner, 2023-01-06:
The noise and error problems look like they are gone after replacing the thermocouple wires. I will test again with the other kiln.

chipgarner, 2023-01-10:
@jbruce12000 Are you up for a Zoom meeting? Topics are thermocouple noise, Blinka testing, code architecture and goals. My time is pretty flexible, so name a day and time and I can set up a Zoom. Need an email address.

jbruce12000, 2023-01-11:
Thermocouple noise - I think this is a non-issue. We have code that allows users to exclude specific errors or all thermocouple errors. It's up to folks to reduce noise via shielding/twisting etc. Maybe @rondoc could create a thermocouple noise page that we can add to the docs?

Blinka testing - Both the max-31855 and max-31856 have now been tested. I'm happy with the changes and I'm ready to move the blinka branch to master. **THIS WILL BE A BREAKING CHANGE FOR EVERYONE.** It will require them to re-wire connections from their TC chips to the RPI.

Code Architecture - open to suggestions here, but I like simplicity. I also like inheritance. We need better test coverage.

Goals - stable, dependable, configurable, accurate control of kilns & ovens. The blinka branch breaks the stability rule, but removes some internal libraries and could expand the user base [by using hardware other than rpi].

chipgarner, 2023-01-12:
Testing and architecture are very much related. Adding tests is difficult now because of how the code is structured. Improving the architecture and adding testing will make it much easier to add features in the future. I will open another issue and make some suggestions.

jbruce12000, 2023-01-12:
Cool. I'd much rather make iterative change than radical.

chipgarner, 2023-01-14:
My MAX31856 quit at at little over 1200f. The call to wait_for_one_shot got stuck and the temperature went to zero. It works fine after the kiln cooled down. I plugged the thermocouple directly into a Fluke thermocouple module and it worked fine through the rest of the firing, no errors and good temperatures. I tried it with two different programs on the Pi, so I think the problem is the MAX31856 itself or the Blinka software. This could be only my system but is a bit worrisome if I am the only one who has tested the MAX318356 with the Blinka library at high temperatures.

jbruce12000, 2023-01-15:
hmm... seems bad. I'll put off merging until we can definitively call that addressed. I'm sure if we find a bug, the blinka team would be happy to get a PR.

there is no one else that has tested the MAX-31856 on the blinka branch that I know of.

chipgarner, 2023-01-17:
The MAX-31856 does hang up in "one-shot" in the Adafruit library. I am trying to test a mod to the library to exit after a while, but of course can't get it to stick now. I have no idea what causes it to hang up but am hoping it will throw and error when it skips out of the while loop. I'll post when I learn something.

The '56 uses a blocking call when it gets temperatures and it takes about 165 ms to return the temp, assuming averaging 1 is selected. (It takes around 670 ms if averaging is set to 16.)  This will effect the temperature averaging. I don't think it will really matter to the outcome but the averaging over 40 measurements will be over about 6.6 seconds instead of the expected 2 seconds.

jbruce12000, 2023-01-18:
ugh. At 165ms, it seems like the default should be 10 instead of 40. 2000ms/165ms = 12.12

jbruce12000, 2023-01-18:
I really wonder what is happening during that 165ms blocking call.

jbruce12000, 2023-01-19:
At the very least, it means that the config default of 40 reads over 2s is too high and that it should be changed to 10 reads over 2s.

jbruce12000, 2023-01-21:
ok, committed that sane default of 5 samples per second... which is 10 over 2s by default.

chipgarner, 2023-01-22:
I have done more testing thanks to a thermocouple emulator from @rondoc. This is a variable power supply that mimics the voltage from a thermocouple, simulating about 460 to 2500f. I have not been able to reproduce the error I had during firing, when the temperature stopped updating because the thread hung waiting forever for the one shot temperature to be computed.

I have been able to reproduce the error, very unreliably, by introducing a thread collision that does not happen in the kiln-controller code. I have not seen any errors at all when testing the latest Blinka branch with the tc emulator. I have run it for several hours and gone through the temperature range multiple times.

I think the error I saw was a one off, and may have something to do with my setup during the firing that is not happening with my testing. @jbruce12000 I don't think you should hold up the release any longer on account of this.

If the error ever does occur, the only symptom is that the temperature stops changing on the UI page. If anyone sees this I recommend shutting off power to the board for about one minute before turning it back on.

Again, this happened once during a firing as the kiln went past about 1200f. I tried restarting the Pi. I may have been hasty or there may have been something in my setup that was still a problem. The next day, after cooling off, it worked fine.

I am leaving for three months in Taiwan and won't be able to test this in a real kiln until my return in May. I will be able to code and test with the emulator.

jbruce12000, 2023-01-22:
understood. thx for your testing effort. enjoy Taiwan.

If anyone else out there has a max-31856 and is interested in testing, lmk. if not, I'll purchase one and test it myself.

rondoc, 2023-01-22:
Anyone want to make the very simple tc emulator? I can post the schematic .
Ronald

phdbrewer, 2023-01-22:
Hello All,I have purchased all of the supplies to build this controller with an Adifruit max chip. I don't have it completed yet so I am unable to test. Send me the schematics for the emulator and I might be able to help with that also.I am building a kiln comtroller that allows me to burn wax out of moulds and then immediately fill with molten metal. I have been following the threads and documentation. I would like a start using this and hopefully will have it finished this week. My health has improved and I would like to contribute.William

rondoc, 2023-01-23:
This is the emulator. It delivers a voltage that mimics the tc. Range with
values shown is 500 to 2400F. Using a battery makes it a floating source
similar to a live thermocople.

captaincaden, 2023-02-02:
@jbruce12000 I've been using 31856 on raspberry pi zero 2W since I started this project. Was looking at updating to this branch, mostly because I want the feature that we discussed a month ago that was merged into the blinka branch. I'm unsure what all will need to change though with switching to this branch, as I see you mentioned something about needing to change the high voltage wiring. I'd be happy to do some testing with 31856, if I knew what exactly I'd need to do to update to this branch.

jbruce12000, 2023-02-03:
@captaincaden no high voltage wiring changes are needed. The only thing that needs to change are the GPIO connections from the PI to the max-31856:

```
spi_sclk  = board.D11 #spi clock
spi_mosi  = board.D10 #spi Microcomputer Out Serial In (not connected) 
spi_miso  = board.D9  #spi Microcomputer In Serial Out
spi_cs    = board.D5  #spi Chip Select
```

jbruce12000, 2023-02-03:
Also, see the **Testing** section in that branch. It has a few scripts for testing connections once you make the wiring changes:

```
test-thermocouple.py
test-output.py
gpioreadall.py

```

jbruce12000, 2023-02-03:
@captaincaden also, I love that you're using a Zero. I always wondered how well it would run. Now I know... just fine. Did you get the one with headers? or install your own? or solder directly to the board?

chipgarner, 2023-02-04:
@jbruce12000 It looks like CircuitPython is capable of using any GPIO pins for the thermocouples via software SPI.  It is a little more low level than using the MAX31855 and '56 libraries but does not look difficult. See:
https://learn.adafruit.com/circuitpython-basics-i2c-and-spi/spi-devices
Scroll down to Software SPI & I2C.

This is could be extremely useful for lot of people who don't want to switch pins but want the new features.

It is slower than using hardware SPI. This will work fine for the thermocouples but it seems like good practice to use hardware SPI when the pins are connected for it.

I have both thermocouple boards and the emulator, so I am well equipped to work on this.

jbruce12000, 2023-02-07:
@chipgarner I love the fact that folks wont need to change hardware. I wonder what the max read rate will be? With HW SPI, the current read rate peaks at about 5/s. I'll need to test this and suggest a sane default for config.temperature_average_samples.

chipgarner, 2023-02-14:
It looks like this is why the '56 doesn't work: https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO/issues/20

From what I can tell by searching the MAX31856 works with software SPI on Arduinos. It would be possible to make a C module and call it from Python. It would need to be compiled for various boards.

@jbruce12000 What do you think of a fallback that uses the old library for the MAX31856 using software pins? It wouldn't work for all the boards, but this is mainly for people who already have something working and soldered up. New boards can use the hardware pins.

rondoc, 2023-02-14:
> adafruit/Adafruit_CircuitPython_BitbangIO#20

Why use the 56? The 55 works fine.

[3 more replies]

closed 2024-07-02 by jbruce12000