jbruce12000/kiln-controller #113: How do I get it out of simulation mode

derrabe1980, 2022-11-10:
When I go into config.py and change simulation for True to False or try to comment it out I can not get it to start up.  If I set it to True again it runs the web console ok but wont actually send power to kiln

jbruce12000, 2022-11-10:
what are you running on? A raspberry pi? using an adafruit-31855 or 31856?

set **simulate = False** in config.py and try to start things up and post the logs so I can see what's happening. if you get an error even starting the process, I want to see the output you get.

From the symptoms you're describing, you probably have some libraries that are not installed. if you run

`python config.py`

it should show you the specific library that is not installed in the virtual environment.

derrabe1980, 2022-11-10:
I am sorry for not being more specific.   I am running a raspberry pi 4
with a Max31855 chip

When I set it to simulate = False, save and reboot.  I get the following
error when I try to manually launch the software

2022-11-10 14:05:20,928 INFO kiln-controller: Starting kiln controller
2022-11-10 14:05:20,931 INFO kiln-controller: this is a real kiln
2022-11-10 14:05:20,931 INFO oven: import MAX31855
2022-11-10 14:05:20,931 INFO oven: init MAX31855
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 43, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 95, in
create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 127, in __init__
    from max31855 import MAX31855, MAX31855Error
  File "/home/pi/kiln-controller/lib/max31855.py", line 2, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'

running python config.py <enter>  just takes me to the next command prompt

derrabe1980, 2022-11-10:
Well I got it to run, I install raspberry pi lite not full os and I think that was my issue as I do not have this issue when running the standard raspberry pi build.  But now I got another weird issue.  No matter what program I run it will say run completed after about 2-3 minutes of running

jbruce12000, 2022-11-11:
ok, sounds like you re-installed the os and got all the libraries installed. Nice work!

Please post the logs here so I can see the problem you're having. Run this script...

`./ziplogs`

in the kiln-controller directory and it will produce a file named...

`./kiln.logs.gz`

Post that here.

derrabe1980, 2022-11-11:
I shut the pi off last night after I was having the issues.   So this was off of a clean boot.   If it does not show what you need let me know and I will run a firing process on it and then create a new log for you

[kiln.logs.gz](https://github.com/jbruce12000/kiln-controller/files/9990476/kiln.logs.gz)

jbruce12000, 2022-11-11:
That file contains no logs.  So, you're probably just running the kiln-controller.py process on the command line yourself. This is fine, just capture the output with something like this...

`source venv/bin/activate; ./kiln-controller.py 2>&1 |tee out.log`

and then post out.log here.

derrabe1980, 2022-11-11:
This is what I got

2022-11-11 16:19:54,090 INFO kiln-controller: Starting kiln controller
2022-11-11 16:19:54,092 INFO kiln-controller: this is a real kiln
2022-11-11 16:19:54,092 INFO oven: import MAX31855
2022-11-11 16:19:54,093 INFO oven: init MAX31855
/home/pi/kiln-controller/lib/max31855.py:33: RuntimeWarning: This channel
is already in use, continuing anyway.  Use GPIO.setwarnings(False) to
disable warnings.
  GPIO.setup(self.cs_pin, GPIO.OUT)
/home/pi/kiln-controller/lib/max31855.py:34: RuntimeWarning: This channel
is already in use, continuing anyway.  Use GPIO.setwarnings(False) to
disable warnings.
  GPIO.setup(self.clock_pin, GPIO.OUT)
2022-11-11 16:19:54,099 INFO oven.dupfree: automatic restart not possible.
state = IDLE
2022-11-11 16:19:54,099 INFO oven: ovenwatcher set in oven class
2022-11-11 16:19:54,101 INFO kiln-controller: listening on 0.0.0.0:8081
Traceback (most recent call last):
  File "/home/pi/kiln-controller/./kiln-controller.py", line 298, in
<module>
    main()
  File "/home/pi/kiln-controller/./kiln-controller.py", line 294, in main
    server.serve_forever()
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/baseserver.py",
line 398, in serve_forever
    self.start()
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/baseserver.py",
line 336, in start
    self.init_socket()
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/pywsgi.py",
line 1545, in init_socket
    StreamServer.init_socket(self)
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/server.py",
line 180, in init_socket
    self.socket = self.get_listener(self.address, self.backlog, self.family)
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/server.py",
line 192, in get_listener
    return _tcp_listener(address, backlog=backlog,
reuse_addr=cls.reuse_addr, family=family)
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/server.py",
line 288, in _tcp_listener
    sock.bind(address)
  File
"/home/pi/kiln-controller/venv/lib/python3.9/site-packages/gevent/_socketcommon.py",
line 563, in bind
    return self._sock.bind(address)
OSError: [Errno 98] Address already in use: ('0.0.0.0', 8081)

jbruce12000, 2022-11-12:
ok. this says you have a process that is already using port 8081 on the pi. You need to find out what it is. My guess is, that you followed the instructions, and that you ran the script start_on_boot. start_on_boot adds kiln-controller.py as a service. the service is configured to automatically start when you reboot. you can check for this in a few ways. the easiest is just to look for the process to see if it is running:

`ps -ef |grep kiln`

you could also just use a browser to see what is running on port 8081.

jbruce12000, 2022-11-12:
it is possible to change the port if you need to [see config.py], but you probably don't need to.

derrabe1980, 2022-11-12:
Thank you, Yes I did run the Automatic start on reboot script on my Pi.
And maybe my current issue is misrepresented by me and I apologize if it
is.  I will try to re-word it below.

The Rpi4 is set up to autostart the kiln-controller script on boot.

I can get to http:\\localpage:8081 web page from a browser on my network
and see the current thermocouple temp readings, set up programs, and even
start the kiln by running a program.

The issue I am having is that no matter what schedule program I run it only
runs for about 90-120 seconds and then just gives me a program complete
message in the browser window and it then stops actively heating the kiln
and I can the start it again but usually only get 30-45 seconds the next go
around.

Example: So lets say I have a 5 hour program and the first step is to go to
300 degrees.   I hit the start button, the kiln starts heating up, the
thermocouple checks regularly and adjusts the temperature readout as it is
supposed to.   About 2-3 minutes after hitting the start button a green box
pops up and says something like schedule complete and tells the kiln to
turn off its heating element and it acts like it has gone through the whole
scheduled program.

I hope that clears up what is going on.

jbruce12000, 2022-11-12:
I've never heard a problem like this. It's super strange. please post config.py so I can see your configuration... and most importantly, I need to see the logs of a run so I can troubleshoot the problem. These logs should be located in /var/log/daemon.log.

I have seen some strange problems happen when folks change time zones with a schedule running...

jbruce12000, 2022-11-12:
I also want to see the schedule "program" you're running. These are located in kiln-controller/storage/profiles/. Each schedule is a separate json file. post the ones you're testing with here.  you can just drag-drop the files if you like.

derrabe1980, 2022-11-12:
[kilncontrol-derrabe.zip](https://github.com/jbruce12000/kiln-controller/files/9995678/kilncontrol-derrabe.zip)
Here are the three files you requested the Log file was about 38MB so it said it was too big unless I zipped them all together

jbruce12000, 2022-11-12:
change these in config.py:

```
ignore_lost_connection_tc = True
ignore_unknown_tc_error = True
ignore_too_many_tc_errors = True
ignore_tc_short_errors = True

```
You have a lot of thermocouple errors. This is likely a hardware issue. Could be the thermocouple itself, connections etc. The settings above will allow you to at least test things out and tune the PID for your kiln.

Your Test schedule is ok, but it's better if you have a more gradual slope when heating so that you don't way overshoot. You're asking the kiln to immediately heat to 300.

Your thermocouple temp is -6 or -7 degrees at start. Is this valid? is your kiln outside or something?

derrabe1980, 2022-11-12:
Well it is in my garage (not well insulated and it is currently snowing outsite. I am going to try with the suggested settings and report back.  Also how do I take it out of autostart mode?

jbruce12000, 2022-11-12:
there's no need to...

`sudo service kiln-controller restart`

if you really want to though:

`sudo service kiln-controller stop`

will temporarily stop it so you can do whatever you want on the command line. When you reboot it will re-start.

derrabe1980, 2022-11-12:
I changed the settings as you suggested.  I also made another test schedule
with less drastic changes and so far it has been running for about 10
minutes and following the graph pretty closely.   I can hear the kiln
turning on and off as it needs to adjust. I am running just a 30 minute
schedule for the concept test.   I do have a question.   Is there a way to
have it hold a temp until I tell it to stop manually?  I do lost wax
casting and the schedule I usually use is something like below.  All are in
C

170 - 2hr
370 - 1hr
570 - 1hr
770 - 4hr
500 - ramp down 1hr then hold until turned off

jbruce12000, 2022-11-12:
The best way to do this is to make a schedule that at the end, holds the temp for a few days [or whatever], then you just turn it off. You could turn it off a few different ways:

1. physically turn off power
2. use the api
3. use the web interface stop button

jbruce12000, 2022-11-12:
if you post the logs of the successful run, I can make graphs using https://github.com/jbruce12000/kiln-stats and make tuning suggestions.

derrabe1980, 2022-11-12:
Sure I can do that but I forgot to add some temp hold times it was just ups and downs no hold at.   I have to leave in about 20 minutes for the evening.   I will have more time to setup a better schedule and run it tomorrow and I will post the logs.   I wanted to say THANK YOU for all your help with this.

derrabe1980, 2022-11-13:
I am thinking my thermostat has reached the end of its life and I purchased a new one it should be here tomorrow.  The current one has an offset of 65 degree C and I think it is time to replace with this and the errors you were seeing on the logs

jbruce12000, 2022-11-15:
I'm gonna close this. We worked through a few issues and solved them. If you need help with your new thermocouple, tuning, or whatever else... open a new issue.

closed 2022-11-15 by jbruce12000