jbruce12000/kiln-controller #114: new ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False

tjerram, 2022-11-13:
any Ideas?  This error comes and goes during a kiln run.  The kiln appears to be running but this error keeps popping up.  Should I just ignore it or do I have a problem with the max 385

rondoc, 2022-11-13:
I suspect that your thermocouple is not electrically isolated from the
ground. Both of the tc boards need complete isolation. If either of the
leads has a slight connection to ground they will fail and then the kiln
controller will report a false. When the connection is correct the kiln
controller will be Ok again. Intermittent problems like this can be tricky
but making sure your Tc wiring is insulated from ground is essential.

tjerram, 2022-11-13:
Thanks for your help.  I'll check for possible grounding of the Tc wires.

By the way this kiln controller is GREAT.  I appreciate all your help and the work that you put in developing it.

jbruce12000, 2022-11-13:
if this is stopping your schedules because you have too many errors, there are a few settings you can enable in order to ignore them. by default 40 readings are taken from the thermocouple over the default 2s duty cycle... so 20 per second. that's a lot and if you have over 30% failures, the controller quits by default.

eygobar, 2022-11-21:
Might not be the same issue, but I saw this a lot when running kiln-tuner.py. My own fault, as the kiln-controller.py automatically starts up when the pi boots. Hence you now have 2 processes polling the max3185x. Point is check you don't have a process (that polls the masx3185x) running more then once.

jbruce12000, 2022-11-21:
@eygobar [you know this already], but you need to stop the kiln-controller.py process before running the tuner.

`sudo service kiln-controller stop`

and once you're done, you can reboot or

`sudo service kiln-controller start`

I'll make this clear in the tuning docs in the blinka branch.

jbruce12000, 2022-12-01:
@tjerram is this still an issue?

tjerram, 2022-12-01:
No everything works great Except I am not generating any logs.. My latest
problem was the thermocouple.

*Tom Jerram*

***@***.***
(607) 235-0316

jbruce12000, 2022-12-02:
If you're automatically starting the kiln-controller process, the logs should go to

`/var/log/daemon.log`

You can open a shell, and run the following to see it in real-time...

`tail -f /var/log/daemon.log`

jbruce12000, 2022-12-02:
The log is sent to stdout of the kiln-controller process, so if you don't want it to start automatically on boot, you can still save the log like so...

`source /home/pi/kiln-controller/venv/bin/activate; /home/pi/kiln-controller/kiln-controller.py 1>/home/pi/kiln-controller/log 2>&1`

that would write the log to /home/pi/kiln-controller/log

rondoc, 2022-12-03:
Thanks a good piece of info for me to remember.

jbruce12000, 2022-12-16:
I'm assuming you have now tested the command above and have access to the logs. if you run into any other problems, open a new issue.

closed 2022-12-16 by jbruce12000