jbruce12000/kiln-controller #115: Time scale

rondoc, 2022-11-14:
Can we get actual time or at least a time scale for hours for the x axis on the charting view. I get strange time intervals like every 33 minutes. 
Thank you.
Ronald.

jbruce12000, 2022-11-14:
I think what you want is whole number hours or minutes or seconds displayed, best fit. So if you have an 8 hour schedule, you want to see 9 intervals on the screen with numbers at the bottom like 0,1,2,3,4 for hours... if you configure minutes, you want 0,60,120 etc... same for seconds.

Is that correct?

jbruce12000, 2022-11-14:
Note to myself. Just changing tickformatter wont do it. gotta examine the jquery flot lib... hopefully for a setting to suggest how wide each tick should be ie 3600 for an hour, 60 for a minute etc.

rondoc, 2022-11-14:
Yes,  you got it right. It's not a biggie but would be nice since
everything else about the charting is soo good.
Ron.

jbruce12000, 2022-11-14:
more notes to myself... looks like jquery flot ticks or ticksize will fix it. see https://github.com/flot/flot/blob/master/API.md

jbruce12000, 2022-11-15:
@rondoc fixed. pull latest master.

**config.time_scale_profile** controls the display and you probably want that set to...

`time_scale_profile  = "h"`

unless your profile is really short.

jbruce12000, 2022-11-15:
[image]

closed 2022-11-15 by jbruce12000

rondoc, 2022-11-15:
Thank you. How do I access this file? I am not a coder.
Ronald

jbruce12000, 2022-11-15:
log in to the pi and...

```cd kiln-controller
git pull
sudo service kiln-controller restart
```

then go check out the changes in a web browser.