jbruce12000/kiln-controller #116: Max31855 may be the culperit

derrabe1980, 2022-11-16:
I would have emailed you directly on this but after looking about I think the errors I was getting from my thermocouple is probably not a bad thermocouple to begin with.  The new one is doing the same thing.   After doing some research I think the issues is the MAX31855 breakout board I bought.   I just used amazon and typed that in and it came up with an assortment of options so I decided to by the 5 pack of them in case I messed one or two up as my soldering skills are not great.  Noob when it comes to some of this stuff.   Well I went back and looked at the one you posted vs the ones I bought that say they are the same thing and they are not.  

Original Purchase
https://www.amazon.com/dp/B09MJX7P99?psc=1&ref=ppx_yo2ov_dt_b_product_details

One you posted and now I purchased:
https://www.amazon.com/gp/product/B00SK8NDAI/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

As you can tell the "knock offs" are missing several other components on the board....Ugh  probably what was throwing my errors and such.   Figured I would share so others dont make the same mistake.  I will update this when the new board comes tomorrow

jbruce12000, 2022-11-16:
Interesting. Make sure and let us know the results. I can change the docs to say something about not using knock-offs if this ends up fixing the problem.

derrabe1980, 2022-11-16:
Will do, do you mind if I post the log tomorrow or thursday after I swap it
out to see if you still see the thermocouple errors that I was getting
prior causing the program to end early on error

jbruce12000, 2022-11-16:
go for it.

derrabe1980, 2022-11-18:
Well I am reporting back and running my first real test right now.   The new board has corrected 2 of the 3 problems. I am getting the correct temp out of it without having to do a big offset.   And the temperature is holding steady not fluctuating as much as 8-10 degrees sometimes.  So the new board (official board) is definitely an improvement.    I am still having error with the thermocouple still and I have tried all day yesterday to try to isolated it but nothing has totally work.  I am getting the errors like #114 I have went down to just the pi and the thermocouple and still getting them.  But using the end on error false  lines mentioned in my other issue it seems to be working fine.

derrabe1980, 2022-11-18:
I want to add follow up I just ran a 3 hour schedule going up to 770c at steps and it ran with out issue and a near straight light following the path presetup.

closed 2022-11-21 by jbruce12000