jbruce12000/kiln-controller #117: NOT an issue just a thought

tjerram, 2022-11-17:
I don't know how to contact you other than here so here is a thought.

I have a kiln with 2 separate electric elements.  Would it be possible to control them independently in the same PID controller.  I realize it would require a separate SDR but it seams to me that other than full heat it would be far easier to control 1/2 of the heat than all of it cycling quickly. perhaps not engaging the second element until the first element has been on full for some period of time. 

Just a thought.  I love the program the way it is the user interface is great and it controls very well but I can't help but tinker

rondoc, 2022-11-17:
this might be off topic but I'm just trying to help here...I think you have
missed the point of the control system. If you have both elements connected
the controller only uses a % that is needed to keep the PID function
capability to match your profile, the one you set up.
Using one element makes no sense if you are using the kiln controller. If
you are manually controlling the heat then using two elements would allow
for series and or parallel connection to adjust heat input ( this is an
aside and nothing to do with the kiln controller, BTW I have only one
element a silicon carbide ( Globar type) one and the kiln-controller does a
great job via my SSR. On the app you can see the % power that the
controller sets. If you are using mechanical relays you should change to
SSR as they can switch on and off many times a second without any problems.
The mech relays will not survive this. SSR relays are abundant and cheap.
just make sure you have a fan/heat sink  to cool them and also pick the
voltage and current to be about 2 x expected.

tjerram, 2022-11-17:
I am using an SSR (and the two elements connected in parallel) and everything works great. I was just wondering if I could get finer low load control.  The thought comes from sizing steam control valves where you size them as 1/3 first valve and 2/3 second valve so that you avoid frequently operating at near closed for low load.  I understand that it is a different situation because the on /off response of an electric element is more linear that of the valve.  As I said the current situation works fine but I am a tinkerer by nature and can't help but try to tweak the last little bit of performance out of everything.

jbruce12000, 2022-11-17:
Interesting idea... I see a few problems with it:

1. Uneven element wear - over time, one element would get used more than others causing it to wear at a faster rate
2. PID environment - changing the number of elements changes the speed with which a kiln can heat. This upsets the balance of the PID config. You would need two separate PID configs to really make this work... one for a single element and one for all elements. For instance, the **Proportional** would probably need to increase in the case the single element. I think it could work ok without doing this, but you'd be compromising one or both modes.
3. Not sure it's going to achieve your goal of better temperature control. Whether you are slicing time on for two elements or slicing time on for a single element, you'd still be introducing the same amount of heat into the system.

Note that the way the system is coded right now, while the temperature is outside the control window, it acts as an on-off controller [100% on to all elements or 100% off]. PID control only happens inside the control window. By default, this window is 5 degrees [f or c] below & above target temp.

K-Type Thermocouples only have a precision of 1/2 degree C. I do some averaging to increase this precision, but don't expect miracles. I take 40 readings/duty cycle. I chop off the top ten and bottom ten readings and average what's left. Using a single element is not likely to give you better tracking to your profile.

jbruce12000, 2022-11-17:
If the goal is "finer, low load control" you might:

1. tweak the PID settings to have less Proportional or more Integral [which means a lower integral number]. Doing these things would decrease overshoot... but also decrease reactivity to change.
2. You could increase the size of the pid control window so that the system doesn't heat at 100% until it gets within 5 degrees of the set point. Instead, PID control would take over earlier to prevent overshoot. The larger you make the window, the more you must pay attention to configuration of Integral because it plays a bigger role.
3. You could change your profile so that it has less abrupt changes. For instance, have it start at or near the normal ambient temp [65F for my basement]... and then slowly heat to 200F over five minutes etc. etc. etc.

rondoc, 2022-11-17:
Jason.

I wonder if the kiln-controller could have a window that sets a fixed power
% e.g. 10% for some time in hours before going to the PID control.
This would be useful for starting from a cold kiln and also for candling
overnight prior to a bisque. It is what I do manually now. I think trying
to set up a PID specially for low temp startup is very tough as the power
output at full will cause big overshoot also it will flash heat the pots ,
not a good idea when they are just past leather hard.
Your suggestion to change the profile for cold weather starts would,
IMNSHO, still have the overshoot problem.
Ronald
Thanks again for your help.

jbruce12000, 2022-11-17:
@rondoc Try the following and give me data with regard overshoot... This is specifically for candling the night before:

1. start profile at near room temp [65F for me]
2. ramp the temp from 65F to 200F over an hour
3. increase from 200F to 212F over 8 hours [or whatever you choose as your max candling temp and max candling time]

send me the log so I can check out the overshoot.

jbruce12000, 2022-11-17:
A critically damped system will have very little overshoot. In this case, it was less than 1F.

[image]

jbruce12000, 2022-11-17:
One interesting possibility to eliminate overshoot [on a much shorter timescale] is to utilize a smooth curve from fastest heating to 200F. I'm going to test this and see how it works.

jbruce12000, 2022-11-17:
With a lot more trouble, you can create a smooth curve for the kiln to follow... let's call it smooth start. With smooth start over 30 minutes, there is some error at the beginning, but it soon settles...

[image]

jbruce12000, 2022-11-17:
you can see the error for the approach to 212 is a quarter degree F each way...

[final.pdf](https://github.com/jbruce12000/kiln-controller/files/10034931/final.pdf)

rondoc, 2022-11-17:
I will try this probably in a week as I am in the middle of some glaze
firings.
For my setup it is no problem to handle it at beginning manually then to
switch over the pi kiln controller.
Ronald

closed 2022-11-21 by jbruce12000