jbruce12000/kiln-controller #118: New problem of mine not the software

tjerram, 2022-11-20:
I am having problems at higher temps with my kiln.  I think it has to do with the rate of change of temperature my kiln is capable of.  I want to run and log time and temperature of my kiln from 65 to 1500 degrees.  But I can't figure out how to read the temperatures from my thermocouple.  I am not great at OOP so I quickly got lost following your code.  I know that this should be a very simple mater of getting the temperature from the thermocouple and the time and logging that to a csv.  My problem is how do I get the temperature from the thermocouple.

Sorry to keep bothering you with my noob questions.

jbruce12000, 2022-11-20:
There are lots of ways to do this. I think the easiest is just to run a profile that goes from 65 to 1500 degrees. This will record the temp as you go to the logs. After the run, you can post the logs here and I can graph it for you and post the pdfs.

make sure kiln_must_catch_up is set to True so that it will pause the schedule and wait for temperatures to be reached. This way, if your profile is too aggressive, it'll wait until it catches up.

rondoc, 2022-11-20:
It is also a good idea to use a electronic thermometer. It can share the tc
leads and will give you a back up reading. Just parallel the tc wires. One
pair goto the pi set up the other pair goto the digital thermometer. I use
a Fluke as they make a well made and reliable product.

tjerram, 2022-11-21:
I have an electronic thermometer connected in parallel so I can check my thermocouple.   

How do I start the logger or does it always run?  
Where are the log files stored?

I am assuming that the logs are a csv file so I can plot them myself I just need to know where they are.

Thanks for your help I really appreciate all the support you provide.

jbruce12000, 2022-11-21:
the kiln-conroller process always logs every run and records the temperature to that log every duty cycle [2 seconds by default]. The logs go to /var/log/daemon.log if the process is automatically starting.

the logs are not csv. if you want csv format, do this just after your start your profile...

login to the pi

```
cd kiln-controller
./kiln-logger.py --csvfile ./out.csv

```

this will start a client that collects kiln data and stores it in the csv file you provide.

tjerram, 2022-11-26:
I think my thermocouple and Max31855 have a problem I can't get it to sense a temperature above 900 +/- but my glass turns into a blob so clearly the temperature is getting well above 900. I am going to replace both the31855 and the thermocouple and see where that gets me.

closed 2022-11-26 by tjerram

rondoc, 2022-11-26:
Are you checking the tc with a electronic thermometer. Also check that both
of the tc wires have a very high impedance to ground.

tjerram, 2022-11-26:
I did check with a n electronic thermometer (not a Fluke but I think its close) and got roughly the same readings that the max 31855 was reporting.  That is why I suspect the thermocouple.  But I would never have dreamed it could ne hundreds of degrees off.

rondoc, 2022-11-26:
something doesnt make sense. If the 31855 and the electronic thermo were
reading same what is the "hundreds of degree off"  about? Can you clarify
please ?

tjerram, 2022-11-26:
The actual temperature has to be hundreds of degrees above the 900 degrees reported as the glass iseltong into a blob.  The electronic thermometer snd the 31855 are sharing the same thermocouple.

rondoc, 2022-11-26:
Ok. My guess is that the tc wires are shorting together somewhere other
than at the end . This way you'd be reading the temp where they are
touching not at the end where it's much hotter. The wires would have to be
making a firm contact for this to happen. A close physical examination
should show this although it might be a nick where the tc wires go over
some sharp object. Or if you run a completely new set of tc wire the
problem should go.

tjerram, 2022-11-26:
Thanks, I'll give it a try.  The electronic thermometer is connected to the
terminals on the thermocouple so it is not using the same leads as the
31855 but a short in the leads to the 31855 could also affect the
thermometer.

*Tom Jerram*

***@***.***
(607) 235-0316

tjerram, 2022-11-28:
I replaced the thermocouple and the connecting wire and vola it works.  I don't think it was the wires as there was no short.  But. I changed them anyway.

Thanks for all your help.

rondoc, 2022-11-28:
I would go back with old wires and see if problem reappears...just so you
know you've fixed it..but that's just me.
Ron