jbruce12000/kiln-controller #119: Graph Scale

Tibblez, 2022-12-02:
I've pulled the latest version and changed the graph scale to minutes in the config because otherwise I couldn't create a firing schedule granularly enough, however the X-axis on the graph is also in minutes.  The page is also verrrrry slow to load now, I imagine this is related to the drawing of this graph.

[image]

jbruce12000, 2022-12-03:
yeah, I saw this when I changed the scale too. I'll have to think about the best way to fix this. For now, create your profile and then switch the time_scale_profile back to "h" when you're done.

jbruce12000, 2022-12-03:
note to myself... I think the best way to handle this is to have the main graph autoscale. when creating the profile, I you need the accuracy of minutes or seconds, great, but the main screen should just autoscale.

Maybe it would be possible to have a multi-select on the web interface to change the view. This would make it easy for folks to switch back and forth as they see fit.

rondoc, 2022-12-04:
Hi jason, I like your second idea , autoscale and use the web interface.
Ron.

jbruce12000, 2022-12-16:
I decided to handle this in a less complex way which I think should be the default. The graph always displays hours. The editor defaults to using minutes. This keeps the display uncluttered, and allows resolution to minutes [or seconds by changing config.py] if you need that for editing profiles.

this change was made in master and in the blinka branch.

closed 2022-12-16 by jbruce12000

jbruce12000, 2022-12-16:
Note that the graph autoscales, just  the tick spacing is set to an hour.