jbruce12000/kiln-controller #120: Feature request: Actual delta-C degrees/hour heating rate

marktilles, 2022-12-10:
I would LOVE to display the actual heating rate/hour during firings, and record in a log file as well. In this way I could compare the heating rates of the same firing curves at certain key temperature thresholds over time, thus being able to determine the health of the kiln heating elements by reviewing these values over time.

Thoughts?

rondoc, 2022-12-10:
I don't see what data will help you determine the element condition...?
Ronald.

marktilles, 2022-12-10:
Let’s say I know that my oven has the strength with new elements to be heating at 96C/hour while passing 1200C.  If at a later dates I see this creeping down to 70C/hour, then I know I may have a problem some and it’s time to check their health.

jbruce12000, 2022-12-10:
It seems to me that you can get more complete information about elements by graphing stats after a firing...

https://github.com/jbruce12000/kiln-stats

Comparison of **Temperature** and the **Elements On** graph will give you a much more complete view than a constant snapshot in time on the web interface.

jbruce12000, 2022-12-10:
For me, the change in temp over time doesn't tell me much. It's usually very close to the profile I'm running [less than a degree on average across a whole firing], so displaying this info live on the web interface gives me no more information than I have from the original profile.

marktilles, 2022-12-10:
I’m just looking for simplicity. And needing to pull statistics out of a graph isn’t what I consider simple. I’d like to be able to get a visual feeling in some way, simply.

In the “skip” sections of a firing curve, where an oven is on full power, would not seeing the rate of heating of the oven at high temperatures give a pretty good indication if something is wrong? Sure you will always be following within a degree or two if the desired curve if the oven has the capacity to heat at the desired eye at that particular temperature, if PID are well set, but that won’t be the case during skip when the elements are full on, Especially for older ovens, which aren’t nearly as powerful as modern ones.

rondoc, 2022-12-10:
Most useful for checking on Kanthal elements is resistance change. Need to
measure at one temperature. I suggest a record of the 1000F resistance
could be available as a log file and plotted for visual access. The slope
will show a gradual increase in resistance corresponding to lower output
power. Since W = v × v/r. The v is the supply voltage. R is element
resistance. Need to measure resistance at a fixed temp as resistance is
also a function of temperature. For kiln users you could have an alert,
warning when resistance increases so much that that the kiln will not reach
required temperature. Since most kiln users seem unable to understand any
of this its probably a waste of Jason's valuable time.
( I use SiC elements and their electrical characteristics are
completely different.)
Ronald.

jbruce12000, 2022-12-10:
So I have a complete understanding of what you want, you're looking to see the change in temperature over the last minute expressed in degrees/hour, updated every duty cycle [2s by default]?

marktilles, 2022-12-10:
Ron:  Of course resistance measurement is the best method to evaluate kiln health, no argument there! 

Jason: I’ve calculated hourly temperature differentials each 2s duty cycle, but with the apparent sensor deviations over such short periods I don’t get good values there. That’s why I thought each minute would be better.

rondoc, 2022-12-10:
You can get an external watt and amp meter very inexpensively.  It simply
is connected to the incoming power . It measures the watts in so if you
checked it with 100% power you would know the resistance at that point in
the temp. profile. Surely this will tell you enough about the kiln elements
condition? Doesn't this meet the KISS requirement?
Since most kilns do not have a current shunt it will not be possible to
determine element condition directly. Mark's idea seems a challenging way
to make the element determination.
Ron.

marktilles, 2022-12-10:
Haha spoken by a true electrical engineer! Ron I myself am on a budget and avoid buying new tools when I can! 

But like I said, this is Jason’s call. My guess is the programming of the math won’t be difficult because the raw data is already available in the code. Jason?

jbruce12000, 2022-12-11:
The data is there. It's deciding where to handle it... server side or client side and where to display it (low real estate on the web interface). The idea is pretty simple:

1. store the temp and time every duty cycle for a little over a minute
2. select the temp closest to a minute
3. calculate the diff between that and current temp
4. convert time  and delta to degrees/hour
5. display
6. trim data
7. repeat

I'm working in the blinka branch right now, so I'm busy with that, buy will pop this on the stack of features to do.

marktilles, 2022-12-11:
Cool! You know, Jason, I’m going to be here in the USA for a while and I’d love to give you a full demo of all the little bells and whistles I’ve added for myself. Let me know if you’d like that, maybe you’ll find something you like. //Mark

captaincaden, 2022-12-14:
I wrote a simple gui python file (a basic python script without a GUI would work just as well) that grabs the temperature from the API when you start the script, then grabs it again 2 minutes later, and then spits out the output in degrees per hour. At that point, it's a simple calculation to multiple the difference into an hour.

jbruce12000, 2022-12-16:
@marktilles how about cooling rate? should that be displayed? just a negative sign in front?

marktilles, 2022-12-16:
Your choice. I don’t need it although others might find it useful.

jbruce12000, 2023-01-01:
ok, started this. stats now contain the heat_rate. works with simulation speedup. roughly (within a few degrees) matches rate shown in profile editor. next is to display the heat_rate on the web interface.

working on branch heating-rate.

jbruce12000, 2023-01-02:
[image]

jbruce12000, 2023-01-02:
max displayed is 9999 degrees because of lack of space... same with cooling -9999. this rarely happens so shouldn't be a problem.

marktilles, 2023-01-02:
Super! Thanks Jason. I’m going to patch your changes into my thread of code. Much appreciated!

jbruce12000, 2023-01-02:
merged with blinka branch.... which will soon become main/master.

closed 2023-01-02 by jbruce12000