jbruce12000/kiln-controller #121: Feature discussion, starting a profile based on what temp the kiln is

captaincaden, 2022-12-14:
I wish there were a way to easily start a profile, based on where the temperature of the kiln is when the profile is started, the way the controllers on digital kilns use. 

For example: 

Kiln is at 150 degrees ( for whatever reason), when I give the start command, instead of starting at the very beginning of the profile, and having to wait for kiln to cool to wherever the start temperature is, it instead starts at the time, and temp of the profile that would match where the kiln was currently sitting (150 degrees in this example). 

It'd be nice if this were an option, so that you could choose when starting to start at ____ or start from the beginning, to rule out other issues that may come along with the kiln starting at it's present temp, or those who want the profile to start at the very beginning. 

I'm not really sure the logistics behind how to do this, or what other consequences that may create, but it'd be a cool feature. Especially if you have to stop the kiln in the middle of a firing for whatever reason, and then begin again without the kiln having to cool or heat back to the beginning, or without having to figure out where to startat with the API.

Just something to discuss thoughts on 😃

jbruce12000, 2022-12-14:
This is interesting and I have run into this problem before. The way I handle this today is by calling the api and start at minute X like so....

curl -d '{"cmd":"run", "profile":"cone-05-long-bisque","startat":60}' -H "Content-Type: application/json" -X POST http://0.0.0.0:8081/api

which starts cone-05-long-bisque at minute 60... but I see you already know how to do this. So, maybe startat_temp is what you'd like? Note that it would probably have to pick the first one it finds at start there. Some profiles might have the temp twice [once during heating and once during cooling].

rondoc, 2022-12-14:
I agree with you. It would be really useful as when power goes out or I
accidently switch off the pi. Lots of idiot activities surround
potters...not called potty for nothing!

jbruce12000, 2022-12-14:
or maybe you were thinking something more automatic... you click start and it seeks the closest point to the start in time where that temp is...

jbruce12000, 2022-12-14:
@rondoc for power outages, the automatic restart feature restarts at the same point in the profile as when the outage occurred [for up to X minutes configurable, 15 by default].

rondoc, 2022-12-14:
That would be great. So this is sequence.
1. Loss of control for any reason
2. Kiln temp slowly drops due to lack.of power input.
3. Press start and the system starts at the place in profile closest to
where it is at that time.
4. System continues to keep on the profile.
5. If system cannot restart alarm set.
Is this correct?
Ronald.

captaincaden, 2022-12-14:
> or maybe you were thinking something more automatic... you click start and it seeks the closest point to the start in time where that temp is...

Yes, I was thinking more automatic like this. Using the startat API works, but it takes a bit of time to think about where in the profile I'd need to tell the kiln to start at based on where I am temp wise in the firing, and then to open a terminal somewhere to send that request. 

A real world scenario was I have a profile built to run a 'candling' that runs for 6 hours below 200 degrees, but I almost always begin my actual bisque profile before that profile is over. So kiln was sitting at 200 degrees when I ended the candling profile, and started my bisque profile, it had to drop to 50 degrees (the starting point of my bisque profile, so technically 60 since catch up kiln_too_hot would kick in then) before it would actually begin the firing, even though I was already 150 degrees hotter. Yes, startat _____ time API would resolve this, but requires a lot more action somewhere where I can access a terminal than just having the option for it to do it automatically (similarly to my other digital kiln)

My other digital kiln, if for some reason I'd need to stop, to change a program, or whatever else, it would just automatically pick up running at whatever temp the kiln was at when I pressed start in relation to the program I selected. 

I imagine this would have some conflict with the kiln_too_hot and kiln_too_cool functions though.

The automatic restart function is great for picking up in the event of a power outage or restart of some kind of the pi. But doesn't address if I physically want to stop the profile myself, and change it, edit it, or whatever else without needing to use the API after to begin again.

jbruce12000, 2022-12-15:
Note to self... implement as a single setting with no changes in the web interface. This seems like a **sane default** to me and is what folks would expect to happen so it should be set to True as the default.

jbruce12000, 2022-12-15:
ok. will add as a feature. this is third in line right now. won't happen right away.

captaincaden, 2022-12-15:
Sounds good! Thank you again for all you do with this project.

chipgarner, 2022-12-19:
@jbruce12000  I think I can do this one. Shall I try it and make a pull request?

I think it should only happen at startup if the kiln is above the start temperature to avoid picking the wrong segment. Just find the first segment in the profile that intersects the current temperature. 

Also, maybe it should wait a few seconds to make sure it is not a thermocouple start issue.

Does this sound right?

jbruce12000, 2022-12-19:
It should: 

1. be a setting that is **True** by default maybe like this... `seek_start=True`
2. the setting should be clearly documented so folks understand what it's for
3. should be completely ignored if `automatic_restarts = True` and a restart is needed
4. should seek the time [of the first instance in the profile] in seconds of the point closest to [without going over] the temperature of thermocouple. there might be more than one time in a profile where a temp exists [usually one while heating and another while cooling].
5. should log that a **seek** is happening and the time in the profile to seek to
6. should call oven.run_profile with the given profile and seconds/60
7. should only happen if oven.state is IDLE
8. should do check in oven.run just after the restart code

Create a branch off blinka for your changes. That's where I want to test it. This one will require significant testing.

That's all I can think of.

chipgarner, 2022-12-19:
This is clear

This will start the firing automatically without pushing Start any time the temperature is above the first temperature. Is this the intent? It may not be expected.

The check could be in run_profile and then it would not start without pushing Start. It would have to re-check if this is an automatic_restart but I think will work.

?

jbruce12000, 2022-12-19:
hmm. not what we want.... so doing the check in run_profile is probably ok. you could just have it check the temp, compare it to current temp in profile, if off by N degrees, set oven.runtime.  All the other stuff you apply though.

chipgarner, 2022-12-23:
Please comment! If you don't understand this no one else will either.

I have the code working and have tested it every way I can think of with the simulator. I will make the pull request soon. The figures below show icons etc. from a merge with Mark Tilles version, not part of the upcoming pull request.

## Seek Temperature on Hot Start

Turn this feature off by setting seek_start = False in your config.py file. It is on by default.

The purpose of this feature is to prevent having to wait for the kiln to cool if it is above the scheduled start 
temperature when the Start button is clicked. This can happen if you have preheated the kiln manually or with 
another schedule, or if it was shut off accidentally, and you needed to restart the firing.

If your kiln is above the scheduled starting temperature when you click the Start button this 
feature will start the schedule at that temperature. Suppose your firing profile starts at 50 
degrees and ramps to 200 degrees in one hour. (Centigrade or Fahrenheit, it doesn't matter in this example.) 
If your kiln is below 50 degrees it will start normally by first heating to 50 degrees. If it is above 50 
degrees when you click Start, say 150 degrees, it will seek that time for 150 degrees in the schedule and 
start the firing at that time. In this case, about 40 minutes after time 0, see the figure below.

[image]

The seek feature always looks for the first segment in the firing profile that meets or exceeds the current 
temperature when the Start button is clicked. In the figure below the kiln is at 500 degrees and the schedule
is started at about 2:40.

[image]

rondoc, 2022-12-23:
I like it! This would be a helpful feature. Would this cover any incidence
during a firing? What would happen if the reason was a power dropped out ,
in which case the pi would also be resetting itself? Having an automatic
feature like this is nice but might lead to unintended consequences so it
somehow must let the user know the status at any time. May be an alert when
it is in a auto restart mode? If you think my comments are unworthy of this
great effort I am happy to not add any more.
thanks for all of this.
ronald.

captaincaden, 2022-12-23:
@rondoc the automatic restart feature already exists. If a profile is running, and then something other than a user or API stop stops it, e.g. power disconnect, power outage, pi reboot, etc. (and you are up to date with updates since this was a new feature added in 2022), it will automatically begin where it ended if within the configured parameters. 

@chipgarner that all sounds like what I was thinking. @rondoc mentioning the automatic restart feature, maybe you should also simulate an automatic restart scenario to see if there are any conflicts? With this feature added, and running as the default, it may be sensical to change the way profiles are written too. For example, I could see (in my uses) setting the first segment always at zero, then the next at whatever temperature. That way the kiln would always be following the exact ramp I wanted in the beginning because it will automatically start somewhere between 0 and segment 2, instead of needing to full heat to get to segment 1 temp.

chipgarner, 2022-12-23:
@rondoc  Thanks for your comments! An alert is a good idea. It shows up on the graph but that could be missed.
@captaincaden  It is not activated during an automatic restart. The automatic restart default is 15 minutes. If a power failure (or whatever) is longer than 15 minutes the seek will happen. This is probably what you would want.

Also, the seek never happens unless you click Start and it happens right away.

It will log, it's not in yet. The segment idea is a good one.

captaincaden, 2022-12-23:
@chipgarner That makes sense. What about if the process is started via the API? Does it still seek then?

chipgarner, 2022-12-23:
The code is complete and logs the seek. First draft of the documentation is above.

I like @rondoc's idea but am not sure how best to implement it. Maybe draw a line from zero to the start in a different color?

I have only tested in the simulator. (I have a test kiln now and have started ordering parts, but so far can only simulate.)

rondoc, 2022-12-24:
@chip...off topic but I have built several kilns and would be pleased to
advise. I use SiC elements as they are good to over cone 10 and have very
long life. Not expensive either. If you are interested my email is
***@***.***

chipgarner, 2022-12-24:
@rondoc Looks like Github obliterated your email, unless it is actuallY ***@***.***

It's because these discussions are public. 

Try this, send me a text at four one five two one eight seven four nine seven

This may not work either ...

chipgarner, 2022-12-24:
@captaincaden  Good point, I need to test with the API. The master branch in KilnController (not kiln_controller) on my Github runs it with the fast simulator if anyone wants to try it. https://github.com/chipgarner/KilnController.git

The seek branch in kiln_controller has the latest for the pull request to the blinka branch here. https://github.com/chipgarner/kiln-controller.git

jbruce12000, 2022-12-24:
@captaincaden not sure about the secondary request to always have profiles start at zero degrees by default [uneditable] and then allow folks to edit point 2 from there. you're making the assumption that point 2 will be above the current temperature of the kiln [which may not be a valid assumption]. I think this idea should not be included in the feature for now... let's discuss it and create a new feature. I want his feature Chip is working on to be small, easy to merge. and easy to test.

jbruce12000, 2022-12-24:
@captaincaden there is no reason why all your profiles could not start at zero... and always have a second point that is higher in temp than the kiln is when you start the profile.

captaincaden, 2022-12-24:
I was thinking more along the lines of just highlighting that profile thing, not an an actual uneditable change in the code @jbruce12000 . I was more so just thinking out loud, and adding where designing profiles around this feature may make them work even better/keep ramps more stable along the profile you want it to follow in the very beginning. As is, unless somehow you're always starting at the exact same temperature, there's always going to be some sort of kiln too hot or kiln too cold situation in the beginning, where the kiln will have to catch up, but if we start at the exact time in the profile based on the current temp of the kiln, (the point of this feature), that won't happen anymore. 

More so just outlining a highlight of this feature. I think it should be something that each person who designs their profiles should think about, but I think the current profile editing works great.

jbruce12000, 2022-12-24:
@chipgarner testing your seek branch. I tested with 

automatic_restarts = True

and stopped and started the process [just control-c on the cmd line], and it automatically restarts, but jumps forward in time each time the process is stopped started. bug.

chipgarner, 2022-12-24:
Fixed this. I though it was not doing the seek on automatic restarts. Now it is.

I have pushed this fix to my for but it has not yet updated the pull request.

I still need to test with the API

jbruce12000, 2022-12-25:
I have not re-tested this yet, but the goal is to be able to have automatic restarts enabled, but seek does nothing if a restart is/should happen.

chipgarner, 2022-12-26:
Don't test yet, I am trying to understand what is happening with API starts. 

The behavior should be no seek on automatic restarts.  Also no seek on API starts with a set start time.

captaincaden, 2022-12-26:
@chipgarner That sounds like the ideal behavior. 

The API calls oven.run_profile with profile data provided in the API request in kiln_controller.py and defaults to startat 0, if one isn't provided in the APIrequest.

chipgarner, 2022-12-26:
If the program is shut down and restarted it checks for an automatic restart. If the schedule is stopped, with the Stop button or the API, there is no check for an automatic restart. I find this a little confusing but this is how it currently works.

If seek is on it is skipped it there is an automatic restart. Seek is also skipped if an API start specifies a start time. Seek is otherwise always checked.

In all cases, the seek option will only do something if it is checked and the current temperature is above the target temperature.
1) Program start. Autostart IS checked, If not auto starting Seek IS checked.
2) The firing schedule was stopped with the Stop button, and the Start button is pressed. Autostart is NOT checked. Seek IS checked.
3) The firing schedule is stopped with the API and the Start button is pressed. This is the same as 2, Autostart is NOT checked and Seek IS checked. 
3) The firing is stopped with the API or the Stop button, and the API is used to start a schedule without specifying a start time greater than zero. As 2 and 3, Autostart is NOT checked and Seek IS checked.
4) The firing is stopped with the API or the Stop button, and the API is used to start a schedule with a start time greater than 0. Autostart is NOT checked, Seek is NOT checked.

Whew ...

chipgarner, 2022-12-27:
Please test! 
Looks like everything is working as discussed. 

You can git clone or git pull the code from the seek branch on https://github.com/chipgarner/kiln-controller.git

This is the same code as in the pull request here,

jbruce12000, 2022-12-29:
confusion about stop:

If someone either stops a profile by clicking stop in the web interface, or uses the api, they have taken an action. automatic restarts would go against that action. why is that confusing? I want to make sure I'm not missing something with this question. The best way to simulate a power outage is to stop and start the process - or yank power to the rpi.

temp high, profile low:

I tested the case where the temp exceeds anything in the current profile... and it does what I would expect... nothing.

start temp of 255F:

I see that you have 255F statically set as the simulation start temp. I understand why and I agree for testing it's great. It should use the variable sim_t_env so that it's configurable. set it to 255 in config.py and have the code pull it in.

I have a lot more testing to do, but I'm not feeling well. going to take a nap.

chipgarner, 2022-12-30:
@jbruce12000 I am easily confused. The start logic makes sense. 

Will add the sim_t_env change.

Hope you are feeling better soon!

jbruce12000, 2022-12-30:
@chipgarner I'm done with my testing and found no other issues. lemme know once you post the above change to the PR and I'll get it merged with the blinka branch.

chipgarner, 2022-12-30:
@jbruce12000 PR is up.

jbruce12000, 2022-12-31:
Nice work Chip. This is also merged with Blinka.

Blinka will become the active master/main in the near future. Still testing to make sure everything is stable.

closed 2022-12-31 by jbruce12000

captaincaden, 2022-12-31:
I will have to do a full update along with transferring some of the changes I've made over the past year sometime in the future, and then play with it then. Still a few updates behind. 

Thanks guys!

captaincaden, 2023-10-23:
I know this was added in the blinka branch, but I'd love to see it added in this version sometime. I wanted to make the switch to the blinka branch, but I just don't have time to disassemble and unsolder the things I've already done in between firings to make my outputs and inputs compatible with the blinka branch (unless that's changed, I remember some discussion was mentioned about it).