jbruce12000/kiln-controller #122: Fast Simulator

chipgarner, 2022-12-15:
HI all
I am interested in the issues here and am trying to get a good understanding of the code. I am looking for a test kiln to experiment with. Since I don't have anything real to test with I am looking at test code.

The simulator is the obvious start. I have been able to get it running at 1000 times normal speed, so the 15 hour simulation runs in about a minute. This only required small code changes but they are in several places. It needs cleaning up but is available on (https://github.com/chipgarner/KilnController.git).

If anyone is interested I can clean this up and create a pull request.

jbruce12000, 2022-12-15:
Hey Chip... make sure you're working from the blinka branch. That branch will soon become Master/Main.

jbruce12000, 2022-12-15:
So I understand, you want simulations to run at light speed. Should this apply to all simulations? or just if you have a configuration parameter set?

I think your goal is to speed up testing times...

chipgarner, 2022-12-15:
A config parameter mainly for testing.  I have it running but it needs cleanup.

rondoc, 2022-12-15:
Please explain why high speed testing for a very slow process.
Thanks.
Ronald.

chipgarner, 2022-12-15:
It's for testing the software, so you don't have to wait to see if a change broke something. It has limitations but can be extremely useful.

The simulator uses a simulated kiln that can heat up at a certain rate. If you tell the simulator code the kiln has been heating for 10 seconds it computes the temperature change, it doesn't know you have tricked it and it was only one millisecond. The PID should respond as if the ten seconds has passed. 

It can also be useful for testing the user interface. An obvious example is if you want something to be displayed at the end of a firing you don't want to wait forever to test your code.

Final testing always needs the real hardware.

jbruce12000, 2022-12-16:
ok. This is the 4th feature in line. I'm currently testing the blinka branch and so it'll be a while before I get to this.

chipgarner, 2022-12-16:
I created a pull request for when/if you have time.

jbruce12000, 2022-12-31:
Nice work Chip. Tested. Merged into Blinka.

closed 2022-12-31 by jbruce12000