jbruce12000/kiln-controller #133: *Not an issue*, add on kiln vent, looking for some guidance.

captaincaden, 2023-03-03:
So I added an additional relay, and some code to flip on my kiln vent when the firing starts, and then off when it ends. This all works great, and I'm happy with all of that aspect and how it's behaving. I added the code in oven.py, and it basically just calls to code I added in the output class to trigger that relay on or off based on the state of the kiln. 

I've been looking around the code, and I can't seem to figure out how the process calls for any of the 'lights' in the top right corner of the websocket to be in their active states, or where it gets information on when to turn them on or off. Ideally, I'd like to have some kind of indication in the websocket that this process of turning the kiln vent's relay on happened. Basically, I'm trying to figure out how I can activate the ds-led air indicator when that process happens. 

Any tips or help anyone could give would be appreciated! I'm pretty fluent with Python, but my js and css knowledge is limited haha

chipgarner, 2023-03-03:
@captaincaden  I think I figured this out. The only call that updates the graph etc. is notify_all in ovenWatcher.py. The front end gets these messages from the web socket. ovenWatcher calls notify_all with the oven_state in its while loop, line 34 in the Blinka branch version I am looking at. It gets this from a call to oven.get_state. Everything sent to the front end is shown nicely in the get-state function in Oven.py, starting at state = { ...  Line 479 in my version. None of the leds are currently set or working in this version.
The front end captures this message in ws_status.onmessage, line 503 in picoreflow.js. It updates a lot of stuff, the leds are at the bottom of the onmessage function, line 584. if (x.air > 0.5) will turn the air led on or off.
I stuck one line in oven.py in the state and it turns on the led:

        state = {
            'cost': self.cost,
            'runtime': runtime,
            'temperature': temp,
            'target': self.target,
            'state': self.state,
            'heat': self.heat,
            'heat_rate': self.heat_rate,
            'totaltime': self.totaltime,
            'kwh_rate': config.kwh_rate,
            'currency_type': config.currency_type,
            'profile': self.profile.name if self.profile else None,
            'pidstats': self.pid.pidstats,
            'air': 0.9
        }

[image]

Hope this helps!

captaincaden, 2023-03-03:
Thank you @chipgarner ! That worked. I just set 'air': to my vent_state variable in oven.py, and then changed the part in picoreflow.js to be if x.air == true { etc. etc. etc. and it works great. Thank you again.

closed 2023-03-03 by captaincaden