jbruce12000/kiln-controller #134: profile - always store using C of F

jbruce12000, 2023-03-08:
Today profiles are stored in the format specified in config.py... either C or F. This means a few things... 1. if you ever change this setting, all loaded profiles will be wrong. 2. folks cannot easily share profiles (folks need to do conversion by hand... then likely store the profile under a different name)

Change profiles so that they always store in Celcius. Convert if needed during load/display.

It would also be possible to store the type C or F. Not sure if this would be confusing for folks or not.

chipgarner, 2023-03-09:
I like the store always in Celcius and convert if needed for display. This makes it easy to share files, and I think is the least confusing both for programmers and users.

My experience on a lot of very different projects is that it works best to always store and compute in one set of units and convert as needed for users as close to the UI as possible. It reduces unexpected results. like opening a profile and finding the temperatures are way off.

steeleh, 2023-03-09:
I agree, Celsius would be the logical universal storage units not least because if a potter was to run someone else's profile thinking it was in F (when it was in C) it would fire low (safe). If a F profile was run in a C set up it would be 9/5 + 32 too high and potentially embarrassing.

jbruce12000, 2024-07-03:
ok this feature is complete and added to the main branch.

profiles will always be stored in degrees c. once loaded, if needed, they're converted to f.

Profiles have a new key, `temp_units`, added if they're saved. It is always set to "c". Some profiles [if they are never saved] will never have this key added. The code deals with this as users would expect.

closed 2024-07-03 by jbruce12000