jbruce12000/kiln-controller #135: Daylight savings time, kiln stopped.

captaincaden, 2023-03-12:
I'm currently pretty behind in relation to updates, but thought it should be mentioned here. I ran a bisque firing last night, and exactly at 3:00am (daylight savings time) the kiln stopped itself, without any errors. I was only 100 degrees from the end of the schedule on a bisque, and this only happens two times a year (for now), but still thought it'd be interesting to mention here. 

Here's the logs from the event: 
Mar 12 01:59:59 raspberrypi python[337]: 2023-03-12 01:59:59,198 INFO oven: temp=1754.44, target=1754.56, error=0.12, pid=79.97, p=0.74, i=85.71, d=-6.48, heat_on=1.60, heat_off=0.40, run_time=36576, total_time=39600, time_left=3023
Mar 12 03:00:01 raspberrypi python[337]: 2023-03-12 03:00:01,206 INFO oven: temp=1754.51, target=1754.64, error=0.13, pid=88.32, p=0.80, i=85.72, d=1.80, heat_on=1.77, heat_off=0.23, run_time=36578, total_time=39600, time_left=3021
Mar 12 03:00:01 raspberrypi python[337]: 2023-03-12 03:00:01,209 INFO oven: kiln outside pid control window, max cooling
Mar 12 03:00:03 raspberrypi python[337]: 2023-03-12 03:00:03,211 INFO oven: temp=1754.68, target=0.00, error=-1754.68, pid=0.00, p=-11086.09, i=85.72, d=0.00, heat_on=0.00, heat_off=2.00, run_time=40180, total_time=39600, time_left=-580
Mar 12 03:00:03 raspberrypi python[337]: 2023-03-12 03:00:03,212 INFO oven: schedule ended, shutting down

rondoc, 2023-03-12:
#captaincaden sure you did not have 1754 as max temp.?

captaincaden, 2023-03-12:
@rondoc positive, schedule is for 1865. Note the time_left value changing in the logs from 3021 minutes to -580 minutes after the change of time, and target changing to 0.

rondoc, 2023-03-12:
Interesting and disconcerting too. I had a similar shutdown a few cycles
ago. Thought it was a power glitch but not repeatable. I tend to watch my
kiln when within a 100 or degrees from the top as in a glaze firing a early
shut down would be bad. I can easily switch to manual control
at any time and back to pi control if I want to..

captaincaden, 2023-03-12:
I do the same. I just finished glazing, reloaded and am running the glaze firing for this bisque now. Hoping the absence of that 100 degrees doesn't affect much in this firing. I had considered that maybe could run into issues with the clocks going forward, but I really didn't think it'd be a problem. When I saw my notification for the ending of the firing at 3 last night, I really didn't think anything of it, until I saw the graph when I woke up this morning and saw it was incomplete.

jbruce12000, 2023-03-24:
updating to latest is not going to fix this. it's a bug. I'll have to think about how to fix it.

In the meantime, can someone lobby congress to pass a bill so time changes just go away?

captaincaden, 2023-03-24:
> updating to latest is not going to fix this. it's a bug. I'll have to think about how to fix it.
I didn't think so, but wasn't sure if maybe somewhere along the line since I last updated something that would affect this would have changed. 

> In the meantime, can someone lobby congress to pass a bill so time changes just go away?
Hoping for this lol.

chipgarner, 2023-03-26:
This might be a fairly easy fix. It uses datetime, Simply changing to ticks (time.time()) should fix it but will probably make times displayed a little irritating to users. Use ticks for all times internally and convert when displaying times of day, probably in the javascript. Also, it might be testable by switiching time zones during a firing.

jbruce12000, 2026-08-16:
fixed in the rewrite branch with test coverage.

closed 2026-08-16 by jbruce12000