jbruce12000/kiln-controller #137: Random no connection with max31856 and quits

MeltTechAu, 2023-05-12:
I have a max31856 actually a few of them they don’t seem to work with jbruces software they spit out random no connection which quits program I know I can ignore error but then the issue of thermocouple breaking and melting down your kiln hmm jbruce how do we fix this issue how to stop it going into emergency quit

rondoc, 2023-05-12:
> I see more errors than I like with the 31856. I have experimented with the
> software and suspect it is the python library. Adafruit mentions a possible
> timing issue, Python is not great with timing. On the other hand, several
> people seem to be firing just fine using the 31856.
>
> The 31856 call blocks and I have seen it hang several times. This is a
> worse problem as it hangs the calling thread, no error it just hangs. This
> could be a problem with the one 31856 I an using.

I have used both 31856 and 31855. Both work well with original pi etc.
Problem with both is that the thermocouple MUST be really floating. No
connection to ground!! Also the 55 seems to be very stable and the 56 seems
more sensitive to electrical noise.
R.

jbruce12000, 2024-08-15:
@chipgarner any updates on the max-31856 or adafruit libs?

jbruce12000, 2024-08-15:
Sorry @chipgarner, I think we've explained @MeltTechAu max-31856 issues. I'm gonna close this.

closed 2024-08-15 by jbruce12000