jbruce12000/kiln-controller #138: Max 31856 in s or r mode floating temp  20 deg out

MeltTechAu, 2023-05-12:
Why does in s or r type thermocouple mode temperature like float counts up or down also not stable even when using offset I have a digital calibration unit to test and s and r are out by 20 degs or so but k and it’s span are ok wits going on here I really need s to work

chipgarner, 2023-05-12:
I am not sure how you are testing but thermocouples are not grounded, and grounding your test unit could be a problem. Have you tried a real thermocouple? What is the voltage range of your test device?

The 31856 should work.

MeltTechAu, 2023-05-15:
Just noticed after all that the ambient temperature can be read but is 10deg c out the dam 4 units I tested won’t read temperature even heating with a lighter wot the hell is wrong?help really needed on this max 31856 please

jbruce12000, 2023-05-15:
What branch of the code are you running? Can you post a schematic or did you follow any particular schematic so we can verify your wiring looks legit?

You might want to try the blinka branch. It's the latest code and would likely require you to change your wiring a bit. It has a script for testing your thermocouple: https://github.com/jbruce12000/kiln-controller/blob/blinka/test-thermocouple.py

MeltTechAu, 2023-05-22:
Thanks for the advice I will look into the blinka code I use max 31855 with no problems I work in the kiln industry and have access to a calibrator I assemble bartletts in a box for the business I work for. Any assistance would be greatly appreciated JB as I reckon your software is better any day you have done an awesome job mate keep up the good work 
Brett from melttech.com.au I have spent a bit of time in the hardware and software last 5 years
Also I have programs done for the whole cone chart if you would like them please download from user melttech on this site my contribution for JB online program storage I also have fast programs but adding to many programs seems to loose some on load if u would like them shoot me an email to send them to you programs were taken as a guide from a Bartlett genesis

jbruce12000, 2023-05-22:
Brett, nice to meet you. 

For kiln schedules, post a link so I can check those out. We have an open issue for creating and sharing schedules. This depends on temperatures being stored in a single temperature scale [like degrees C]. Once that is done, we'll have a github repo where anyone can contribute schedules...

MeltTechAu, 2023-05-28:
Nice to meet you to Jason
have not had a chance to sort ma31856 out been busy as for the firing profiles
https://melttech.com.au/Firing_Profiles.html
you can download as a zip or individual they are in degrees C 
i tried to figure out your online storage to no avail so if you could add them would be great
cheers Brett

MeltTechAu, 2023-06-26:
Hi Jason I have tried blinka and get no thermocouple at all just says  zero degrees have tried it with both max chips 55 and 56 to no avail am I missing something it still reads in your software and mine master branch  I have also tried Vanderins  3 zone version which is similar it doesn’t read a 56 chip either 
Any help would be greatly appreciated thanks in advance

jbruce12000, 2023-06-26:
So you're using the blinka branch... make sure the right chip is configured, make sure the right GPIO pins are configured... since it is SPI, you must use specific pins [check out config.py].

Did you test with test-thermocouple.py script?

jbruce12000, 2023-06-26:
I've never looked at Vandarin's fork, but I can tell you the concept of having each zone inside a single kiln separately controlled by a PID will not work. Even if there was a separate TC for each zone, this would not work. The PIDs would constantly struggle with each other.

MeltTechAu, 2023-06-28:
I got it to read temperature on the max 31855 but after a reboot I have to turn off spi  module then reboot then after reboot reactivate as it will only show 0 deg c why is this bit of a pain have you got any idea around this problem blinka

jbruce12000, 2023-06-28:
I don't understand this at all. I don't know what "turn off spi module" means. If you're typing commands in a shell, I want to see them. I'd also like to see the logs so I can help troubleshoot.

MeltTechAu, 2023-07-12:
Jason if i do this commands below t reads 0 deg c from boot up
sudo raspi-config
interfacing options -> SPI -> Select Yes to enable
select reboot
But if i set in desktop settings spi on/for the first time  it works until i reboot then reads 0 deg c even though it says SPI is on
would love your input
Again thank you for your valued time and excellent work you have done alot of it :)

jbruce12000, 2023-07-12:
hmm...

`ls -la /dev/spidev*`

and post the output after your reboot and it reads 0c. if there is no output, do this...

try adding `dtparam=spi=on` to `/boot/config.txt`

jbruce12000, 2023-09-28:
@MeltTechAu did you get either the 31855 or 31856 working on any branch? I'm trying to clean up and close your issues.

MeltTechAu, 2023-09-28:
Hi jason as of yet no i got no idea y  boughT a new board and all i got the
older branch to show it dor about a minute but blinker wouldnt show it in
spi i sus it out again if you want to clean up no probs im sure ill get to
work will move to blinka soon ill test it out than use gptchat program to
add some more features have a good one

jbruce12000, 2023-09-28:
ok. closing this issue. open another when you have the time to test.

closed 2023-09-28 by jbruce12000