jbruce12000/kiln-controller #139: MAX31855 25th bit not working

harthenry, 2023-05-21:
Kiln Controller all finished and running.  Except, there is one issue with the MAX31855 thermocouple board.  It appears the 25th bit is not working.  Once the temperature gets high enough (over 175 degrees - which is the 25th bit), the temperatures are all wrong.  I have ordered the MAX31856 to see if using a different type of thermocouple conditioner board will help.
[image]

harthenry, 2023-05-24:
Installed new conditioner module (31855), same issue.  rewired the conditioner to use another set of PI pins -- now works just fine.  Perhaps the PI has some bad inputs?? Anyway, back on track.  Now tracking down why thermocouple show 100 degrees too cool.  CJC is correct, so, perhaps some calibration error. thx

closed 2023-05-25 by jbruce12000

harthenry, 2023-06-02:
FYI: I found the 100 degree offset problem -- inside the enclosure that has the Solid State Relay, I have a set of 115 VAC fans that blow air across the heat sink mounted to the SSR.  The two 115VAC muffin fans are connected to Leg A 115VAC supply side of the SSR (so the fans are on all the time).  The sinusoidal waveform output of Leg A A/C output is back feeding noise through the Analog Output connected to the SSR to the Raspberry PI.  This interference provides almost an exact 100 Degree error on the thermocouple input reading.   Wiring and running the fans from a nearby outlet solved the problem -- for now.  I will need to figure out some filtering so I can plug the fans back inside the SSR enclosure

rondoc, 2023-06-02:
I don't see how induction fans can produce any RFI. The SSW can and will
unless they are zero crossing switched.  Nice to see you're playing
electronic detective...but look elsewhere than the fans.
It's easier to use a zero crossing SSW than trying to suppress or isolate
high current SCR switching.
RD E.E.