jbruce12000/kiln-controller #142: Schedule taking much longer than specified

sactheskydiver, 2023-07-06:
Using the premade cone 5 long bisque I started firing the 15 hour schedule approx 13 hours ago yet the time remaining is 10 hours rather than 2. Watching the count down timer I can definitely confirm the clock is counting down much slower than the actual passage of time.

jbruce12000, 2023-07-06:
wow. that's bad. can you post the log and config.py so I can check that out?

jbruce12000, 2023-07-06:
There are lots of possible causes for this. Most likely is that the temperature is constantly outside the window and the kiln is waiting to catch up.

sactheskydiver, 2023-07-06:
Where will I find the log? I'll post it asap 

I haven't done any calibration or tuning yet so being outside the window is extremely likely

jbruce12000, 2023-07-06:
see https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md

that has instructions for how to zip up the log.

If you want the kiln to strictly follow time and never wait to catch up, set the following in config.py:

kiln_must_catch_up = False

This is NOT the default because most people want to heat as fast as possible during some part of their run.

sactheskydiver, 2023-07-07:
So I just went ahead and did the PID auto tune because of the ossilations I got in the previous test and checking the log it was very much due to being outside the window. 

It recommended these settings. 

Kp= -20.8 (a negative)
Ki= 10.2
Kd=1116

The P and D seem widely outside what the suggested starting point was. Is this likely a hardware issue?

jbruce12000, 2023-07-07:
haha. that's not right. Kp should always be positive. The higher the number the more impact it has. Ki should always be positive. The smaller the number, the larger the impact. Kd should always be positive. The larger the number, the higher the impact.

You can do manual tuning. This is an iterative process where you change one setting Kp,Ki or Kd and test then reflect. This is a pain and takes time, but you'll get close after a few tests. No tuning will be perfect at all temperatures... at low temps, it's ok for temps to swing a little as long as it settles. Check out the graphs here for a well tuned kiln...

https://github.com/jbruce12000/kiln-stats/blob/main/output/final.pdf

You can see temps swing at the start, but then settle to an average error of just a degree F. 

See this doc for manual tuning:

https://github.com/jbruce12000/kiln-controller/blob/master/docs/pid_tuning.md

Maybe @adq can suggest why the autotuner does not work in this case. It's likely related to the gpio output inversion you have.

jbruce12000, 2023-07-07:
since we know the cause of the issue and you're working on tuning [the fix], I'm gonna close this. open another issue if ya need.

closed 2023-07-07 by jbruce12000