jbruce12000/kiln-controller #144: SSR power

antipop001, 2023-08-15:
Hello I noted your comment "Note: I tried to power my ssr directly using a gpio pin, but it did not work. My ssr required 25ma to switch and rpi's gpio could only provide 16ma. YMMV." What was your resolution to that?

rondoc, 2023-08-15:
Solution is to use an optical isolator chip that is designed to drive SCRs.
They are zero crossing types so prevent electrical noise. It's a simple
connection from the pi
but you might need to use a transistor to invert the signal. If you want
full details email me direct at ***@***.***
Ronald.

antipop001, 2023-08-15:
Hello, I would love full deatils.  Unfortunately it hid your email address. mine is ahermon at gmail  if you want to send me the info.

rondoc, 2023-08-15:
This is using two scrs but you could use a solid state relay instead. Make
sure the switches can carry the full current. You might need a heat sink
and or a fan. I use two scr s as they work well and have plenty of carrying
and voltage ratings. Some of the SSR claim higher current ratings than they
should for long time ON use. Like a 12 hour kiln cycle.

antipop001, 2023-08-15:
> This is using two scrs but you could use a solid state relay instead. Make sure the switches can carry the full current. You might need a heat sink and or a fan. I use two scr s as they work well [...]

Thanks for helping. Unfortunately your

antipop001, 2023-08-15:
unfortunately your attachment did not come through.

antipop001, 2023-08-16:
> > This is using two scrs but you could use a solid state relay instead. Make sure the switches can carry the full current. You might need a heat sink and or a fan. I use two scr s as they work well and have plenty of carrying and voltage ratings. Some of the SSR claim higher current ratings than they should for long time ON use. Like a 12 hour kiln cycle.
> > […](#)
> 
> Thanks for helping. Unfortunately your

Hello The attachments are not coming through. Could you send to ahermon at gmail ?  Thanks!

antipop001, 2023-08-17:
@rondoc Is there a way you can send me the attachments?

rondoc, 2023-08-17:
Simple answer is to use a NPN TRANSISTOR. It will invert the output but
that seems ok . The transistor will drive the SSR OK. Almost any NPN
transistor will work.
My attachment  for a 2 scr plus zero crossing chip was sent to yr email .
Ronald.

antipop001, 2023-08-17:
@rondoc I so appreciate the help I am not seeing the email unfortunately.  ahermon @ gmail com or ahermon @ outlook com. or I could send you an email to reply to if you enter it in a spaced out fashion. Again I appreciate you sticking with this.

rondoc, 2023-08-17:
Rdoctors at gmail

antipop001, 2023-08-17:
Got the pic

closed 2023-08-17 by antipop001