jbruce12000/kiln-controller #145: Trigger SSR

antipop001, 2023-08-18:
Hello. I can trigger my relay with the commands:
raspi-gpio set 23 dh & raspi-gpio set 23 dl
But I can't get it to trigger running a kiln job from the web ui.  I also see that the target temp is not changing. 
Attaching config.y and scree
[image]
nshot. 
[config.txt](https://github.com/jbruce12000/kiln-controller/files/12381183/config.txt)

rondoc, 2023-08-18:
I use either  my cell or my pc and both work great. I seem to need to open
kiln.py and tell it open in window. Then when I open the app I can see that
it is connecting. This is all on my local network.
Once you get over this stuff you will be pleased as the app is very stable
and reliable. I like that I can view , on my phone, the progress in the
kiln. Your next step will be getting your PID numbers..there is a very
helpful process that you can follow.
Ronald.

jbruce12000, 2023-08-21:
hi @antipop001 . If you have not yet got this working, can you post the logs so I can see what happens when you try to start a run?

see https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md

which describes a script to grab and zip up all logs to a single file. post that somewhere I can download it. (you can do that here on github if you like).

antipop001, 2023-08-22:
@jbruce12000 @rondoc I am trying to get back into workshop. Been sick. But I will be getting back on this asap.

antipop001, 2023-08-23:
Hello. I found my issue. The thermocouple connection was going up and down. I replaced my MAX module and the issue has been resolved.

closed 2023-08-23 by antipop001

jbruce12000, 2023-08-24:
great to hear. glad you got it fixed.