jbruce12000/kiln-controller #148: Kiln stopped after 6h

whomble, 2023-08-23:
Hi, I started my kiln for the first time in real condition yesterday and after approximately 6h of runtime (21250 seconds) the controller just stopped functioning. I wasn't around when it happened and when I came back the terminal was closed and the power of the kiln was at 100%. I was able to reproduce the issue two times (with the power disconnected).
I have a hardware safety feature so there's no risk but I want to be able to run profiles of more than 6h.
Any idea of what can cause this issue?

jbruce12000, 2023-08-23:
Not sure what happened. Can you post the logs? 

https://github.com/jbruce12000/kiln-controller/blob/master/docs/logs.md shows how to run ziplogs... which will give you a single archive with all the relevant logs in it.

jbruce12000, 2023-08-23:
You have a hw solution in place, but there is also the watcher.py script which can keep tabs on your kiln and ping you in slack if temps don't look right.

see https://github.com/jbruce12000/kiln-controller/blob/master/docs/watcher.md

whomble, 2023-08-24:
I dont know why but the logs generated are empty... I can send the daemon logs if it can help!
I run my kiln at work so even if the watcher notify me that there's a problem i can't do anything until the next day...

jbruce12000, 2023-08-24:
The logs normally go to /var/log/daemon.log which the ziplogs script should grab, but yes, if you have the logs post, zip them up and post them here so I can take a look and see why it stopped.

whomble, 2023-08-25:
I've started the server at 17h15 and it stoped 7200 seconds after! The 4 last times it was 21000s after.
 Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 6 threads of 3 processes of 1 users.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Successfully made thread 18237 of process 18128 owned by '1000' RT at priority 10.
Aug 24 17:12:12 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:14 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:14 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:19 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:19 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:19 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:19 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:22 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:22 raspberrypi rtkit-daemon[719]: Supervising 7 threads of 4 processes of 1 users.
Aug 24 17:12:49 raspberrypi systemd[1]: Reloading.
Aug 24 17:12:50 raspberrypi systemd[1]: /lib/systemd/system/plymouth-start.service:16: Unit configured to use KillMode=none. This is unsafe, as it disables systemd's process lifecycle management for the service. Please update your service to use a safer KillMode=, such as 'mixed' or 'control-group'. Support for KillMode=none is deprecated and will eventually be removed.
Aug 24 17:12:50 raspberrypi systemd[1]: Stopping VNC Server in Service Mode daemon...
Aug 24 17:12:50 raspberrypi systemd[1]: vncserver-x11-serviced.service: Succeeded.
Aug 24 17:12:50 raspberrypi systemd[1]: vncserver-x11-serviced.service: Unit process 677 (vncagent) remains running after unit stopped.
Aug 24 17:12:50 raspberrypi systemd[1]: vncserver-x11-serviced.service: Unit process 910 (vncserverui) remains running after unit stopped.
Aug 24 17:12:50 raspberrypi systemd[1]: vncserver-x11-serviced.service: Unit process 915 (vncserverui) remains running after unit stopped.
Aug 24 17:12:50 raspberrypi systemd[1]: Stopped VNC Server in Service Mode daemon.
Aug 24 17:12:50 raspberrypi systemd[1]: vncserver-x11-serviced.service: Consumed 20.302s CPU time.
Aug 24 17:22:04 raspberrypi rngd[532]: stats: bits received from HRNG source: 800064
Aug 24 17:22:04 raspberrypi rngd[532]: stats: bits sent to kernel pool: 749728
Aug 24 17:22:04 raspberrypi rngd[532]: stats: entropy added to kernel pool: 749728
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2 successes: 40
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2 failures: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Poker: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Runs: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Long run: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: HRNG source speed: (min=347.581; avg=495.925; max=532.956)Kibits/s
Aug 24 17:22:04 raspberrypi rngd[532]: stats: FIPS tests speed: (min=21.773; avg=50.586; max=62.742)Mibits/s
Aug 24 17:22:04 raspberrypi rngd[532]: stats: Lowest ready-buffers level: 2
Aug 24 17:22:04 raspberrypi rngd[532]: stats: Entropy starvations: 0
Aug 24 17:22:04 raspberrypi rngd[532]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Aug 24 18:22:04 raspberrypi rngd[532]: stats: bits received from HRNG source: 820064
Aug 24 18:22:04 raspberrypi rngd[532]: stats: bits sent to kernel pool: 779968
Aug 24 18:22:04 raspberrypi rngd[532]: stats: entropy added to kernel pool: 779968
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2 successes: 41
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2 failures: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Poker: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Runs: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Long run: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: HRNG source speed: (min=347.581; avg=496.729; max=532.956)Kibits/s
Aug 24 18:22:04 raspberrypi rngd[532]: stats: FIPS tests speed: (min=21.773; avg=50.810; max=62.742)Mibits/s
Aug 24 18:22:04 raspberrypi rngd[532]: stats: Lowest ready-buffers level: 2
Aug 24 18:22:04 raspberrypi rngd[532]: stats: Entropy starvations: 0
Aug 24 18:22:04 raspberrypi rngd[532]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Aug 24 19:22:04 raspberrypi rngd[532]: stats: bits received from HRNG source: 860064
[... cut ...]
Aug 25 08:22:04 raspberrypi rngd[532]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Aug 25 08:22:04 raspberrypi rngd[532]: stats: HRNG source speed: (min=347.581; avg=508.354; max=533.816)Kibits/s
Aug 25 08:22:04 raspberrypi rngd[532]: stats: FIPS tests speed: (min=21.773; avg=51.510; max=63.578)Mibits/s
Aug 25 08:22:04 raspberrypi rngd[532]: stats: Lowest ready-buffers level: 2
Aug 25 08:22:04 raspberrypi rngd[532]: stats: Entropy starvations: 0
Aug 25 08:22:04 raspberrypi rngd[532]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us

jbruce12000, 2023-08-25:
I need the logs from the kiln-controller. Each log line that matters will contain one of the following strings...

```
oven
kiln-controller
gevent

```
Maybe you are running the process manually and everything is being written to stdout and stderr?

whomble, 2023-08-28:
Ok sorry. IThe kiln worked for the weekend without any issues by starting the server on boot! 
There's two remaining issues:
- The logs files are about 27Gb this morning so impossible to send anything
- The browser freeze after a while (13h this weekend) so i have to restart it

jbruce12000, 2023-08-28:
27G is a lot of logs. Can you try the ziplogs script again? this will grab just the lines I'm interested in and will compress the logs so they'll be about 10X smaller.

27G worth of logging probably means you have some bad stuff going on. You should examine them closely for problems that you need to fix.

jbruce12000, 2023-09-28:
Did you take a close look at those logs? is everything working as expected? can I close this issue?

jbruce12000, 2023-09-28:
nice. glad you got everything working.

closed 2023-09-28 by jbruce12000