jbruce12000/kiln-controller #149: More Question than Issue about Watcher

antipop001, 2023-08-24:
Hello,
The link the the slack webook instructions is broken. But can you post you json from that. I would like to try to use a discord webhook. It should work similarly. Thanks.

jbruce12000, 2023-08-25:
I think you want the json from the kiln_url endpoint.

you can use curl to get this yourself...

`curl http://127.0.0.1:8081/api/stats`

your ip address and port may be different that what I used above. This is probably why you cannot connect. Maybe you have a different port configured in config.py and maybe the ip address of the kiln-controller is not 192.168.1.84.

This is the json returned [if the kiln-controller.py is currently running a schedule]:

```
{
    "time": 1692989065.0,
    "timeDelta": 2.002047,
    "setpoint": 65.45046575,
    "ispoint": 65.00000153538473,
    "err": 0.4504642146152804,
    "errDelta": 0.2249992330925708,
    "p": 11.26160536538201,
    "i": 0.09018538104744522,
    "d": 44.99984661851416,
    "kp": 25,
    "ki": 10,
    "kd": 200,
    "pid": 56.35163736494362,
    "out": 0.5635163736494362
}
```

closed 2023-08-25 by jbruce12000

antipop001, 2023-08-26:
Thanks. How does watcher behave if the kiln scheduleis idle?

jbruce12000, 2023-08-28:
The json from the endpt returns: {}

The watcher would fail after six attempts trying to make sure the temperature is within range.