jbruce12000/kiln-controller #154: Another question

antipop001, 2023-09-12:
I have my Kiln working!  I have a separate thermocouple just to see the temps quickly and it is about 30 degrees different than what the one on the pi is reading. What are the best ways to verify the temp reading and adjust  your thermocouple?

chipgarner, 2023-09-13:
You need to start with a known temperature. Is this happening at room temperature? If so one of them will be the obvious wonky one.  30 degrees C or F?

Make sure your TCs are connected correctly and you are using TC wire. The two wires are two different metals and have to be connected the right way. There are actually four possible ways to connect the wires, only one of which will work right. (You can try all four without damaging anything.)

Congrats on getting it running.

MeltTechAu, 2023-09-13:
To achieve correct ambient you must ground out the positive thermocouple
wire to ground on the pi  just a tip with max31855 now that’s sorted it
will be spot on within 1deg c

antipop001, 2023-09-13:
> To achieve correct ambient you must ground out the positive thermocouple wire to ground on the pi just a tip with max31855 now that’s sorted it will be spot on within 1deg c
> […](#)

So I can just attach a ground from the pi to the same terminal as the positive wire on the max?

rondoc, 2023-09-13:
The thermocouple wires must be floating NOT connected to ground or +. Some
people have had problems with accidental grounding.
You DO NOT need separate TCS you can monitor the same ones with different
meters or W.H.Y.  I monitor the pi readings with a Fluke meter and tracking
is within a few degrees over room temp to 1300C.
Ron.

antipop001, 2023-09-13:
ok so my readings are about 10F off.   From the output below you can see that I am not seeing a lot of noise.  The internal temp is very close to correct so the temps should be very close to each other.
Thermocouple Temperature: 18.750*C / 65.750*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.250*C / 66.650*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.500*C / 67.100*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.250*C / 66.650*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.250*C / 66.650*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.250*C / 66.650*F
    Internal Temperature: 25.562*C / 78.013*F
Thermocouple Temperature: 19.250*C / 66.650*F
    Internal Temperature: 25.562*C / 78.013*F

chipgarner, 2023-09-13:
This looks like the output from the thermocouple test program(?) If so, the internal temperature is the temperature of the sensor board where the thermocouple wires are connected and the voltage is read. This will usually be higher than the TC temperature because of heat from the electronics. The sensor board needs to know the temperature at the end of the TC wires to compute the correct temperature at the TC

MeltTechAu, 2023-09-13:
With the max31855 I ground out the positive thermocouple lead to ground on
the pi this gets true ambient then tested with my thermocouple calibrator
max chip is only one deg c out compared to a Bartlett or tap kiln
controllers they are usually 5 to 6 deg c out
Hope this helps
Have a great day!

antipop001, 2023-09-14:
Grounding the pi ground to thermocouple lead on max just caused 90 9

rondoc, 2023-09-15:
Please look at the instructions again..the thermocouple is floating NOT
connected to anything but the input terminals of the board. By accident or
poor insulation some users end up with tc leads shorting to ground then
results are unknown as often it is not a solid short making diagnosis
harder.
Ronald.

rondoc, 2023-09-15:
If you look up the max31855 chip.info from the maker you will see it is
designed for fully floating input. It has a built in FAULT
detector system for shorts from the tc to either gnd. or + 5. Please,
please don't make life more difficult for yourself by trying unproven ideas
and expecting good results...
Ronald.

MeltTechAu, 2023-09-16:
H all
some controllers ground out the thermocouples to get a true ambient and
scale also for safety
 i had to do this and i have heaps of spare parts bought over the years so
i cant make my max chips work at all with jbruces software i had to modify
the driver take out some of the error handling to make it work lol no 2
chips are the same lol something going on here i cant with blinka branch
get 31855 or 31856 to work either where you say it does but i doent work i
tried spi 8 9 10 11 no hope lol and i spent years playing around with this
software hurts my head sometimes
i also put .01 and 10nf caps on the thermocouple solder 1 to the
positive and the negative on the max31855 and also across negative and
positive on the thermo side with th 10nf yellow cap then all becomes good
for accuracy.
hope this helps
i would love to get the blinka branch going just can not make it read the
temp with either chip? sending me nuts the old software works brilliant
with max31855 but seems to get no connection faults with max31856 which i
tried multiple time with multiple rpi's
any how guys talk soon busy looking into cone mode and a few things ,the
home people like the thought of cone mode but it was never right usually
under fires customers tell me ye

 i box up bartletts in my day job and night job i sell controllers we sell
a lot of them because of cone mode though
ill keep you up dated if i get anywhere should nt be too hard they did 30
years ago with old tech here in australia the famous ausse controller its
in every school etc
it is patented by bartlett as per there manual but there are ways around
that  gatherm!!

i have sold kiln controllers for a lot of years so i sort got some clue
just not a programmer but can sort of program
anyhow can we first get a timer put on catch up mode in older software
please as blinka no go for me yet
been great watching this awesome project grow over the past 5 years i been
here yeah
 i have looked at most other people coding on here some awsome ideas each
and every oen of you have.
cheers
Brett
CEO
Melttech

rondoc, 2023-09-16:
If you follow the information given you will find the app works fine. I am
an EE and also creative but when someone has taken a long time to work out
a great controller with a fantastic app why not just use it as given?
Controlling a kiln is not that complex but there are some things that I am
sure you understand.
1. The thermocouple used is the type that the app addresses . I use a K
type as they are fine upto cone 10, 1350C. The app allows for other
thermocouples but there is no need for this for hobby potteries. ,( I did
fire at cone 20 for some special materials and used an optical thermometer.)
2. If you are using the PID control equation you do need to have optimized
the values of the constants. The pi controller information includes how to
get the best numbers.  I did it and get tracking within a degree upto
1350C. At the start there will be some oscillations but all smooth out at
about 200C
3. When setting up the temp time slopes choose values that your kiln can
do. Also I have a dwell time at max temp so large pieces reach the same
temp as small ones.
( BTW I built my own kiln and designed a digital controller 35 years ago
that all still works fine. I especially like that now I can monitor my kiln
with my cell phone .)

I appreciate your efforts to "improve" on Jason's efforts but please first
try the app as it is.
Ronald

antipop001, 2023-09-21:
Hello. I belive I have bad chip as a borrwed chip didn't have issues.

closed 2023-09-21 by antipop001