jbruce12000/kiln-controller #157: 31856 config

antipop001, 2023-09-22:
Hello. I am switihcing from a MAX31855 to a MAX31856. (because my 5 is kaput, and I have a 6) Just making sure these are all the changes needed (other than connectiing the other wire physically) 

``### Thermocouple Adapter selection:
#   max31855 - bitbang SPI interface
#   max31856 - bitbang SPI interface. must specify thermocouple_type.
max31855 = 0
max31856 = 1
# see lib/max31856.py for other thermocouple_type, only applies to max31856
# uncomment this if using MAX-31856
thermocouple_type = MAX31856.MAX31856_S_TYPE

### Thermocouple Connection (using bitbang interfaces)
gpio_sensor_cs = 27
gpio_sensor_clock = 22
gpio_sensor_data = 17
gpio_sensor_di = 10 # only used with max31856``

Brett308, 2023-09-22:
i get this error i n  the old one and blinka ive tried 4 boards but software wont load in 31856 anyone got a clue im only uncommenting thermo type and import its sending me nuts why is config not importable is there a fix also put 0 max31855 and a 1 on 31856 and ther eis no other issues when in 31855
>>> %Run kiln-controller.py
Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup.

Process ended with exit code 1.

jbruce12000, 2023-09-22:
Hi @Brett308. Can you post config.py so I can check it out?

jbruce12000, 2023-09-22:
@Brett308 try this to see the error in config.py:

`source venv/bin/activate
python ./config.py
`
That should give you the exact error instead of a generic... could not import error.

Brett308, 2023-09-22:
> @Brett308 try this to see the error in config.py:
> 
> ```
> source venv/bin/activate
> python ./config.py
> ```
> 
> That should give you the exact error instead of a generic... could not import error.

dosn't give m anything

jbruce12000, 2023-09-22:
At the bottom of this comment window is an area that says attach files by dragging and dropping. I need you to drag-and-drop config.py from a folder into that area...

Brett308, 2023-09-22:
here it is jason
cheers for your time
brett

jbruce12000, 2023-09-22:
ok, it says that it doesn't support files with that extension. copy it to another file...

  cp config.py config.py.txt

then drag/drop config.py.txt to the area below the box you're typing in... like this...
[config.py.txt](https://github.com/jbruce12000/kiln-controller/files/12703195/config.py.txt)

Brett308, 2023-09-22:
import logging
import os

# uncomment below if using MAX-31856
from lib.max31856 import MAX31856

########################################################################
#
#   General options

kiln_name = "MeltTech"
function_passcode = "0000"

## Logging
log_level = logging.INFO
log_format = '%(asctime)s %(levelname)s %(name)s: %(message)s'

### Server
listening_ip = "0.0.0.0"
listening_port = 8081
########################################################################
# Cost Information
#
# This is used to calculate a cost estimate before a run. It's also used
# to produce the actual cost during a run. My kiln has three
# elements that when my switches are set to high, consume 4800 watts.

kwh_rate        = 0.00030 # = 30 cents Australian cost per kilowatt hour per currency_type to calculate cost to run job
kw_elements     = 2400 # if the kiln elements are on, the wattage in kilowatts
currency_type   = "$"   # Currency Symbol to show when calculating cost to run job
### Cost Estimate
oven_kw         = 00000   # Watt capacity of oven we now have actual costs not correct kw??????

########################################################################
#
#   PID parameters
#
# These parameters control kiln temperature change. These settings work
# well with the simulated oven. You must tune them to work well with
# your specific kiln. Note that the integral pid_ki is
# inverted so that a smaller number means more integral action.

# If you have oscillations that don't stop or increase in size, reduce pid_kp
# If you have an oscillation but the temperature is mostly below the setpoint, decrease pid_ki.

pid_kp = 40   # Proportional was 40
pid_ki = 200  # Integral was 200
pid_kd = 800  # Derivative was 800

####################################################################
# emergency shutoff the profile if this temp is reached or exceeded.
# This just shuts off the profile. melttech duel safety switches means 2 switches need to fail closed 
#which relay switchs usually fail open 90 % of the time so below feature is an extra procaution on top of meltech hardware

emergency_shutoff_temp = 1316

########################################################################
#
#   Time and Temperature parameters
#
# If you change the temp_scale, all settings in this file are assumed to
# be in that scale.

temp_scale          = "c" # c = Celsius | f = Fahrenheit - Unit to display
time_scale_slope    = "m" # s = Seconds | m = Minutes | h = Hours - Slope displayed in temp_scale per time_scale_slope
time_scale_profile  = "m" # s = Seconds | m = Minutes | h = Hours - Enter and view target time in time_scale_profile

#####################################################################
# If the kiln cannot heat or cool fast enough and is off by more than
# pid_control_window the entire schedule is shifted until
[... 136 lines cut ...]
#   These were tested on a Pi B Rev2 but of course you
#   can use whichever GPIO you prefer/have available.

######################################################################
#
#   Simulation parameters
simulate = False
sim_t_env      = 60.0   # deg C
sim_c_heat     = 500.0  # J/K  heat capacity of heat element
sim_c_oven     = 5000.0 # J/K  heat capacity of oven
sim_p_heat     = 5450.0 # W    heating power of oven
sim_R_o_nocool = 0.5   # K/W  thermal resistance oven -> environment
sim_R_o_cool   = 0.05   # K/W  " with cooling
sim_R_ho_noair = 0.1    # K/W  thermal resistance heat element -> oven
sim_R_ho_air   = 0.05   # K/W  " with internal air circulation

#######################################################

########################################################################
# load kiln profiles from this directory below or download from melttech website under firing profiles
# created a repo where anyone can contribute profiles. The objective is
# to load profiles from this repository by default.
# See https://github.com/jbruce12000/kiln-profiles
kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),"storage", "profiles")) 
#kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),'..','kiln-profiles','pottery'))

jbruce12000, 2023-09-22:
cool. thx. checking on config...

jbruce12000, 2023-09-22:
confirm the branch you are using... is it blinka?

MeltTechAu, 2023-09-22:
One before but blinks did same

jbruce12000, 2023-09-22:
ok, I need a branch. type

`cd kiln_controller`

`git branch`

post the output of git branch. next type

`git log`

and post the first few lines of output.

jbruce12000, 2023-09-22:
I can say with certainty that this config file will not work with blinka. There are imports missing at the top.

Brett308, 2023-09-22:
commit 8af50334f764e54bc0d1408f2cfb8ede36f93bfa (HEAD -> master,
origin/master, origin/HEAD)
Author: MeltTechAu ***@***.***>
Date:   Sat Apr 15 19:28:14 2023 +1000

    2023

commit a59f3742c7544202b5995bffa0cca6667d372435
Author: MeltTechAu ***@***.***>
Date:   Sat Apr 15 19:23:41 2023 +1000

    Update README.md

commit b2329d2eb2143574ca91aba84f4898d6df403fc8
Author: MeltTechAu ***@***.***>
Date:   Fri Apr 14 22:50:36 2023 +1000

    spi gpio

commit d9115b9b0cafce7a9ab72815f4900e83a7de9f89
Author: MeltTechAu ***@***.***>
Date:   Wed Apr 12 22:36:57 2023 +1000

jbruce12000, 2023-09-22:
This is MeltTechAu's fork of the kiln_controller. It's got lots of modifications. You'll need to ask her/him for help and open an issue...

https://github.com/MeltTechAu/kiln-controller

I did download the head of master and use your config file. It worked fine with **simulation=True** set, but I don't have a PI sitting around [and I've never had a max-31856], so I cannot test it live.

Jason

MeltTechAu, 2023-09-22:
Thanks Jason for your help have a great weekend

antipop001, 2023-09-22:
I have having this issue (config attached):
Traceback (most recent call last):
  File "/home/ahermon/kiln-controller/./kiln-controller.py", line 43, in <module>
    oven = RealOven()
  File "/home/ahermon/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
  File "/home/ahermon/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/ahermon/kiln-controller/lib/oven.py", line 95, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/ahermon/kiln-controller/lib/oven.py", line 140, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
AttributeError: module 'config' has no attribute 'thermocouple_type'
[config.txt](https://github.com/jbruce12000/kiln-controller/files/12704522/config.txt)

rondoc, 2023-09-22:
I don't see any reason to use the 56 chip since most potters use K type TC
s. It just seems trying to resolve problems that don't need to exist is a
futile effort.
The 55 chip on the right board works fine. I have several and no problems.
The idea of choosing the tc should IMNSHO be taken out of the basic app.
Less choices makes for less challenges especially for potters who are only
interested in getting a great kiln controller..which thanks to J we have.
Ronald

chipgarner, 2023-09-23:
@antipop001 You need to uncomment line 48 that loads the thermocouple_type

antipop001, 2023-09-23:
> @antipop001 You need to uncomment line 48 that loads the thermocouple_type

With that change I get the following:

Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup.
[config.txt](https://github.com/jbruce12000/kiln-controller/files/12707282/config.txt)

chipgarner, 2023-09-25:
@antipop001 I am stumped.  Changing a line in the config.py file should not make the code unable to read the same file unless the name of the file changed or it got moved somehow.
Any luck?

antipop001, 2023-09-25:
No. I am very confused.

chipgarner, 2023-09-25:
Make sure you have the config.py file in the same directory as kiln-controller.py. That's all I can think of. I have had small typos give me errors like this.

jbruce12000, 2023-09-26:
Start with a file that you know works. Change one line at a time, start the kiln-controller, until you arrive at the config you need.

antipop001, 2023-09-28:
Went back to a 55

closed 2023-09-28 by antipop001