jbruce12000/kiln-controller #158: Blinka auto tune no longer works in the new bookworm RPIOS Help needed

MeltTechAu, 2023-10-24:
pi@raspberrypi:~/kiln-controller $ python kiln-tuner.py zn zn.csv
Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup.
 
kiln controller runs but only through terminal

old software now wont work with bottle either

is there any other way to make tuning work again

auto boot working :) thonny complains about bottle which old software wont install on new os bottle issue 
all i have done is new OS downloaded blinka installed and tuning gives above error

jbruce12000, 2023-10-25:
remove the try/except section at the top of the kiln-tuner.py so it looks like this...

```
#!/usr/bin/env python

import os
import sys
import csv
import time
import argparse

sys.dont_write_bytecode = True
import config
sys.dont_write_bytecode = False

```

So that you can see the real error. Post the error here.

MeltTechAu, 2023-10-26:
I don’t see that in blinka that I downloaded I see that in your original
code which does work in new is  but not in blinka tuner hmmm

MeltTechAu, 2023-10-26:
Sorry I was clicked on the wrong thing the logger lol I will try that asap

MeltTechAu, 2023-10-29:
> pi@raspberrypi:~/kiln-controller $ python kiln-tuner.py zn zn.csv Could not import config file. Copy config.py.EXAMPLE to config.py and adapt it for your setup.
>
> kiln controller runs but only [...]

python kiln-tuner.py recordprofile zn.csv
usage: kiln-tuner.py [-h] [-c] [-t TARGET_TEMP] [-d TANGENT_DIVISOR] [-s]
kiln-tuner.py: error: unrecognized arguments: recordprofile zn.csv 
im getting that in bullseye as well havent tried bookworm yet seems no go in bullseye too i cn not seem to get it going anyone else getting this

MeltTechAu, 2023-11-03:
Error is no module named digitalio when doing tune after removing those
lines hmmm that was in bookworm this should be part of adafruit I believe
should be installed with drivers I tried to install again no luck Jason I
have never got it to work on tuning test no matter what I do on blinka also
seems to be a bit floaty on temperature aswell hope this helps I get that
same error on one of my mods I did a safety switch which I had working on a
disk but when I now install it gives me pin error which lead me to same
problem digitalio has something changed since there new upgrade and updates
I think it’s part of adafruit isn’t it currently I’m moving to your old
software and trying to add new features ,as software I been using errors on
bookworm so officially will be using yours mate as it stable from tests,
hope we get blinka sorted that’s where I wanted to get to

Try reinstalling says already satisfied

anyone else having same issues
Sorry to be a hassle
Again thankyou for your valued time and major work you have done so far :)
Tia Brett

MeltTechAu, 2023-11-03:
Also cannot run any of your software anymore on bookworm has bottle error
aswell when run as a real kiln I’m at a lose tried reinstalling about 20
time none of the software works anymore can anyone confirm this I’m sorta
stuffed without it software only works on bullseye what has raspberry pi
done  everything in book work no go only works under simulator mode only
pip installing says can’t find a better match bottle installs but thonny
says bottle error and terminal says real kiln issue with real kiln hmmm
this is not good I didn’t expect rpi to stuff this right up for us hope
someone can fix this I’m lost works on bullseye but updates will stop
eventually

chipgarner, 2023-11-05:
I have the master branch code running on the new Raspberry Pi Bookworm OS. Here is what I did.

Create a fresh install on a flash card using the Raspberry Pi Imager. I did not try upgrading from Bullseye to Bookworm as it is not recommended. I had a little trouble getting it to work with SSH but I was not using the latest version of Imager.

SSH into the pi.

    sudo apt update
    sudo apt ugrade

Clone the repository:
    git clone https://github.com/jbruce12000/kiln-controller.git

(Note that pip install does not work on Bookworm unless you are working in a virtual environment.  Bookworm comes with Python 3.11.2 and does not have the ancient Python 2.)
Create a virtual environment with access to the system packages:
    cd kiln-controller
    python -m venv --system-site-packages env

Activate the environment and install the requirements:
    source env/bin/activate
    pip install -r requirements.txt

The program should now run the simulator. Edit the config.py file as needed to run your kiln.

I tested this on a Raspberry Pi 3B and checked that it reads my kiln temperature and switches the relay, I did not do a firing.

chipgarner, 2023-11-05:
kiln-tuner.py also runs, reads my thermocouple and switches the relay.

chipgarner, 2023-11-05:
I tried the above on the blinka branch. The relay switches but I did not get the thermocouple working right - I think my pins are not compatible.

MeltTechAu, 2023-11-07:
Hi all I can only make it run once after installation after that I get bottle error still it won’t auto boot and load either anyone got past the working only once , I will keep trying,that was with the 32 bit version so far hmmmmm anyone or do we need to get it working on Ubuntu or another flavour can anyone confirm at all so frustrating. JBruce have you got any idea  I had upgraded rpi imager as well to latest something seriously wrong with getting it running

chipgarner, 2023-11-08:
I have only tested with a clean install of the recommended Raspberry Pi Desktop version, which is 64 bit. I am not having any problems with the master branch or the blinka branch. My multi zone software is also working and uses bottle. (See https://github.com/chipgarner/MultiZoneKilnController)

MeltTechAu, 2023-11-08:
Hi chip
Any chance you could make a zip file of the disk 32gb and put it online
with a link for download say in drop box for me that would be highly
appreciated it just won’t play nice if you have time I’m using a pi 4 also
could you install your 3 zone for me to try as well as jbruces could get
yours  working either happy to pay for your time I really keen to build a 3
zone aswell
Thanks mate
Have a great day
Brett

MeltTechAu, 2023-11-08:
Hey chip it runs only through ssh from my pc trying to do it on the pi stil says bottle error I can start it from my pc not a problem works only once as well then bottle error I really don’t get why it’s not working the os must be faulty I really need this working with a pi with a screen this ones got me confused anyone can they confirm

chipgarner, 2023-11-08:
I am able to reproduce the bottle error by trying to run the program without activating the virtual environment. This would also explain why the program wouldn't run with auto boot. Sorry for assuming everyone knows how to use a python virtual environment. You need to type the following to activate:

source env/bin/activate

Your terminal prompt will show something like (env) jkg@pi: instead of jkg@pi: to show you are in the environment. The programs will now run normally.

All the software packages installed in a virtual environment (by pip install ...) are only available with the environment activated. See https://docs.python.org/3/library/venv.html for more.

I have only tested with a clean install of the recommended Raspberry Pi Desktop version, which is 64 bit. My multi zone software is also working with Bookworm. (See https://github.com/chipgarner/MultiZoneKilnController)

MeltTechAu, 2023-11-09:
> PS you need to be in the program directory to activate the environment.

I have simulator auto booting now I get this error on real kiln I tried
installing rpi gpio module says already satisfied hmmm I try tonight after
I get off work
Dunno what’s going on thanks chip I’ll let you know how I go I am using as
what you said hmmm looks like I’ll be up all night if vnc worked on it I
would let you show me but vnc don’t even work on it
Again thank you for your valued time
Will get there in the end I hope
Cheers
Brett

MeltTechAu, 2023-11-09:
So after I still could not make it work I now have it running I reflashed
the pi and jbruces now works straight up like WTF lol why so many different
errors anyhow happy to pay for your time so a pi4 with older firmware is
the issue send me details I can send you some beer money I now need a beer
that was stressing me out nothing was consistent I still get bottle error
in thonny for some reason so will have to program on the older bullseye
Thanks so much mate
Cheers Brett
Ps send me your bank details for payment

I’ll try yours tonight yes I’m still at my day job so I better do some work
lol

closed 2023-11-11 by MeltTechAu