jbruce12000/kiln-controller #159: SSR working in reverse

salperta, 2023-11-06:
Hi All,
Thanks for this wonderful project. I'm getting mine set up, and have "successfully" gotten it to work in testing with one major flaw--my SSR is running in reverse. That is, it is closed by default when hooked up to the pi, and open when it should be heating while running a schedule.

I've set it up following carefully the instructions from the main github page. I'm sure I've just done something wrong, but I wonder if anyone has insight--I'm banging my head against a wall here.

Thanks kindly for any advice.

captaincaden, 2023-11-13:
Can you include the model of the SSR you used? 
Also, did you build a transistor circuit (optional step in the instructions) to switch the relay, or are you using it straight from one of the GPIO pins? 

If you can confirm that the output of the pi is behaving as expected, than the issue lies somewhere either in the transistor circuit if you used one, or in the SSR. Either that it's configured to behave that way, or it has a problem.

rondoc, 2023-11-13:
It just needs a single transistor to reverse the function. Almost any
transistor will work. I can send you the very simple circuit
diagram if needed. It's probably easy to do in the code but a transistor
inverter is useful as it isolates the SSR from the pi. Not complete
isolation but enough to prevent a blown pi from a shorted SSR.
Ron.

jbruce12000, 2023-11-20:
this has never happened before. If it's truly backwards, then you can change two lines of code to flip it.

See lib/oven.py lines 45 and 50. Change those to the opposite they are now.

jbruce12000, 2023-11-20:
This could be easily made into a setting, but this is the first time I've seen it.

We-Sing, 2024-02-24:
:-) I have the same reverse problem with a SSR 40-DA

rondoc, 2024-02-24:
You might need a transistor inverter as the polarity to drive your SSR may
be inverted.
This 'problem' has been mentioned before.
Ronald.

We-Sing, 2024-02-25:
Hi Ronald,

I fixed it in oven.py. I had it before with an other project. Maybe I bought the wrong SSR's. But evrything works fine. Thanks for the  solutions! Jan

jbruce12000, 2024-06-27:
fixed by changing a line in oven.py by requestor.

closed 2024-06-27 by jbruce12000