jbruce12000/kiln-controller #164: High Volume Network and CPU consumption due to continuous sending of config from backend to web.

chriswhsu, 2023-12-03:
Problematic Code:

@app.route('/config')
def handle_config():
    wsock = get_websocket_from_request()
    log.info("websocket (config) opened")
    while True:
        try:
            message = wsock.receive()
            wsock.send(get_config())
        except WebSocketError:
            break
    log.info("websocket (config) closed")

I changed to the following which just sends config once; could probably stand to add in some error handling to retry after a wait if initial attempt fails.

@app.route('/config')
def handle_config():
    wsock = get_websocket_from_request()
    log.info("websocket (config) opened")
    try:
        wsock.send(get_config())
    except WebSocketError:
        log.error("Error with Websocket in Config")
    log.info("websocket (config) closed")

jbruce12000, 2023-12-04:
high cpu, yes. 

but, high network consumption is not possible if the websocket is not connected... right?

Not saying this should not be fixed... and it's certainly a bug. I think the best way to handle this is just to put a sleep statement in the current code so that the cpu does not spin.

This impacts config, storage, status, and control routes... because they all work the same way. I think a sleep needs to be put into place for each.

jbruce12000, 2023-12-04:
changes committed to the blinka branch for this. blinka will become master before the end of the year.