jbruce12000/kiln-controller #165: Max31856 thresholds (not a issue)

Linkeor, 2023-12-11:
A little more reading here to avoid polluting the other thread on blinka migration...
I played again with the max31856 thresholds...
I'm using the script https://github.com/adafruit/Adafruit_CircuitPython_MAX31856/tree/main/examples that I'm modifying and I was able to draw some interesting conclusions.

- If no threshold is set, I reread the thresholds set previously.
- If we do not set any threshold after electrically disconnecting the max31856, the values read are "extreme"

```
Time: 21.82 :: cj: 22.81
Thresholds:
TC low: -2048.00 high: 2047.90
CJ low: -64.00 high: 127.00
```

- If we change the type of thermocouple, the values do not change.
Nothing is therefore fixed in the chip according to the thermocouple other than -2048 -> 2047.90 and -64 -> 127.

- The problems that I encountered initially therefore came from the adafruit_bitbangio library and from the fact that I always launched max31856_thresholds_and_faults.py mentioned above before any other .py in the project. Incorrectly set thresholds therefore returned faults and naturally kiln-controller.py too.
Not setting the thresholds of the 56 is therefore not a problem if it is "electrically" reset and uses the "very wide" default values to not interfere and report faults.

However, I tell myself that if these thresholds exist, it could be good to exploit them and I see two ways.
- Simply set the thresholds according to the oven, the thermocouple used and the environment of the cold junction, but is this not redundant with what is already in place software for the 55?
- Set thresholds throughout the cooking process close to the desired curve to overcome any possible drift. I tell myself that this hypothesis can be interesting in the case where several groups of resistors and thermocouples are implemented and interact together (which is not the case in the configuration of the initial project of course)

chipgarner, 2023-12-13:
This is mostly useless but I did manage to get very strange range settings, like (-1071, 84) by having a wrong pin setting. I believe I had the CS pin connected to a '55, which was also running at the time. Setting the ranges resulted in different, also strange settings. Getting the pins right fixed this.

jbruce12000, 2024-07-05:
there has been no activity on this. there does not seem to be any action to take. closing this.

closed 2024-07-05 by jbruce12000