jbruce12000/kiln-controller #166: Not an issue, PID tuning frequency

captaincaden, 2024-01-19:
I don't think I've seen it asked, but I'm just curious if there's a recommendation for how often or when a kiln running this project should look into retuning the PID variables in config.py 

For example, I remember running the auto tuner when I first started the project, but would this information ever need to be 'retuned'? 

I recently changed the configuration of my kiln, which results in profiles up to cone 6 that I've tested so far being less than a half a degree of error the whole firing now, which is significantly better at the higher temps than what I was getting with the old configuration + old elements. 

During my tests, using the old PID data that I got from the autotuner and that I think @jbruce12000 and I I think determined would be good for my kiln a few years ago, the new element configuration fired fantastically with the profiles I've tested so far, but it just got me wondering what circumstances would make you look into or change the PID values in config.py 

This question comes from someone who has extremely limited, very basic knowledge of how PID works.

rondoc, 2024-01-19:
> First thing is that the PID is only used inside the PID window... which is
> a total of 10 degrees by default. This window means you can be a lot more
> aggressive with PID configuration than you would if the window did not
> exist.
>
> Second, if you're happy with the performance of the kiln, don't change the
> PID config.
>
> If you find that your PID settings are constantly causing temperatures to
> go outside the PID window, then it's time to fix them. If there is constant
> oscillation of temperatures that does not settle to nearly static [within a
> degree or two], it's time to fix them.
>
> Precision of type K thermocouples sucks. Mine jumps in 1/2 to 3/4 degree F
> increments. This means that my PID will never settle completely and that
> control will never be better than 1/2 to 3/4 of a degree. This level of
> control is still incredible and the PID can do a great job of reacting to
> the jumpiness.

K type thermocouples do not suck! The
Thermoelectric output is dependent on dissimilar metals and is a basic
physical phenomona
What might suck is the system used to convert this tiny electrical current
to a usable digital signal. The voltages are very small and other
electrical currents can affect the resultant data. This is called noise!
The combination of small electrical signals from the K and the thousands of
times larger kiln element currents is a challenge.
The PID equation works well for most industrial controls. However it is not
necessarily the best for a kiln. The reason I say this is because kiln
firing is unidirectional , you go from low to high temperatures. PID
controls are mostly used for set point control where a fixed temp , water
flow etc is required.
My old control with a finely stepped heat setting works extremely well and
is logicaly simpler than the pi PID system.  Now I have the pi system ,
thanks Jason et al. , that works great and has monitoring down well and
when I want I can switch to my old stepped one. ( the electronics in the
stepped one are over 40 years old!)
Ronald.

captaincaden, 2024-01-23:
Thanks for the replies! I figured keeping it where it was was the best option if I was happy with how it was firing, but was just curious on the frequency of retuning.

closed 2024-01-23 by captaincaden