jbruce12000/kiln-controller #167: Changing from max31855 to max31856 causes error in config.py file

artcank, 2024-01-22:
I have had this working before but switched to a RaspberryPi 4 with a touchscreen to use on my kiln. 

Using the updated setup procedure everything works as it should for the default test setting.

When I change the config file to use my type-s thermocouple and specify the max31856 I get an error when running ./kiln-controller.py

```
could not import config file.
copy config.py.EXAMPLE to config.py and adapt it to your setup.
```

artcank, 2024-01-24:
Never mind, I'm an idiot and and didn't uncomment the lib.max31856 at the top of the config file!

However when I change it from simulation = True to simulation = False, I get this error:
`
Traceback (most recent call last):
  File "./kiln-controller.py", line 43, in <module>
    oven = RealOven()
  File "/home/pi/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
  File "/home/pi/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/pi/kiln-controller/lib/oven.py", line 95, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
  File "/home/pi/kiln-controller/lib/oven.py", line 143, in __init__
    ac_freq_50hz = config.ac_freq_50hz,
  File "/home/pi/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
  File "/home/pi/kiln-controller/venv/lib/python3.7/site-packages/Adafruit_GPIO/GPIO.py", line 420, in get_platform_gpio
    import Adafruit_BBIO.GPIO
ModuleNotFoundError: No module named 'Adafruit_BBIO'
`
I'm going to see if I can resolve this but just wanted to update for the facepalm!

jbruce12000, 2024-01-27:
glad you got it fixed. closing.

closed 2024-01-27 by jbruce12000