jbruce12000/kiln-controller #169: Moving from Master branch to Blinka

artcank, 2024-02-02:
Having trouble migrating to the blinka branch. I'm doing this because I am having trouble with the master branch stopping a run  prematurely. I am on a RasPi4 with a max31856 that was working a year ago on a RasPi3.

Using this to document to migration from the master branch:

Following the text doc [https://github.com/jbruce12000/kiln-controller/blob/blinka/docs/old-to-new.md](url)

I wasn't sure how to merge to the new blinka branch. After some research I found after I ran:

`git checkout master`

I had to run:
`git stash save`
`git pull origin blinka`

I also had to run these:
`git config --global user.email "you@example.com"` 

`git config --global user.name "Your Name"`
Entering my email and name.

I'll update later if things are working.

artcank, 2024-02-02:
After changing the config.py file to match my GPIO pins, the tests for temperature reading and sending heat signal to SSR were good.

However, attempting to run kiln-tuner.py resulted in thermocouple errors:
cold junction range fault, cold junction temp too high, and temp thermocouple range faults.

Trying to run a firing schedule results in the run stopping as soon as it starts.

artcank, 2024-02-03:
Working on figuring out why it is not running.
I tried to install kiln-stats and when installing the requirements file I got a hash error for statsmodels:

`scipy>=1.3 from https://www.piwheels.org/simple/scipy/scipy-1.7.3-cp37-cp37m-linux_armv7l.whl#sha25                            6=2fec5b97046d51212017a3246996fb3bc99fe3225949e3b6e1dbb1e1be0eb57c (from statsmodels->-r requirements.t                            xt (line 6)):
        Expected sha256 2fec5b97046d51212017a3246996fb3bc99fe3225949e3b6e1dbb1e1be0eb57c
             Got        215a64e2b3d49bc9d209415c1efe62ab9d54ca14eb6f0a4b98597755d7a71b37`

I installed it from the github page with:

`python -m pip install -i https://pypi.anaconda.org/scientific-python-nightly-wheels/simple statsmodels --upgrade --use-deprecated=legacy-resolver`

I was able to finish the install for kiln-stats after that.

artcank, 2024-02-03:
I tried another run and used the fast-test run instead of one I had created.... It ran for a while but got hung up around 260°. 
I ran  
`  ./test-thermocouple.py`
in another putty window and at first the temp read the as the WebApp but in a few seconds it jumped up to over 700° while the WebApp still showed 260°. 

I decided to shut the PI down and work on it this morning. Now the Pi won't boot and I am going to have to troubleshoot that in order to get the Log file.

chipgarner, 2024-02-04:
@artcank  I don't think reading the max31856 from two programs at the same time will work very well.
Set log_level = logging.DEBUG near the top of the config.py if you haven't already. Are you running kiln-controller in a putty terminal? If so you should see errors if it hangs up. 
Make sure everything works in the simulator first, this often makes it easier to find problems.
You may have already tried all this, these are just some things I do when there is an issue. Good luck with the boot problem!

jbruce12000, 2024-02-04:
I don't think kiln-stats will help you troubleshoot the problem. It shouldn't hurt anything to have it installed, but I would focus elsewhere.

You may have done this already, but start with the latest config.py on the blinka branch and modify that to match your specific pins. Don't use your old config.py file.

Don't use test-thermocouple to get the temperature while the kiln-controller is running. You can either look at the logs or do the following while sitting in the kiln-controller directory:

`watch -n 2 'cat state.json'`

That'll display all vital stats every two seconds. Focus on the logs if you see issues. What was the error? Post them here, we can help.

The fact that your PI wont boot is the biggest issue. Maybe you have SD card issues or other hardware issues. You'll need to look at the logs to diagnose that:
`
dmesg >logs.txt`

and take a look at logs.txt.

artcank, 2024-02-04:
So my boot problem is somewhere in the buck converters I had. I have a 19v power supply that I split into 3 5v inputs for the Rpi, 7" screen, and the touch screen input. 

I'm just powering the Rpi with a separate 5v power supply for now and working over Putty.

I am currently running a ramp up test. 
[image]

I also had the watch gpioreadall.py running and had some questions about what I was seeing.

[image]

The GPIO4 keeps changing from 0 to 1 with Input even when the kiln-controller is not heating and sometimes it switched to OUT - 0.

GPIO17 also changes from 0 to 1 when the Kiln is not heating.

I just want to make sure that is normal.

artcank, 2024-02-04:
@jbruce12000 Yes, I am using new config.py for the setup.

Here are the Logs

[kiln.logs.gz](https://github.com/jbruce12000/kiln-controller/files/14157883/kiln.logs.gz)

I saw lots of errors but I put ignore all errors to true so it would try to keep running.

Do I need to be in a Virtual Environment in Kiln-Controller every time I run a script in the repository?

@jbruce12000, I like this: `watch -n 2 'cat state.json'`
Thanks

jbruce12000, 2024-02-05:
I don't think you should run gpioreadall.py when the kiln-controller.py process is running. I think they would conflict. I have not examined the underlying libs, but that would be my guess. gpioreadall.py was intended to troubleshoot during initial set up only.

Do I need a virtualenv? Well, if you plan to run the process from the command line every time, yes. I would not do this though. I'd set mine up to autostart when the pi powers on... once you get things stable there are instructions for doing this on the main docs page.

jbruce12000, 2024-02-06:
OK, you have a bunch of range faults. I think there is a fix you need to make to one of the libraries in the virtualenv on a specific line. Can someone with MAX-31856 experience point artcank to the correct fix?

```
14 Feb  2 20:22:13 raspberrypi python[419]: 2024-02-02 20:22:13,776 ERROR oven: Problem reading temp (ignored) thermocouple range fault
14 Feb  2 20:22:14 raspberrypi python[419]: 2024-02-02 20:22:14,012 ERROR oven: Problem reading temp (ignored) cold junction range fault
14 Feb  2 20:22:14 raspberrypi python[419]: 2024-02-02 20:22:14,243 ERROR oven: Problem reading temp (ignored) thermocouple range fault
14 Feb  2 20:22:14 raspberrypi python[419]: 2024-02-02 20:22:14,476 ERROR oven: Problem reading temp (ignored) cold junction range fault
14 Feb  2 20:22:14 raspberrypi python[419]: 2024-02-02 20:22:14,710 ERROR oven: Problem reading temp (ignored) thermocouple range fault
14 Feb  2 20:22:14 raspberrypi python[419]: 2024-02-02 20:22:14,942 ERROR oven: Problem reading temp (ignored) cold junction range fault
14 Feb  2 20:22:15 raspberrypi python[419]: 2024-02-02 20:22:15,170 ERROR oven: Problem reading temp (ignored) thermocouple range fault
14 Feb  2 20:22:15 raspberrypi python[419]: 2024-02-02 20:22:15,395 ERROR oven: Problem reading temp (ignored) thermocouple range fault
14 Feb  2 20:22:15 raspberrypi python[419]: 2024-02-02 20:22:15,629 ERROR oven: Problem reading temp (ignored) thermocouple range fault

```

chipgarner, 2024-02-09:
@artcank Any luck with this? I am running the MAX31856 with no issues using the Adafruit setup instructions. No library changes. Are you using hardware or software pins? Both should work.

artcank, 2024-02-09:
No, I set up the max56 several years ago according to the schematics of the old master and had all kinds of trouble with my kiln(never could reach cone 6), not related to the Rpi3. I have since changed to an Rpi4 and an updated kiln. I used the sd card from my old attempt and have tried to update to the Blinka branch. 

Honestly, I guess I don't understand what the difference is between hardware and software pins so maybe that is the issue. Any help or guidance would be appreciated.

chipgarner, 2024-02-10:
What Pi os version are you using? ( You can run cat /etc/os-release) 

If you are using the card straight from the Rpi3 it may not work on the Rpi4. kiln-controller requires a lot of libraries, some of these, and possibly the OS, may check the hardware and load different things on different pi's. Also, you could have an older version of one library and a newer version of another - this sometimes creates problems.

The Blinka boards, including the Rpi, have pins designated by the manufacturer for MOSi, MISO etc. These hardware pins are designed for the task and the firmware in the board takes care of some of the required tasks. Other GPIO pins can be used but require another layer of software (called bitbangio), Using the software pins is faster, but his is not an issue for us as we are sending a small amount of data at every 2 seconds or so.

The software pins have proven to be more finicky but we think we have this worked out on the blinka branch. As far as I know, you and I are the only ones who have tested this. Your input is valuable!

I am doing some more testing, You may need to do a clean install staring with the OS.

chipgarner, 2024-02-11:
@artcank @jbruce12000 I have reproduced what I think is the same thing @artcank is seeing with the software pins on the '56. I am getting constant errors but a reasonable temperature. kiln-controller quits prematurely, I think because of all the errors.

I am working on sorting this out.

chipgarner, 2024-02-12:
So this is clearly the Adafruit library error. @artcank Sorry I didn't recognize this sooner! I had seen the issue with the many errors but did not realize kiln-controller could work for a while with the errors, and thus thought you were seeing something else.

WOW! I added a comment on the Adafruit bitbangio library issue pull request from @Linkeor a few minutes ago and they just merged it. Looks like we just need to reinstall the library. It is not up on pip yet, but the following should work as soon as it is. (I have no idea how long it takes for an update to appear on pip.)

pip3 install adafruit-circuitpython-bitbangio

chipgarner, 2024-02-12:
@artcank  
pip3 install --upgrade adafruit-circuitpython-bitbangio
should fix your problems. Do this inside the virtual environment if you are using one. (You may need sudo if not.)

Let us know if you still have issues or questions.

artcank, 2024-02-13:
@chipgarner That seems to have fixed it. Thank you. I am running a test fire right now with some new PID parameters.

@jbruce12000 The PID parameters I got from the auto tuner were:

kp = -13.76698
ki = 6.765
kd = 320.548

Is it normal to have a negative kp?

rondoc, 2024-02-13:
No negative P ..it would mean that as the error increases so would the
output. Not in the right direction!

MeltTechAu, 2024-02-13:
Keep tuning with all zeros  000 as k p ki kd it should not over shoot or
anything I have kilns o build identical it works perfect from 30 degs c and
up and should stay within 1 deg c with jbruces software I use

MeltTechAu, 2024-02-13:
Not one controller that I assemble\ Bartlett or tap etc can do this
cone mode needs every one’s help as well to make this controller beautiful
if you want to help with that go to MeltTechau on hit hub select from
master to cone mode see where I’m at and help of u can I’m also working on
profile identifying so we can list the cone list in correct order join the
crew let’s all be able to make and sell this we can all profit from this in
our little group I have a pottery group with 10000 people feed up for this
just needs some extras like profile speed control also been working on a
fast forward rewind slider and so much more

msloan75, 2024-02-19:
So I tried two cone 6 test fires this weekend and had trouble on both. If you look at the logs, on the 17th you can see the issues arise at around 2060 F. Eventually the Temp starts to drop, even though the kiln is still heating. Once I opened the kiln the next day my Cone 6 cones had melted to a Hot Cone 6. So the temp must have been rising even though the probe said it was going down. 

On the 18th the temp freezes at 1656 and then jumps to 1736. I eventually stop and then start the program again to keep it from cooling too much.

Any ideas?

[kiln.logs.txt](https://github.com/jbruce12000/kiln-controller/files/14334556/kiln.logs.txt)

rondoc, 2024-02-19:
I suggest that you use a digital thermometer connected to same tc that your
control uses. I like the Fluke as it has been reliable and accurate for a
very long time. Having another check on what is going on with the
controller is valuable and might give you a better idea of what going wrong
with your system. I have an independent controller so I can switch from the
pi if I have a problem. A failed firing is not pleasant and might trash a
kiln load of ware.
Ron.

chipgarner, 2024-02-19:
@msloan75 Are you using the MAX31856? If so I may have seen one of your errors.

In your log, the temperature reading gets stuck, it stays the same for several readings in a row. It then comes back to a reasonable reading. You pointed out one of these cases (jumps to 1736) above. This may be something I have see before but only with the MAX31856.

I also see a separate problem. The kiln begins to cool rapidly on the 17th at about 14:07:36 and on the 18th at about 18:30:36. In. both cases heat_on=2, heat_off=0 continues to show the power is full on, but the kiln is cooling as if it is off. For some reason, the commands to switch the power on are not working. (This happens on my kiln if the kiln sitter shuts off, but this depends on how things are hooked up.) The first thing I would do is check the torque on all the power connections.

artcank, 2024-02-19:
@chipgarner- Yes, I am using the MAX31856. 
So you have the Kiln Sitter setup as a backup? Maybe I should have kept my old Kiln Sitter hooked up...

I am also using the cheaper S type thermocouples from ebay: http://tinyurl.com/2fru5c52

rondoc, 2024-02-19:
You can get a power and current meter that will monitor the power being
delivered. They are cheap and also will give kwhrs so can check cost with
the pi system. Get one with voltage that matches your kiln. Usually 240
volts.
Also put a LED on the pi output to the relay or SS switch that turns power
on. The LED will flicker until the input reaches 100%  , ( if it does).
Just connect the LED with a 3 k resistor to the pin and other end to grnd
or +5 depending on how you have it set up. FYI .. LED s are polarity
sensitive

R.

chipgarner, 2024-02-20:
@artcank Yes, I could not see bypassing the trusty ancient kiln sitter when I made my manual kiln automatic. It's over 50 years old and still working, although it may have been unused for much of that time. I wired it so the Pi still has power. This lets me see what is happening with temperatures if it trips, and also allows testing the Pi software without  actually heating the elements.

chipgarner, 2024-02-20:
@jbruce12000 @msloan75 The above results seem to confirm a really bad problem that I have seen with the MAX 31856.
I am calling this the halting problem.  The call to get the temperature from the library does not return. This halts the calling tread.
 
It does seem to return after some, very variable,  period of time. (Easily many minutes.) In this case the kiln will start working again and resume the firing. This problem can be easy to miss, but could result in an over firing if it occurs at the wrong time.

I have reproduced this halt many times with various software, not just kiln-controller. It is definitely a problem in the library or in th MAX 31856 itself. I was hoping it was a one off problem with my 31856 but the results above show that it is at least two 31856's..

As I understand it, kiln_controller just keeps reporting the same temperature when this occurs.  Sorry, this is a bit convoluted.
The call to temp = self.thermocouple.temperature, line 317 in raw_temp() does not return. No error is thrown. raw_temp() is called by get_temperature() in line 138. This halts the call to temp = self.get_temperature(), line 158, in the while loop. The stopped while
 loop no longer updates the temperature by calling self.temptracker.add(temp).

Calls to get_avg_temp(), called by another (the main?) thread, simply return the same temperature whenever it is called. I tod you it was convoluted.

@msloan75 's log above shows this happening several times. Two identical temperature readings in a row from the MAX31856 are very unlikely because of the sensor noise. Three or more are probably an error and several minutes is a potentially a serious problem. There are many examples of this in the log. @msloan75 points out probably the longest, the temperature freezes at 1656 and then jumps to 1736.

I have no idea of the remedy for this. I thought maybe detect the problem and restart the '56, but this does not seem to work. I have tried restarting the Pi and re powering the whole system while this was happening and it did not fix the problem. It does seem to recover on its own, may be always, but who know when.  It could be a strange wiring or grounding issue, jiggling wires seemed to fix it one time, but could have been the timing. I have also not been able to repeat this issue recently after running two 56's for about 50 hours after re-wiring things.

Again, this could have been happening for years with kiln-controller users without anyone noticing. The graph would not show it as the time stops for the kiln to catch up. You would have to notice that the temperature had not changed for a while. It would only show up as an unexplained over firing.

rondoc, 2024-02-20:
I am not a code guy but it should be easy to track the 56 output
independently of the pi kiln software.
My guess is that the 56 output might need to be buffered to avoid a loading
issue. Also supply noise might be a problem with the 56 board.
To eliminate these possibilities the supply should have a reasonable size
cap from + to - , around 100mfd with a 1000pf ceramic in parallel. These
need to be right at the 56 board.
Buffering could be a single emitter follower single transistor,  this will
not invert the digital signal . Any small signal npn transistor will be ok.
Also is it possible that when the pi code requests the temperature it needs
to ask several times quickly and then if all same ( since temps not
changing quickly)  then uses that data. If not same then an alert, error
would be needed.
FYI I have zero problems with my pi setup
using K type tc and the 55 board from adafruit.
Ron.

jbruce12000, 2024-07-01:
this problem sucks. I think a custom `watcher` script could be created that checks if the kiln trying to catch up and the temp is not changing. This would detect when the issue happens... but I'm not sure what the action would be.

jbruce12000, 2024-07-02:
The main branch, which was created from blinka, replaced master as the default branch. master is still there, but should be avoided since it is no longer maintained.

closed 2024-07-02 by jbruce12000