jbruce12000/kiln-controller #172: Question (not issue): API function to put current profile on "hold"

samuelgduncan, 2024-03-18:
We live in South Africa where we have load shedding (i.e. mains power gets turned off during certain scheduled hours of each day, different each day and week).  My wife is a potter and has a 10KW single phase kiln with a digital controller (but without any input to control apart from buttons on the housing). 

I have split the 10KW element into two 5KW elements that can be independently controlled.  In order to be able to fire during load shedding, we start the firing a few hours before load shedding  so that the kiln is at ±400C when load shedding starts.  When the load shedding starts, I turn one of the 5KW elements off and manually put the kiln controller into a mode where it simply holds its current temperature.  We then use our solar + inverter to power the 5KW element. In this way the kiln is able to hold the temperature for the 2 hours during which mains power is out. 

When mains power returns, I switch the second 5KW element back on and manually advance the kiln controller to resume its ramp up and complete the firing with mains power.

I am already able to remotely switch the 5KW elements, but am looking for an automated way to put the kiln controller into "hold" mode when mains goes off,  and then automatically resume the firing once mains returns.  I'm looking to do this by using a Raspberry Pi controller instead of the current controller.  I'll be able to send commands via the API from my inverter using Node-RED.

Is there any way to easily add an API function to the raspberry pi kiln controller to put it into "hold" mode?  I know I could use the "startat" command via API to move it to a specific time in in a firing profile, but the problem is that I don't know exactly what temperature the kiln will be when the mains goes.. Hence I really need to be able to just make it "hold" whenever this happens.

Many thanks
Sam

samuelgduncan, 2024-06-26:
> I want to make sure I understand the feature request.
>
>    1. power goes out
>    2. you detect that somehow or schedule it somehow
>    3. you call api to put kiln on hold... which means save state, do not
>    advance time, hold the current temperature using some back-up power solution
>    4. you detect when power returns or schedule it somehow
>    5. you call the api to take the kiln off hold... load state, resume.

You are 100% correct :). For step 2, I would schedule it as we have a
published schedule of when the power goes off.

For step 4 I would detect the return of mains power (via my inverter)

jbruce12000, 2024-06-26:
What if the outage occurs at a high temperature... would you still want to hold?
What about really low temps...would you still want to hold?
Are there limits to the hold time (how long is too long)?

I feel like this has the potential to get complex.

samuelgduncan, 2024-06-26:
I don't think you're wrong about the potential to get complex ;). But I'm
keen to give it a bash keeping it simple as below:

What if the outage occurs at a high temperature... would you still want to
hold? - If the outage occurs at high temp, it would be an unplanned outage
and I wouldn't want to hold.  In this case I'll have a separate mechanism
to simply shut the whole kiln off.
What about really low temps...would you still want to hold? I'll plan the
firing so that it starts before the scheduled outage and gets to 300 or
400C.  There wouldn't be any need to hold at temps below that.
Are there limits to the hold time (how long is too long)?  Hold time
shouldn't ever be more than 3 hours max, so we could hard code that as a
safety stop.

Really appreciate the fact you're even looking at this - it's been quite a
saga getting the kiln working in the midst of our "load shedding"  (
https://en.wikipedia.org/wiki/South_African_energy_crisis#:~:text=Loadshedding,-Eskom%20Power%20generation&text=In%20South%20Africa%2C%20loadshedding%20has,on%20coal%2Dfired%20power%20plants.
in case you're interested)

Thanks

jbruce12000, 2024-06-30:
api pause and resume added to the api on the blinka branch (which is the only branch with active development). It's not perfect (I'm sure there are edge cases with issues), but it works.

closed 2024-06-30 by jbruce12000

rondoc, 2024-06-30:
It would be easier to use a gen set or solar plus battery. May be rethink
firing process. I am so sorry for your situation.
R.

samuelgduncan, 2024-07-09:
This is amazing, thank you!  Apologies for delayed response, I've been
away. I'm going to get onto testing the new function asap

samuelgduncan, 2024-07-09:
I am actually using solar and battery.  The full picture is as below, not
shwon is the automatic transfer switch to move onto inverter power when
mains is off.  We only have a 5kVA inverter, hence need to drive the 10kW
kiln at half power when the mains is off. At half power, we can hold the
temperature, not ramp up - hence the need for the API hold function.

jbruce12000, 2024-07-09:
note that the blinka branch is now main.