jbruce12000/kiln-controller #173: No Support for Three-Zone Control

kulps, 2024-03-29:
Hey everyone, 

Our studio is considering replacing our V6-CF controllers with a Kiln-Controller board, but it looks like the current build only supports single-zone reading. Our Kilns are setup with 3 thermocouples each. 

Is this something that's been discussed or attempted before? 

Any insight would be greatly appreciated!

rondoc, 2024-03-29:
Are there three element groups? What do each tc do now. The pi controller
is so cheap you could easily just use three pi's.
Then it would be like it is now but more precise in heating since each
section would have its own pi .
With some extra code and a bit of hardware I would think that one pi could
sequentially manage all three.
Ronald

chipgarner, 2024-03-29:
I have a project based on kiln-controller that supports up to 4 zones. I have never a seen a one zone kiln, including manually controlled kilns, bigger than about one cubic foot. @rondoc 's kiln is an exception.

See https://github.com/chipgarner/MultiZoneKilnController

This project is nowhere as far along, or as user friendly, as kiln-controller. I have tested it multiple times on a two zone kiln. @kulps
let me know if you are interested. The main thing missing right now is documentation.

turt2live, 2024-03-29:
@chipgarner this looks interesting! Does your project support compensating for a bad element or thermocouple too, or is it more just trying to get a consistent firing?

chipgarner, 2024-03-29:
@kulps The UI should be pretty user friendly. The setup and documentation is not. It would be a big help to work on this with other people - I need feedback on what is clear and what is not. Also, it would be motivating to have help of any sort.

I am not very happy with the reliability of the MAX 31855 and MAX 31856 TC boards. The '55 reports more errors than readings at high temperatures, and the '55 halts and recovers at some unknown time later. These don't show up on the kiln-controller UI, only in the logs. Users are not likely to notice unless a problem occurs at the wrong time.

rondoc, 2024-03-29:
@garner have you actually monitored the output from the 55 tc board? Not
via the pi kiln controller. I wonder if there is a timing error or noise
that you see. The 55 board cannot produce errors it is a simple single chip
and I don't see where you could get erratic output from except from noise
via it's  tc wiring .
Be useful to nail this "error" anomaly down as the simple pi controller
with its great use depends on it.
My use through about 20 firings showed no problems but I have not searched
the log to see what's going on. I fire to cone10. FYI.
Electrical conductivity of the surfaces that are in contact with the tc
wires and with high temps could make the tc give an error. If the tc is not
in a high fire high alumina shield.
Ronald.

chipgarner, 2024-03-29:
@rondoc The board sends errors like short to ground, short to power, over and over. Connecting the TC to a Fluke or a '56 shows no errors. The errors come more and more often as the temperature increases. The errors continue in the same manner as the kin is cooling, with the high voltage switching shut off. I have tried different software - my software graphs the errors real time. To further define "errors", the board sends out a set of bits that corresponds to a specific error as defined in the board documentation. This has been described as "noise" in other issues here. Kiln-controller gets a good temperature every several seconds or so, so it continues to work,

The '56 error is completely different. The Pi sends a message to the board for a temperature. The board never answers, or may answer several minutes later. In kiln-controller this hangs the thread that reads the temperature. I have tried software the avoids the hang, but it also does not get a response, neither an error or a temperature.

rondoc, 2024-03-30:
@Chip Garner ***@***.***>  this does not make electronic
sense to me. The fact that errors increase with temperature is a clue that
it's  a tc mounting problem. I suggest connecting a tc to the 55 board with
short leads then heating the tc with a flame / torch and monitoring the
output of the board. No other electronics needed or wanted. You might have
to make a small insulated box to achieve high temps.
Also try same with the 56 board.
Ronald.

chipgarner, 2024-04-01:
@turt2live I have not looked at compensating for TC or element problems yet. I don't think this would be difficult to add. I have made it so you can switch to manual mode at any time. This allows you to set the heating values for each set of elements with sliders. I have used this a couple of times to save a firing, once when a '56 quit and another time when my experiment with a model based controller (replacing the PID controller) went awry.

chipgarner, 2024-04-01:
@rondoc This is the wrong place for the TC board issues. It has been discussed before under several issues here. I have tested everything I can think. I would like to consolidate what I have done and some of the previous discussion all in one place. Once I have done this, I would really like your input on other things that could be tried. I am leaving this week for Texas for the eclipse, so I may not get this done for a couple of weeks. I will post a link when it is up.

rondoc, 2024-04-01:
@Chip Garner ***@***.***> if this is not the place then why
do I see so many questions, here, coming up about tc errors? I have no
problem discussing electronic and electrical questions elsewhere,  but
where?
Ronald.

chipgarner, 2024-04-01:
@rondoc I just meant that this issue is No Support for Three-Zone Control, not TC reliability. Also, the discussion has become fractured over several issues. I am largely responsible for this, and plan to set up a better place to continue and preserve the discussion. I will let you no where as soon as I get this going.

jbruce12000, 2024-04-09:
I can confirm the issues with the 31855. The number of errors increase with temperature. I believe this is a problem with the 31855 interpretation of the signals coming from the tc. I don't believe this is a tc problem.

jbruce12000, 2024-04-09:
To answer the question about multi-zone control... it has come up before. I've never been a believer in it. PID control is based on a single reading... this reading does not strictly have to come from a single device [ie thermocouple], but it must be a single reading. If it's coming from multiple thermocouples, the readings must be combined in some way - like averaging them.

It is impossible to have multiple PID controllers all managing different zones of the same kiln. They would constantly fight each other for control of the space.  You could fake it by having one dominant controller, but that's basically the same as having a single PID controller.

I don't have plans to add multi-zone control.

It would certainly be possible to connect three 31855 controllers to read from 3 TCs while connected to a single rpi. The 3 readings would be averaged for use by the PID software. A single multi-phase SSR could be used to control heat application to the entire kiln [no individual zone control which is again, a farse].

rondoc, 2024-04-10:
@jason. The signal from the tc is very low
Voltage in the millivolt region.its just DC . The chip takes this and also
measures ambient temperature. It takes the difference via analog circuitry
does an A to D and then this is read by wharever is attached to the board.
This process doesn't change with the tc temperature. Based on the above it
really seems unlikely that the resultant interpretation of temperature will
be anything except a digital stream.
The fact that some users have problem with errors that look like tc reading
is still IMNSHO due to the way the tc is connected.
Until a test of just the 55 board and a digital readout is done of the
boards output code at all temperatures I will stay with my opinion.
I have not any problem using the 55 board with the kiln controller and am
very happy using it.
Ronald.

rondoc, 2024-04-10:
Re 3 element control. What is wrong with one pi three TCS  and sequentially
switching both the TCS and the SSRs?
Could the same pid settings work for this?
It seems doable but messy
I have only one element but I achieve very even heating by rotating the
ware during firing.
Ronald.

jbruce12000, 2024-04-11:
One space [like a kiln] cannot have multiple PID controllers effectively controlling the temperature. The controllers will compete with each other. Without a kiln vent, this would likely mean the bottom controller is the most active. With a kiln vent, it would be a crap shoot.

As stated previously, you could have multiple TCs connected to a single controller and average the TC inputs. That single controller would control all elements.

rondoc, 2024-04-11:
Hi I have reviewed the 55 data sheet again. It states that capacitors
should be put across the tc input at the 55 board and also across the 5volt
input. It seems clear that unless this is done noise will cause errors. The
voltage levels from the tc are very tiny so electrical noise is a potential
problem.
What is the kiln controller's process for having the data from the 55
board? Since there is no need for fast processing I wonder how this is
handled. Perhaps knowing ahead what the data will be within a known spread
errors could be ignored?
The method of ensuring even heating with multiple elements seems like an
interesting discussion.  Has anyone attempted any of the solutions offered?
Ronald.

jbruce12000, 2024-04-11:
I have multiple elements...three to be specific. They are ALL either on or off at the same time for a specific period during each duty cycle. They are not individually controlled.

I'll have to check if I have caps at power and tc input. not sure.

turt2live, 2024-04-11:
Out of interest, how big is your kiln? Our BX2827D comes standard with three zone control, but it's also 10 cubic feet. We've found the consistency and reliability of 3 zone control to be valuable, particularly when trying to heat such a large volume of air.

We also upgraded our Skutt 1027-3 to have 3 zones (previously it just had a single middle zone), and we've had more consistent firings since.

Granted, these results are with the factory controllers. We'd like to get some monitoring on the kilns to review the firing and see when it's cool remotely, which is why we're interested in the raspberry pi solution.

rondoc, 2024-04-11:
The pi controller is a gear monitoring system. I use it even when I use my
manual controller.

rondoc, 2024-04-11:
Sorry "great monitoring ...."
My rotary kiln is 8 cuft. It is extremy well insulated so low cost in
electricity. Insulation means slow cooldown 16hrs from cone10.
Ronald.

jbruce12000, 2024-04-11:
Skutt KS-1018... so much smaller than yours I expect.

chipgarner, 2024-04-12:
Multiple zone control has been available for many years and is not difficult. A separate TC and power switch is needed for each zone. For example, my 2 zone Paragon A-88 B has four elements and two thermocouples.  The top two elements are connected to one SSR, and the bottom two elements are connected to a second SSR. The controller uses the reading from the top TC to control the top SSR, and from the bottom TC to control the bottom SSR. The controller treats the two zones independently. This is essentially the same for all the commercially available controllers.

I have not seen any issues with interactions between the kiln zones. Radiative heat transfer increases with the 4th power of temperature. This means that radiation dominates the heating in the kiln at higher temperatures, and is why the kiln temperatures tend to even out at high temperature. If the zones are kept at the same temperature there is no interaction. If one zone gets hotter, it tends to help the other zones catch up by transferring heat to them.

You do have to deal with lagging at full power. If the desired temperature profile is faster than the kiln can keep up with one of the zones will lag. All the zones will go to full power but will not (in most cases) stay at the same temperatures because the power and heat loss is not the same for each zone. I have used the same method as the Bartlett controllers, slow down the profile slope until the coolest zone can keep up.

I do not know if modern commercial controllers are using PIDs, but is very likely that hey are using PIDs or a simpler method, such as proportional control only. If a present day control engineer were to design a new system it would probably use model based control.

jbruce12000, 2024-04-17:
thanks chip. great info. sounds like this should be added as a feature.

jbruce12000, 2024-04-17:
I'll think about implementation. It's complex because it impacts config, execution, logging, ui... basically every aspect.