jbruce12000/kiln-controller #174: Real temperature <> Kiln-controller

We-Sing, 2024-04-23:
Hi, I'm very happy with the kiln controller. I can control my kiln almost properly. I only have a deviation between the actual and the temperature of the kiln controller. When I also connect an external temperature meter to the oven, it measures a different value.

The external meter is also a K-type without any deviation. The difference between the kiln controller reading and the actual value increases linearly. When the kiln controller reads 280°C, the actual value is 300°C, at 1072°C, the actual temperature is 1270°C. So, everything in between is linear. I hope someone knows where the problem could be. I've tried different thermocouples, but the difference remains the same.

Thanks!!
Jan

rondoc, 2024-04-23:
I have a Fluke meter and it reads same as the kiln controller. I suggest
you check the connections. The single tc can feed both the pi and the
meter. Use tc wire for both and note polarity as the tc wire is critical.
There might be a small delay due to the sampling time in the pi and this
will look like an error.
Your error must be related to the wiring as it is linear. In other words
you have two effective TCS  working due to wiring error.
Ron.

We-Sing, 2024-04-23:
Thanks Ron, I will check this weekend!

We-Sing, 2024-04-25:
Hi Ron, 
Thanks for the tips. I've connected the meter (unfortunately not a Fluke) to the TC. TC is also connected to the Pi (3 b+). Additionally, I replaced the wiring from TC to Pi. There's still a linear difference, but it's 10% less. Do you have any other tips? I've been reading a lot about grounding... or should I just buy a new Raspberry Pi and rebuild everything? Thanks in advance! Jan

rondoc, 2024-04-25:
Check out your meter. You can use ice for a 0C and boiling water for a 100C
test.
The problem is not with the pi. Could be the tc 55 board..not likely
either.
Have you done a firing with the kiln controller and a cone . The cones are
a good way to check upper temp.
R.

We-Sing, 2024-04-26:
Thanks, I will buy cones and do the final check. Otherwise I just adjust de kiln schedules. Have a nice weekend!

sirquince, 2024-05-10:
I'm having the same issue, did you get it figured out?
I tried using different type k thermocouple wire but that didn't solve it. I have an accurate meter attached in parallel so I can tell it's incorrect. I also switched GPIO pins as well as the wires from the max31856 to the pi with no luck. I can try replacing the pi and/or max31856 but doubt that'll fix it.
If we can't figure this out, is there a way to put in a "thermocouple_output = thermocouple_output * correction_multiple" kind of line in the scripts?

rondoc, 2024-05-10:
I use a k type tc reader from a reliable vendor. I set the pi to read the k
type. So far it works fine
I do find that the pin connections on the pi are not good. They are not
hold so corrosion sets in. I routinely  clean them and make sure they are
solid. If not the tc readings go all over the place.
You can check the system by making a schedule "firing" and letting it run.
Then check the logs if it doesn't follow the path set. Oscillation is
normal at low temps as long as they converge. The PID cannot work over the
total range to cone5 or 10 for me. At about 400F it does follow within a
degree.
Things I'd try are:
1. Connections...show up as erratic results
2. Pi could be defective
3. Grounding both the tc DONT. Also your system.
4. Wiring especially SSS usually at high temp when % is high..erratic
pathway
5. Set catchup to false.
6. Try a new 55 board if using a k tc.
7? Hope!
Take care Ronald

We-Sing, 2024-05-10:
Hi, I replaced the wiring on my K-type thermocouple. The measurement is now better but still not entirely parallel to the actual reading. A workaround is simply adjusting the firing schedule to match the value of the actual measurement.

Jan

rondoc, 2024-05-10:
Making sure no part of the tc wiring gets grounded is critical. Often the
wires going into kiln get close enough to be a problem especially at high
temperatures when some ceramics become electrically conductive. The tc 55
and 56 chips require isolated inputs but the meter doesn't care. This could
be why a difference in reading. Also keeping chip in a room temp
environment is important as it has to measure ( its called the cold
junction) the ambient temperature to provide the reference for the tc
itself.
Ronald.

sirquince, 2024-05-13:
Nothing hardware wise I tried ended up working. I ran my kiln for a few hours and took note of what this kiln controller said the temp was vs what my accurate handheld meter said (they're wired in parallel to the same thermocouple). I plotted the readings and found the equation of the trendline that the points generated. Since the points are actually linear, I can go ahead and just use the generated equation as-is. 
(The line is the conversion from what the max31856 says the temp is to what it actually is)

[image]

I then went into oven.py and added a line to convert the temperature readings based on the generated equation

```python
def get_temperature(self):
        '''read temp from tc and convert if needed'''
        try:
            temp = self.raw_temp() # raw_temp provided by subclasses
            if config.temp_scale.lower() == "f":
                temp = (temp*9/5)+32
#I added this line here vvv~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                temp = (temp*1.31)-21.9
            self.status.good()
            return temp
``` 

And that works well enough that I'm calling it good for now. I'll keep my handheld meter wired in so I can keep an eye on if the controller starts to drift over time. If I get so inclined in the future and fix the hardware issue then I'll just comment out that line.

jbruce12000, 2024-06-26:
The requestor solved this problem by adding a linear equation in get_temperature.

closed 2024-06-26 by jbruce12000