jbruce12000/kiln-controller #177: Issue with temperature reading

lluisllenas, 2024-06-05:
Hello,
I've installed the controller on an old kiln and I'm having issues with the temperature reading. I'm using a MAX 31855 with a K type thermocouple
I wired everything on a breadboard and worked well, with no errors. Then, I used a soldering protoboard and I welded the electronic components.
After soldering I had "????" errors, but I managed to fire the kiln configuring the controller to ignore them.
The controller stopped at around 850ºC, I started it again and then it stopped at 900ºC and I let it cool down. The autoboot was enabled and I don't know how to open a terminal to read the errors and the log seems to not be working, so I don't know what error trigered the stopping of the kiln.
I disassembled the thermocouple and tried to  resolder the wires and now I get the following error:
ERROR oven: Problem reading temp N/C:True GND:True VCC:True ???:False
It may be because of the MAX31855 board but I'm not sure, it wasn't easy to get it from adafuit and I bought a replica of it.
Can you help me on getting it working?
Thanks so much.
Lluís

rondoc, 2024-06-05:
I don't know but soldering the tc wiring is NOT a good idea. Use direct
contact between tc wires only. Usually a screw and washer combo works
fine.
Ronald.

lluisllenas, 2024-06-05:
Thanks for the answer, but I explained myself wrong. I soldered the MAX31855 and the RP connections, not the TC. The TC is connected with a screw connector (not sure if it's called this way)
Lluís
[image]

rondoc, 2024-06-05:
There are too many unknowns in your problem for me. I would guess that
since it all worked as a breadboard you have misconnected something. Also
check that the config file is set up correctly to.match your conditions.
I do this when running my kiln firing, might be helpful..:
1. Turn on power and check that pi is running ok.
2. Check with external meter that the tc is reading ok.
3. Open the config file and make any changes needed. Usually none.
4. Open kiln file and check run in terminal.
5. I then look at my Android phone and connect to the pi app.
6. I select the firing profile I want to use.
7. I then goto start on the app on my phone.
8. The temperature shows up and is close to my meter reading.
9. The heating % starts to show and the temp starts to try to follow the
profile
10. The kiln temp wanders about until about 400F when it tracks within a
degree.
11. I make a cup if tea and get on with life until it gets close to max
temp.
12. I usually watch the last 30 minutes but usually nothing unusual occurs.
13.shut down occurs then I watch the kiln cool down on my phone.
14. Open kiln at about 350F cause I am impatient ...when handleable take
out my
priceless treasures !
15. Thank everyone who has made this controller possible..thank you.
Ronald.

jbruce12000, 2024-06-26:
hmmm. there's a lot going on here and too much changing to be able to give you a specific answer about how to fix it. It's really important that you are able to look at the logs so you can troubleshoot... or that so that you can post them here and we can help you troubleshoot.

is your problem fixed?

lluisllenas, 2024-06-27:
Hello @rondoc and @jbruce12000 ,
Thanks so much for your answers and your support, much appreciated!
I haven't had much time to work on the kiln lately.
I still have the problem, I've checked the connections and everything seems to be fine.
I'm not getting the logs, maybe because I changed the name of the RP and the route on the script that writes the log is not updated. I think I will flash again the RP and leave the "pi" name so I avoid any problems like this and try again.
Also, I will take another look at the connections to see if I can find anything wrong.
Again, thanks so much for your answers and for the project itself!
Lluís

jbruce12000, 2024-06-27:
if you're automatically starting the kiln controller at boot, your logs should be going to /var/log/. Post the output of

`ls -la /var/log`

Trying running ziplogs and post the output

```
cd kiln-controller
./ziplogs

```

lluisllenas, 2024-07-02:
Hello @jbruce12000  and thanks for your answer.
I could get the ziplogs but I haven't copied to my PC to upload them yet, I have to spend some time learning how to do it.
I've solved part of the problem I had, it was a poor connection on one of the welding paths...
I still get the following error every 2-5 seconds:
ERROR oven: Problem reading temp N/C:False GND:True VCC:False ???:False
Reading some other issue posts, I see that it's a common problem when using the MAX31855, so I think everything is working well now!
I'll fire it probably on Thursday and see how it goes.
Another question, for setting up a webhook on Slack you have to be a premium user now, is there any other alternative to have the updates of the watcher on the mobile for free?
Again, thanks so much for your support!
Lluís

jbruce12000, 2024-07-03:
bummer about slack and not being able to set up a webhook without premium.

not sure of any other services that offer this kind of functionality for free. If you find one, lmk.

closed 2024-07-03 by jbruce12000