jbruce12000/kiln-controller #178: Thermocouple Issue

gumboman4, 2024-06-13:
I have been using kiln controller to run an older kiln without issues for a couple of years.
I recently had an issue with the run  ending at exactly the same time/temperature before I was unable to start  a profile at all.
I notice that the TC now constantly reads 32F when running test-thermocouple.py
I am thinking my TC has failed? I have checked for continuity on wire from TC to boards and that checks out ok. I am using 31855 with a type K TC. Should I just go ahead and order a new TC? 
Or is there something else I can check?

chipgarner, 2024-06-13:
You can check the thermocouple with a voltmeter on millivolts scale. 32F (OC)  doesn't sound like a TC failure. Check the wiring from the 31855 to the Pi if you haven't already.

gumboman4, 2024-06-13:
I unplugged the 31855 completely and it's still just showing 32 F and the kiln hot icon illuminated. Would this mean the Raspberry Pi itself is damaged? The old TC was falling apart upon inspection. I have a replacement TC and Raspberry Pi I can try to swap out.

rondoc, 2024-06-13:
When the 55 board is unplugged from the pi you will get 32F showing. My
guess is that the 55 was not correctly plugged into the pi and/or the 55
board is not correctly powered. It's easy to misread the pinout on the pi.
I use a template to avoid this.
R.

jbruce12000, 2024-06-25:
did you get this fixed with replacement hardware?

jbruce12000, 2024-08-15:
gonna leave this open one more week while waiting on a response