jbruce12000/kiln-controller #179: error on setup

djs1969, 2024-06-16:
hi getting this error on setup had it working once but something went wrong and could not get back on to server have tried several times but still same error.
env) kiln@kiln:~/kiln-controller $ pip3 install --upgrade setuptools
Configuration file could not be loaded.
File contains no section headers.
file: '/etc/pip.conf', line: 1
'break-system-packages = true\n'
thanks in advance

jbruce12000, 2024-06-25:
try...
`sudo mv /etc/pip.conf /etc/pip.conf.old
`
then re-run.

djs1969, 2024-06-29:
thanks for getting back to me can now get it to work but as soon as i put simulate = False get the following result 
2024-06-29 15:32:19,717 INFO kiln-controller: Starting kiln controller
2024-06-29 15:32:19,720 INFO kiln-controller: this is a real kiln
2024-06-29 15:32:19,720 INFO oven: import MAX31856
2024-06-29 15:32:19,720 INFO oven: init MAX31856
Traceback (most recent call last):
  File "/home/kiln/kiln-controller/./kiln-controller.py", line 43, in <module>
    oven = RealOven()
           ^^^^^^^^^^
  File "/home/kiln/kiln-controller/lib/oven.py", line 498, in __init__
    self.board = Board()
                 ^^^^^^^
  File "/home/kiln/kiln-controller/lib/oven.py", line 67, in __init__
    self.create_temp_sensor()
  File "/home/kiln/kiln-controller/lib/oven.py", line 95, in create_temp_sensor
    self.temp_sensor = TempSensorReal()
                       ^^^^^^^^^^^^^^^^
  File "/home/kiln/kiln-controller/lib/oven.py", line 140, in __init__
    self.thermocouple = MAX31856(tc_type=config.thermocouple_type,
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kiln/kiln-controller/lib/max31856.py", line 127, in __init__
    gpio = Adafruit_GPIO.get_platform_gpio()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kiln/kiln-controller/venv/lib/python3.11/site-packages/Adafruit_GP                                             IO/GPIO.py", line 426, in get_platform_gpio
    raise RuntimeError('Could not determine platform.')
RuntimeError: Could not determine platform.
(venv) kiln@kiln:~/kiln-controller $
below is my config
import logging
import os

# uncomment this if using MAX-31856
from lib.max31856 import MAX31856

########################################################################
#
#   General options

### Logging
log_level = logging.INFO
log_format = '%(asctime)s %(levelname)s %(name)s: %(message)s'

### Server
listening_port = 8081

########################################################################
# Cost Information
#
# This is used to calculate a cost estimate before a run. It's also used
# to produce the actual cost during a run. My kiln has three
# elements that when my switches are set to high, consume 9460 watts.

kwh_rate        = 0.1319  # cost per kilowatt hour per currency_type to calculate cost to run job
kw_elements     = 9.460 # if the kiln elements are on, the wattage in kilowatts
currency_type   = "£"   # Currency Symbol to show when calculating cost to run job

########################################################################
#
#   GPIO Setup (BCM SoC Numbering Schema)
#
#   Check the RasPi docs to see where these GPIOs are
#   connected on the P1 header for your board type/rev.
#   These were tested on a Pi B Rev2 but of course you
#   can use whichever GPIO you prefer/have available.

### Outputs
gpio_heat = 23  # Switches zero-cross solid-state-relay

### Thermocouple Adapter selection:
#   max31855 - bitbang SPI interface
[... 119 lines cut ...]
# reading was correct anyway
ignore_tc_short_errors = False 

########################################################################
# automatic restarts - if you have a power brown-out and the raspberry pi
# reboots, this restarts your kiln where it left off in the firing profile.
# This only happens if power comes back before automatic_restart_window
# is exceeded (in minutes). The kiln-controller.py process must start
# automatically on boot-up for this to work.
# DO NOT put automatic_restart_state_file anywhere in /tmp. It could be
# cleaned up (deleted) by the OS on boot.
# The state file is written to disk every sensor_time_wait seconds (2s by default)
# and is written in the same directory as config.py.
automatic_restarts = True
automatic_restart_window = 15 # max minutes since power outage
automatic_restart_state_file = os.path.abspath(os.path.join(os.path.dirname( __file__ ),'state.json'))

########################################################################
# load kiln profiles from this directory
# created a repo where anyone can contribute profiles. The objective is
# to load profiles from this repository by default.
# See https://github.com/jbruce12000/kiln-profiles
kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),"storage", "profiles")) 
#kiln_profiles_directory = os.path.abspath(os.path.join(os.path.dirname( __file__ ),'..','kiln-profiles','pottery')) 
thanks Darren

jbruce12000, 2024-07-05:
The blinka branch is now the main branch (and is the default).

djs1969, 2024-07-07:
Thanks got it running without switching but when run hits first target and says run complete and stops

jbruce12000, 2024-07-08:
I assume "without switching" means that you did not re-wire anything.

1. look at every line in config.py and make sure it matches your current configuration / what you want / need... pay careful attention to this section:
```
    spi_sclk  = board.D17 #spi clock
    spi_miso  = board.D27  #spi Microcomputer In Serial Out
    spi_cs    = board.D22  #spi Chip Select
    spi_mosi  = board.D10 #spi Microcomputer Out Serial In (not connected) 
    gpio_heat = board.D23 #output that controls relay```
2. run ./test_thermocouple.py to make sure your thermocouple is working as expected
3. run ./test_output.py to make sure that your output is configured correctly and that it turns your kiln on and off

jbruce12000, 2024-08-14:
did you get your setup working yet?

djs1969, 2024-08-15:
Hi  Jason did a first bisque firing at the weekend and work ok trying glaze this weekend taken this long as been on holiday and also had no Internet where kiln is.
Cheers Daz

jbruce12000, 2024-08-15:
Congratulations on your first successful firing! Nice work.

closed 2024-08-15 by jbruce12000

rondoc, 2024-08-15:
Not sure where to.post this.
System was all.working until.I had to change my local.network login. Now
although the pi is connected to the network my cell and local.pc cannot
access the pi kilncontroller as they did.
What should I do?
Thanks
Ronald.

rondoc, 2024-08-15:
My web address had changed due to new router. Thanks for help all ok now.
Ronald.

jbruce12000, 2024-08-22:
I'd set up static IPs on that router for your kiln. That way you always know what it will be.

regardless, changing routers would still be a problem.