jbruce12000/kiln-controller #181: Server connection issue

Maditiss, 2024-07-06:
A few days ago, there was electricity shut down and my kiln run was left in complete. Tried starting it today but the online kiln controller server is not working. It’s showing an error saying that it’s not connected with the server. I’m not a technical person. Please do share insights how to resolve this

jbruce12000, 2024-07-08:
I would start by restarting the raspberry pi. This should've happened when the power came back on, so there might be a bigger problem here.

You could have a corrupt disk (unlikely), or a whole  bunch of other problems. Maybe you don't have the kiln-controller.py process set to auto start?

To really be sure, you'd have to post the logs.

jbruce12000, 2024-08-14:
Any more info on this?

Maditiss, 2024-08-15:
Hey ! Thank you for responding! I just checked your mail!

The issue got resolved. Idk how and why but the name of the server changed.
For eg. 192.168.1.104 it was .. it became 192.....1.106

I scanned all the active servers via an app. And then used the current
active raspberry Pi server

jbruce12000, 2024-08-15:
This is common on home networks. Your router is likely handing out ip addresses using DHCP which is configured to have a lifetime for each ip address lease. It uses a pool and can changed.

You can fix this by configuring a static ip address for that rpi on your router.

closed 2024-08-15 by jbruce12000