jbruce12000/kiln-controller #182: Auto tune issue

MeltTechAu, 2024-07-19:
When trying to autotune  as per instructions it says can not import config adapt it copy examaple config but the kiln controller runs fine any thoughts jbruce im moving to the max 31856 and the new blinka software please help again thank you for your valued time
cheers brett from melttech

och29, 2024-07-20:
I'm getting the same error. My process:
 - From a clean boot (with autostart turned on)
 - Confirm web interface is active, thermocouple and SSR are functional (using pre-built scripts)
 - Via SSH `systemctl stop kiln-controller`
Output:
```

pi@kilnpi:~/kiln-controller $ systemctl stop kiln-controller
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Authentication is required to stop 'kiln-controller.service'.
Authenticating as: ,,, (pi)
Password:
==== AUTHENTICATION COMPLETE ====
pi@kilnpi:~/kiln-controller $ python kiln-tuner.py recordprofile zn.csv
Could not import config file.
Copy config.py.EXAMPLE to config.py and adapt it for your setup.
pi@kilnpi:~/kiln-controller $
```

config.py is present in the folder
```
pi@kilnpi:~/kiln-controller $ ls
config.py           kiln-tuner.py  requirements.txt  test-output.py
docs                lib            start-on-boot     test-thermocouple.py
gpioreadall.py      public         state.json        venv
kiln-controller.py  __pycache__    storage           watcher.py
kiln-logger.py      README.md      Test              ziplogs
```

jbruce12000, 2024-07-22:
can one of you comment out the try/except block near the top of kiln-tuner.py?
post the error.

What I'll likely do is remove this try/except block so that folks can see what the real error is.

och29, 2024-07-23:
> can one of you comment out the try/except block near the top of kiln-tuner.py? post the error.
> 
> What I'll likely do is remove this try/except block so that folks can see what the real error is.

```
pi@kilnpi:~/kiln-controller $ python kiln-tuner.py recordprofile zn.csv
usage: kiln-tuner.py [-h] [-c] [-t TARGET_TEMP] [-d TANGENT_DIVISOR] [-s]
kiln-tuner.py: error: unrecognized arguments: recordprofile zn.csv
```

MeltTechAu, 2024-07-23:
Confirmed getting same error och29

jbruce12000, 2024-07-25:
This is an old branch and old version. you need to grab the latest code. I did a simulation on the head of the main branch and found a bug... then fixed it.

MeltTechAu, 2024-07-25:
Getting same error with fresh dl today????

jbruce12000, 2024-07-25:
post the output of:

```source venv/bin/activate
./kiln-tuner.py
```
The kiln-tuner parameter handling was changed last Dec and should no longer be called like it is above.

jbruce12000, 2024-07-25:
The _master_ branch is no longer supported. The _main_ branch is now the active, supported branch. Switching branches may cause you some configuration or hardware re-work.

och29, 2024-07-25:
Ah, then it's an issue with the link here then! https://github.com/jbruce12000/kiln-controller?tab=readme-ov-file#pid-tuning

jbruce12000, 2024-07-25:
fixed the README. All links that pointed to _master_ or _blinka_ now point to _main_.

good find. lemme know if tuning still does not work for you.

MeltTechAu, 2024-07-25:
Legend jbruce thank you so much have a great day mate!

jbruce12000, 2024-07-26:
@MeltTechAu how did that tuning work out? any problems?

jbruce12000, 2024-08-14:
closing. create a new issue if you find a new problem with the code or docs.

closed 2024-08-14 by jbruce12000

och29, 2024-08-17:
I have tested this now, confirmed solved.