jbruce12000/kiln-controller #183: Subprocess Exited with error

TheHarmacist, 2024-07-21:
Setup: RPI Zero 2W with 32bit Raspberry Pi OS Lite (Debian Bookworm no Desktop environment)

Going through the stages of setting up the device via terminal, when I get to the '$ pip install -r requirements.txt' I receive this error: 
```

Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      + /home/brandon/kiln-controller/venv/bin/python3 /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5/vendored-meson/meson/meson.py setup /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5 /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5/.mesonpy-y542757w -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5/.mesonpy-y542757w/meson-python-native-file.ini
      The Meson build system
      Version: 1.4.99
      Source dir: /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5
      Build dir: /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5/.mesonpy-y542757w
      Build type: native build
      Project name: NumPy
      Project version: 2.0.1
      C compiler for the host machine: cc (gcc 12.2.0 "cc (Raspbian 12.2.0-14+rpi1) 12.2.0")
      C linker for the host machine: cc ld.bfd 2.40
      C++ compiler for the host machine: c++ (gcc 12.2.0 "c++ (Raspbian 12.2.0-14+rpi1) 12.2.0")
      C++ linker for the host machine: c++ ld.bfd 2.40
      Cython compiler for the host machine: cython (cython 3.0.10)
      Host machine cpu family: arm
      Host machine cpu: armv7l
      Program python found: YES (/home/brandon/kiln-controller/venv/bin/python3)
      Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
      Run-time dependency python found: NO (tried pkgconfig, pkgconfig and sysconfig)

      ../meson.build:41:12: ERROR: Python dependency not found

      A full log can be found at /tmp/pip-install-4d2sevkz/numpy_0e4b2c662f5c42159b91c4383c333cf5/.mesonpy-y542757w/meson-logs/meson-log.txt
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

I am not terribly well versed in coding so I am not 100% sure where to go from here: Any help would be greatly appreciated!

jbruce12000, 2024-07-22:
numpy is not be included in requirements.txt. Maybe something else is pulling in numpy.

Are you using the latest from the "main" branch?

jbruce12000, 2024-08-14:
I'll wait a week. If there is no response, then I'll close.